SzczepanLeon/esphome-components #1: VERSION "1.2.3" Not working properly

jakobs84, 2023-01-16:
After the last change 
`
12:57:30	[I]	[wmbus:044]	
Using driver 'apator162' for ID [0x0045XXXX] RSSI: -64 dBm T: 3E440106XXXX450005077A5000308512A2E18AE696258EDCD70E07EFBD6123E3E95243A2B0ED4A3387DC2B664DE3278EA17B914E2E866BA5072AF4925AB459 (63)
12:57:30	[E]	[wmbus:071]	
Can't get value from telegram for ID [0x0045XXXX] 'apator162'
12:57:30	[E]	[wmbus:072]	
T : 3E440106XXXX450005077A5000308512A2E18AE696258EDCD70E07EFBD6123E3E95243A2B0ED4A3387DC2B664DE3278EA17B914E2E866BA5072AF4925AB459 (63)
12:57:52	[I]	[wmbus:044]	
Using driver 'apator162' for ID [0x0045XXXX] RSSI: -64 dBm T: 3E440106XXXX450005077A510030852508D8045AE72F59636ABF543A93F734A7CD4F82EAAE2AEC971FAB17793B150891683FC4A5196FFB0B4A8F5458B0A8A5 (63)
12:57:52	[E]	[wmbus:071]	
Can't get value from telegram for ID [0x0045XXXX] 'apator162'
12:57:52	[E]	[wmbus:072]	
T : 3E440106XXXX450005077A510030852508D8045AE72F59636ABF543A93F734A7CD4F82EAAE2AEC971FAB17793B150891683FC4A5196FFB0B4A8F5458B0A8A5 (63)
`

SzczepanLeon, 2023-01-16:
Please add key option to sensor definition:
`key: "00000000000000000000000000000000"`

SzczepanLeon, 2023-01-16:
Please add key option to sensor definition:
key: "00000000000000000000000000000000"

SzczepanLeon, 2023-01-16:
RSSI is directly copied from CC1101. Do You have WiFi on Arduino? You can try to move CC1101 far from ESP WiFi.

closed 2023-01-16 by SzczepanLeon

jakobs84, 2023-01-16:
I have an arduino without WIFI, about 1m usb wires there orangePi with WIFI. Will try to increase the distance between esp and CC1101.