SzczepanLeon/esphome-components #2: How can I evaluate more telegram data ?

zibous, 2023-01-27:
Congratulations on the project. I want to reduce the dependencies and evaluate more 
data from the WMBUSMETER telegram. 

Running WMBUSMETER as a client needs a computer. So that I don't have this dependency, 
I would like to evaluate more data from the telegram

Is there a way to get addtional the following data:
- last_month_total_m3
- last_month_measure_date
- remaining_battery_life_y
- current_alarms
- previous_alarms

My configuration:
[image]

```
Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 43434418 dll-id (18444343)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 41150013CE5C5C324C2DBD48EEF6CB mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"********",
    "total_m3":448.798,
    "last_month_total_m3":438.675,
    "last_month_measure_date":"2023-01-01",
    "remaining_battery_life_y":10.5,
    "current_alarms":"no_alarm",
    "previous_alarms":"no_alarm",
    "transmit_period_s":8,
    "timestamp":"2023-01-27T14:55:40Z"
}

Using: wmbusmeters: 1.11.0-60-g89c42ab
89c42ab27dc8e7ac566e8b3a7810243a3bebd84b
https://wmbusmeters.org/
```

SzczepanLeon, 2023-01-27:
Please check:
https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_izar.cc
and
https://zewaren.net/wmbus-izar-meter.html

You can prepare pull request.

zibous, 2023-02-07:
@SzczepanLeon 

> You can prepare pull request.
Thank you for your information.

I started an inquiry, but it looks like nobody is reporting. 
The IZAR driver from wmbusmeters would be the right one, but I can't integrate it. 

I got a little further see: 
https://github.com/zibous/ha-watermeter/blob/master/esphome/README.md

closed 2023-05-08 by zibous