SzczepanLeon/esphome-components #4: Chances for Apator Amiplus?

KrzysztofHajdamowicz, 2023-02-01:
Hey!  
Thank you for a great project, I'm looking forward to implement it in my scenario.  
Only thing missing is a support for [amiplus](https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_amiplus.cc) driver.
Is there any chance to include it in your project?

SzczepanLeon, 2023-02-01:
Please check version 1.3.7

KrzysztofHajdamowicz, 2023-02-03:
I managed to run this today, looks like CC discovers only apator162, but no amiplus :(
`wmbus` initialized, but nothing in `sensor` section:
```
[12:27:51][C][wmbus:240]: wM-Bus v1.3.8:
[12:27:51][C][wmbus:257]:   CC1101 SPI bus:
[12:27:51][C][wmbus:258]:     MOSI Pin: GPIO13
[12:27:51][C][wmbus:259]:     MISO Pin: GPIO12
[12:27:51][C][wmbus:260]:     CLK Pin:  GPIO14
[12:27:51][C][wmbus:261]:     CS Pin:   GPIO15
[12:27:51][C][wmbus:262]:     GDO0 Pin: GPIO5
[12:27:51][C][wmbus:263]:     GDO2 Pin: GPIO4
[12:27:51][C][wmbus:269]:   Available drivers:  amiplus, apator08, apator162, bmeters, elf, evo868, hydrocalm3, izar, mkradio3, ultrimis, unismart
[12:29:36][D][wmbus:089]: Meter ID [0x03057338] RSSI: -54 dBm not found in configuration T: 6E44[redacted]E10 (111)
[12:30:55][D][wmbus:089]: Meter ID [0x03057338] RSSI: -54 dBm not found in configuration T: 6E44[redacted] (111)
```

What is weird, same antena was connected to RTL-SDR v3 pro, wmbusmeters were running with config
```
device=rtl433:868.95M:t1
```
In this case, wmbusmeters had no issues with picking up signal from electricity meter.

SzczepanLeon, 2023-02-03:
As you wrote antenna is the same but Rx electronic is not the same.
CC1101 is not so sensitive.

KrzysztofHajdamowicz, 2023-02-03:
OK, I managed to connect device to powerbank, get larger antenna and move it 10m closer to electricity meter.
Now, signals are read at around -71dBm 🎉 

Is it possible to expand amiplus driver with more decoded values?
Currently, there is `TOTAL_ENERGY_CONSUMPTION` and a lot more data can be decoded:
```
$ /wmbusmeters/wmbusmeters --listenvs=amiplus

METER_TOTAL_ENERGY_CONSUMPTION_KWH # it's currently here
METER_CURRENT_POWER_CONSUMPTION_KW
METER_TOTAL_ENERGY_PRODUCTION_KWH
METER_CURRENT_POWER_PRODUCTION_KW
METER_VOLTAGE_AT_PHASE_1_Volt
METER_VOLTAGE_AT_PHASE_2_Volt
METER_VOLTAGE_AT_PHASE_3_Volt
METER_DEVICE_DATE_TIME
METER_TOTAL_ENERGY_CONSUMPTION_TARIFF_1_KWH
METER_TOTAL_ENERGY_CONSUMPTION_TARIFF_2_KWH
METER_TOTAL_ENERGY_CONSUMPTION_TARIFF_3_KWH
METER_TOTAL_ENERGY_PRODUCTION_TARIFF_1_KWH
METER_TOTAL_ENERGY_PRODUCTION_TARIFF_2_KWH
METER_TOTAL_ENERGY_PRODUCTION_TARIFF_3_KWH
```

SzczepanLeon, 2023-02-03:
> Is it possible to expand amiplus driver with more decoded values?

Yes, it is possible.
But.... 
It require code refactoring (or rather re-writing), so it means free time :-(

KrzysztofHajdamowicz, 2023-02-03:
Thanks from the mountain ;)
Until that time, I managed to stream telegrams to wmbusmeters, so it can process telegrams and decode more data
https://gist.github.com/KrzysztofHajdamowicz/199c815c3a7fe7d2b445ee801d70f90e#file-home-assistant-esphome-wmbusmeters-md

SzczepanLeon, 2023-04-29:
Please check version 2.x

closed 2023-04-29 by SzczepanLeon