SzczepanLeon/esphome-components #5: CC1101 connection confirmation

MiG-41, 2023-02-18:
In ESPHome log of device i have this just after start:

[10:00:45][C][wmbus:259]:   CC1101 SPI bus:
[10:00:46][C][wmbus:260]:     MOSI Pin: GPIO13
[10:00:46][C][wmbus:261]:     MISO Pin: GPIO12
[10:00:46][C][wmbus:262]:     CLK Pin:  GPIO14
[10:00:46][C][wmbus:263]:     CS Pin:   GPIO15
[10:00:46][C][wmbus:264]:     GDO0 Pin: GPIO5
[10:00:46][C][wmbus:265]:     GDO2 Pin: GPIO4
[10:00:46][C][wmbus:275]:   Check connection to CC1101!

The same in  web page , the status is NA .
Is that a signal ,that is something wrong with CC1101 or ESP8266 ( i use chinese Wemos)  ,or finally with wirering , or it means nothing ,and my watermeter just not sending anything...

For info my connection CC1101 - Wemos is ( with above settings) :
VCC - 3,3V
GND -GND
MOSI - D7
SCLK- D5
MISO - D6
GDO2 - D2
GDO0 - D1
CS - D8

MiG-41, 2023-02-18:
I was told , that status NA on web page is till CC1101 not receive first telegram , so means nothing....
What about "Check connection to CC1101!" in log ?

SzczepanLeon, 2023-02-18:
> Check connection to CC1101!

It means that ESP can't connect to CC1101. So please chwck wiring and config.

MiG-41, 2023-02-18:
Right, wiring is ok , changed Wemos to NodeMCU , still the same. Looks like broken CC1101 then...

MiG-41, 2023-02-23:
Yes , confirmed , with new CC1101 log is like this:
[17:23:54][C][wmbus:259]:   CC1101 SPI bus:
[17:23:54][C][wmbus:260]:     MOSI Pin: GPIO13
[17:23:54][C][wmbus:261]:     MISO Pin: GPIO12
[17:23:54][C][wmbus:262]:     CLK Pin:  GPIO14
[17:23:54][C][wmbus:263]:     CS Pin:   GPIO15
[17:23:54][C][wmbus:264]:     GDO0 Pin: GPIO5
[17:23:54][C][wmbus:265]:     GDO2 Pin: GPIO4
[17:23:54][C][wmbus:272]:   Available drivers:  amiplus, apator08, apator162, bmeters, elf, evo868, fhkvdataiii, hydrocalm3, izar, mkradio3, mkradio4, ultrimis, unismart

On web page is still status NA ( but also not received any telegram yet...)

closed 2023-02-23 by MiG-41