SzczepanLeon/esphome-components #6: meter_id (Required, int): Meter ID (usually from sticker)....

zibous, 2023-02-27:
Hi,

> meter_id (Required, int): Meter ID (usually from sticker). Can be specified as decimal or hex.

In my case only `hex` is working:

```
- platform: wmbus
    name: "My Water from Izar"
    meter_id: 18444343
    type: izar
    accuracy_decimals: 3    
    
[14:03:14][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418A251150013CEE5577BE7B53D9135EFCA (26)
[14:03:23][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418A261150013CEDC26A11AD749FB58C4CD (26)
[14:03:32][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418A271150013CECB09E8B1F69A2283DDCF (26)

```

This is working:
```
- platform: wmbus
    name: "My Water from Izar"
    meter_id: 0x43430778
    type: izar
    accuracy_decimals: 3
    
[14:08:40][I][wmbus:060]: Using driver 'izar' for ID [0x43430778] RSSI: -63 dBm T: 1944A511780743434418A231150013CE97B4CE1D71D545EFB9C5 (26)
```

SzczepanLeon, 2023-02-27:
But:
18444343 in DEC is 0x1197037 in HEX
0x43430778 in HEX is 1128466296 in DEC

closed 2023-02-27 by SzczepanLeon

zibous, 2023-02-27:
o.k But 18444343 is my Meter ID from sticker. Mybe this is Austria based.

[image]

SzczepanLeon, 2023-02-27:
> 18444343 is my Meter ID from sticker.

This is serial number, not meterId (dll-id from wM-Bus frame)

You can see it here:
https://wmbusmeters.org/analyze/1944304C72242421D401A2013D4013DD8B46A4999C1293E582CC

meterId - 0x21242472
serialNumber - 145842

zibous, 2023-02-27:
yes:

Decoded telegram

> 1944A511780743434418A241150013CE0766324C94EE48EEF6C8

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 43434418 dll-id (18444343)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 71150013CE3C17E8B1F69A2283DDCF mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"18444343",
    "total_m3":451.399,
    "last_month_total_m3":451.114,
    "last_month_measure_date":"2023-02-01",
    "remaining_battery_life_y":10.5,
    "current_alarms":"no_alarm",
    "previous_alarms":"no_alarm",
    "transmit_period_s":8,
    "timestamp":"2023-02-02T13:49:16Z"
}

Using: wmbusmeters: 1.11.0-60-g89c42ab
89c42ab27dc8e7ac566e8b3a7810243a3bebd84b

So the `18444343`is  the id and not the serial number.

I was just not sure because with wmbusmeters I could use the id.

```
name=watermeter
type=izar
id=18444343
key=
```

Thanks, good job

SzczepanLeon, 2023-02-27:
Let's go back one step.

You have telegram:
`1944A511780743434418A241150013CE0766324C94EE48EEF6C8`

and it can be decoded [here](https://wmbusmeters.org/analyze/1944A511780743434418A241150013CE0766324C94EE48EEF6C8).

From that telegram You have:
```
004   : 43434418 dll-id (18444343)
...
    "id":"18444343",
```

And that ID value is meterId and also it is equal to serialNumber (this is special case, as You can see there is no serialNumber field in telegram).
But that meterId (dll-id) from wmbusmeters.org is in HEX format, so in my component You should have:
```
- platform: wmbus
    name: "My Water from Izar"
    meter_id: 0x18444343
    type: izar
```

or
```
- platform: wmbus
    name: "My Water from Izar"
    meter_id: 407126851
    type: izar
```
Anyway both traces
```
[14:03:14][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418 ...
[14:08:40][I][wmbus:060]: Using driver 'izar' for ID [0x43430778] RSSI: -63 dBm T: 1944A511780743434418 ...
```

are strange, because there is no 0x43430778 ID in telegram ;-)
Probably You have old component version (please reinstall) with some bug around meterId calculation.

zibous, 2023-02-27:
> are strange, because there is no 0x43430778 ID in telegram ;-)

Yes incomprehensible (43434418 dll-id (18444343)

```
meter_id: 0x18444343 
[16:40:21][D][wmbus:101]: Meter ID [0x43430778] RSSI: -64 dBm not found in configuration T: 1944A511780743434418A211150013CEBCEA5D4B3272F6598BC0 (26)
[16:40:21][D][wmbus:101]: Meter ID [0x43410778] RSSI: -92 dBm not found in configuration T: 1944A511780741434418A231150013CCC3ECFE796F737B8C9993 (26)
[16:40:29][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418A221150013CE859B87B650069C34A0C7 (26)
[16:40:37][D][wmbus:101]: Meter ID [0x43430778] RSSI: -62 dBm not found in configuration T: 1944A511780743434418A231150013CE92B4CE1D71D545EFB9C5 (26)
[16:40:46][D][wmbus:101]: Meter ID [0x43430778] RSSI: -63 dBm not found in configuration T: 1944A511780743434418A241150013CEF778324C94EE48EEF6C8 (26)

meter_id: 0x43430778
[16:44:37][I][wmbus:060]: Using driver 'izar' for ID [0x43430778] RSSI: -63 dBm T: 1944A511780743434418A271150013CECE09E8B1F69A2283DDCF (26)
[16:44:46][I][wmbus:060]: Using driver 'izar' for ID [0x43430778] RSSI: -63 dBm T: 1944A511780743434418A201150013CEABC514E013A12F8292C2 (26)

```

but  on wmbusmeters i can see many `coded differently from standard?...`
https://github.com/weetmuts/wmbusmeters/blob/master/src/manufacturer_specificities.cc

But it is not so tragic, since the log shows which ID to use. 
Even if it is a bit difficult for me, because I get several data from water meters 
that are installed at my neighbors. But since I know the amount of water, I can find out mine.

You just have to know that this water meter only works with HEX.

**Text sensor:** 
I don't understand it, because if I use it, I won't get any data.
```
text_sensor:

  - platform: wmbus
    name: "Wasseruhr"
    meter_id:  0x43430778
    type: izar
```
 
What is this for?

SzczepanLeon, 2023-02-27:
That text sensor was in one of old version just for debug purposes. Now it is not used, but not removed yet.