SzczepanLeon/esphome-components #8: Support of Quindis driver "qwater" &6 "qheat"

uwec, 2023-03-09:
Is it possible, that driver for water and heat meter from Qundis can be supported?

Water meter:  `qwater` https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_qwater.cc
Heat meter: `qheat`  https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_qheat.cc

Use C1101 + ESP8266.

Thanks in advance for any support. I am not sure, how to adapt these drivers form wmbusmeters, that they are working with your code?!

At the moment I get in the Logs:
`[C][wmbus:340]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, elf, evo868, fhkvdataiii, hydrocalm3, itron, izar, mkradio3, mkradio4, ultrimis, unismart, vario451
[D][api:102]: Accepted 192.168.178.47
[D][api.connection:918]: Home Assistant 2023.3.2 (192.168.178.47): Connected successfully
wMBus-lib: Error during decoding '3 out of 6'wMBus-lib`

SzczepanLeon, 2023-03-10:
qwater added in 1.4.7
qheat added in 1.4.8

Please check results.

uwec, 2023-03-10:
Thanks a lot for the fast implementation!

Get still error in Log `wMBus-lib: Error during decoding '3 out of 6'`
Using version 1.4.8., `qwater` and `qheat` shwoing up in the log.

Tried with both meters together, and each one by one. Always the same entry in the log.

[image]

Meters are sending telegrams. If I use a RTLSDR receiver, I receive correct  telegrams.

SzczepanLeon, 2023-03-10:
Show me log from RtlSdr. Maybe it is not T1 or it is too far.

uwec, 2023-03-10:
good hint:

attached the telegram of the water meter
`telegram=|3C4493442071413533377246736324934433065E000020_0C13980100004C1317000000426CDF2CCC081316000000C2086CFC22326CFFFF046D1D12EA23|+42`

seems like it is C telegram

SzczepanLeon, 2023-03-10:
My lib dupports only T1 :(

uwec, 2023-03-10:
oh, what a bummer.
The heat meater seems also using C telegrams.
No plans to extend to C telegrams :-) ?

SzczepanLeon, 2023-03-10:
I have plans but no time :(

uwec, 2023-03-10:
Can understand. Neverthless thanks a lot for your support und your effort! Will close the issue. 
Not sure if there are T1 meters from Qundis, then your libary can be used.

SzczepanLeon, 2023-05-05:
Can You send me logs from rtlwmbus. I'm trying to add C1 mode.

uwec, 2023-05-05:
Hi Leon, do you mean additional logs or the sent telegrams?
For qheat meter:
`telegram=|3C449344110521372337720735036793442304B8000020_0C05060000004C0500000000426CDF2CCC080506000000C2086CFE24326CFFFF046D1D11E525|+333`

For qwater meter:
`telegram=|3C4493442071413533377246736324934433065E000020_0C13980100004C1317000000426CDF2CCC081316000000C2086CFC22326CFFFF046D1D12EA23|+42`

SzczepanLeon, 2023-05-06:
I tought rather about raw logs from rtlwmbus, but anyway I don't need it now.

I have first beta (not everything is OK). Can You test it?

If yes then config is below:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@wmbus_c1
    components: [ wmbus ]

wmbus:
  log_unknown: True
```

Please don't define sensors and send me logs. They should be like:
`Meter ID [0x12345678] RSSI: -62 dBm LQI: 142 Mode: C1`

uwec, 2023-05-06:
Tested it several times but I am getting only such log
[image]

SzczepanLeon, 2023-05-06:
Great. C1 frame received.
As I wrote it is not completed. Should be ready around 2 days.

SzczepanLeon, 2023-05-06:
Can You try one more time (install & collect logs)? I've added couple traces.

uwec, 2023-05-06:
After your updates I get some errors while compiling
[logs_mbusmeters_compile.txt](https://github.com/SzczepanLeon/esphome-components/files/11413254/logs_mbusmeters_compile.txt)

SzczepanLeon, 2023-05-06:
Ach, typo. Try now.

uwec, 2023-05-06:
Still an error
[logs_compile.txt](https://github.com/SzczepanLeon/esphome-components/files/11413295/logs_compile.txt)

SzczepanLeon, 2023-05-06:
Sorry, it is hard to edit on small screen. Try now.

uwec, 2023-05-06:
Looks good, compilation successfull again.
```
pm open,type:0 0
wMBus-lib: L=336 l=336
wMBus-lib: Frame: 0xD10x960x9A0x530xA40xD50x950xE80x7F0x7A0xDF0x830x100xE80x630xE70xC70x160xE20xEB0x760xA70x9B0x8C0x8B0x3E0xAA0x750x3F0x860x250xF70x210xDD0x700x2A0x970xE30xAF0xA20xC60x170x5A0x990x5F0x170x1E0x850xE10x4C0xE20x2A0x000x120x820x590x120x000x5D0x000x250xE00xB00x4C0x8D0x560x620x180xFE0x000x000xC00x1E0xF10xC00x5C0xC60x450xC50xE70x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x00
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'
[V][component:204]: Component wmbus took a long time for an operation (0.12 s).
[V][component:205]: Components should block for at most 20-30ms.
wMBus-lib: L=221 l=221
wMBus-lib: Frame: 0xB10x660x6C0xDD0xF00x910x2D0xF70xC70xF50xBC0xEB0x4E0x930xDF0x7D0x690xC70x8F0x9E0x320xD20xD70x3E0x4A0x490x630xEB0xAF0x000x590x7F0xA00xC40xE10xFA0xD90xF80x7E0xFD0x700x990x440x7F0x310x620xC70xC30x7F0xFF0xFF0xFF0xFF0xFF0xEF0x3C0x690x4B0x8B0x9E0x8E0x2C0xB50xD60x150x1D0xD20x7B0x460x090x700xFD0x890x260x320x3E0xDC0x810x8A0x5C0x700xC70xCF0x160x7D0x970x4B0xF50xDD0x590x0D0xF80xFF0xFF0xFF0x9F0xFF0xFF0xFF0x7B0x140x440x8E0xF30x890x000x1C0x5C0x7B0xD40x2A0xAF0xB70x080xA60x8F0x9D0xA20xE50x430x7F0x910xC70x460xCB0xBA0xC70x9A0x390xDA0x7A0x150xEC0xF00x820x570x0F0xFF0xFD0xFF0xFF0xFF0xFF0xFD0x7E0xD30x960x0B0x630x420x830x780xFB0x480x680x1A0xED0xFC0x080x2A0xC70xA90x7C0x490x290x9E0xC00x210xDB0xE70x500x000xBA0xFB0xDB0x3B0x4D0x2F0x7C0xD80x9D0xE30x3F0xFF0x7F0xFF0xFF0xFF0xFF0xB70xBC0x310x190x5E0x3C0xCB0x920xD70x1C0xB10x420x470xBE0x6C0xC50x8F0xA20xC30x540x910x7D0xEA0x110x4E0x5E0xDB0x410x1A0xAC0x930xEF
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'
[V][component:204]: Component wmbus took a long time for an operation (0.08 s).
[V][component:205]: Components should block for at most 20-30ms.
wMBus-lib: L=73 l=107
wMBus-lib: Frame: 0x540xCD0x3C0x440x930x440x110x050x210x370x230x370x670x8E0x720x070x350x030x670x930x440x230x040x910x000x000x200x0C0x050x060xBF0xE50x000x000x000x4C0x050x000x000x000x000x420x6C0xDF0x2C0xCC0x080x050xA20x9B0x060x000x000x000xC20x080x6C0xFE0x240x320x6C0xFF0xFF0x040x6D0x290x570x120x140xE60x250x3D0xDA
wMBus-lib: Processing C1 A frame
wMBus-lib:  Error during decoding 'CRC'
wMBus-lib: L=428 l=428
wMBus-lib: Frame: 0xA60x630xA50xF30x000xB00x750xF80x6F0xBB0xAF0x110x330xAB0xBA0x000xCA0xCF0x720x110x830x2F0xFC0xB20xFB0x5F0xE20x760x160xFD0xD40x040x110x730x290x300x670xF60xFF0x550x930xCC0xE20x060x110xDE0x800x0A0x6B0x9F0xFF0xFF0xFF0xFF0xFF0xD50xE20x510x2E0x1E0xFE0xA30x8C0x7C0xF00xDC0xF70xB70xD70x070xC00xF30xB40x850x790x7F0xBC0xC00xD50x730x710x9C0x660x260x4C0x3C0xDE0x3B0xB20x8C0x2B0x730xB60x450x200xEF0xFF0xFF0xFF0xFF0xFF0xFF0x7B0x880xF40xD80x730xA00x4D0x310x6F0xA20xCA0xD60xE70x000x970xEA0xC60xFC0x830xD20xCE0xEF0xC30x9D0x9A0x590x3A0xED0xEC0xD20xA60x2B0x8E0x010x090x070xDC0xED0xC70xBF0xFF0xFF0xFF0xFE0xFF0xFB0xF70x780x0D0x8D0xCE0xF20x5F0x230xBF0xF60x220x5B0x5D0xE00x220x3E0xD90x990x340x140xB50x800x920x6D0x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x00
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'

uwec, 2023-05-06:
another extract from the log....

```
wMBus-lib: L=73 l=107
wMBus-lib: Frame: 0x540xCD0x3C0x440x930x440x110x050x210x370x230x370x670x8E0x720x070x350x030x670x930x440x230x040x920x000x000x200x0C0x050x060x940xC10x000x000x000x4C0x050x000x000x000x000x420x6C0xDF0x2C0xCC0x080x050xA20x9B0x060x000x000x000xC20x080x6C0xFE0x240x320x6C0xFF0xFF0x040x6D0x300x2D0xC50x140xE60x250x3D0xDA
wMBus-lib: Processing C1 A frame
wMBus-lib:  Error during decoding 'CRC'
wMBus-lib: L=341 l=341
wMBus-lib: Frame: 0xD20xCC0x710x070x320x780x2A0x670xEE0xEF0xDB0xAE0xE50x420x710xF70x810x280x290x7D0xE30x540xB50x3E0xDF0x290xAB0xC30xFC0xB50x210x510x460x390x920xE70xF50xFA0x180x7A0xF90xE90xA10x880x5C0xEA0x800x9D0x8E0x5D0xFF0xF10xFF0xDF0xFF0xFE0xF70x120x0D0xB50xB10xC00x4E0x160x1D0xE20x500x010xA60xB30x710x2F0x230x3C0xE10xA90xB60xD70x910xC60x330x750x8B0x470xFC0x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x000x00
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'

jonas240, 2023-06-02:
[logs_esp-watermeter_logs.txt](https://github.com/SzczepanLeon/esphome-components/files/11638880/logs_esp-watermeter_logs.txt)

Some more logs generated with this branch, maybe it will help. I have a qwater water meter; however, I am uncertain if the logged frames are actually from it or from something different (like from devices from neighbors)

SzczepanLeon, 2023-06-05:
Please collect more logs with latest version on that C1 branch.

jonas240, 2023-07-07:
Hi, @SzczepanLeon , some more logs:
[logs_esp-watermeter_logs_debug.txt](https://github.com/SzczepanLeon/esphome-components/files/11983924/logs_esp-watermeter_logs_debug.txt)

SzczepanLeon, 2023-07-21:
Please try version 2.2.22 from main branch.

uwec, 2023-07-22:
I have tested succesfully a `qwater` and a `qheat` meter with version 2.2.22. 
See attached logs. 

````
[C][wmbus:369]: wM-Bus v2.2.22:
[C][wmbus:382]:   LED:
[C][wmbus:383]:     Pin: GPIO0
[C][wmbus:384]:     Duration: 1000 ms
[C][wmbus:386]:   CC1101 SPI bus:
[C][wmbus:387]:     MOSI Pin: GPIO13
[C][wmbus:388]:     MISO Pin: GPIO12
[C][wmbus:389]:     CLK Pin:  GPIO14
[C][wmbus:390]:     CS Pin:   GPIO2
[C][wmbus:391]:     GDO0 Pin: GPIO5
[C][wmbus:392]:     GDO2 Pin: GPIO4
[C][wmbus:399]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, iperl, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451
[C][wmbus:417]:   Meter:
[C][wmbus:418]:     ID: 893481248 [0x35417120]
[C][wmbus:419]:     Type: qwater
[C][wmbus:420]:     Key: ''
[C][wmbus:422]:     Sensor '0x35417120 Wasser_OG'
[C][wmbus:422]:       Device Class: 'water'
[C][wmbus:422]:       State Class: 'total_increasing'
[C][wmbus:422]:       Unit of Measurement: 'm³'
[C][wmbus:422]:       Accuracy Decimals: 3
[C][wmbus:422]:       Icon: 'mdi:water'
[C][wmbus:417]:   Meter:
[C][wmbus:418]:     ID: 924910865 [0x37210511]
[C][wmbus:419]:     Type: qheat
[C][wmbus:420]:     Key: ''
[C][wmbus:422]:     Sensor '0x37210511 Heizung'
[C][wmbus:422]:       Device Class: 'energy'
[C][wmbus:422]:       State Class: 'total_increasing'
[C][wmbus:422]:       Unit of Measurement: 'kWh'
[C][wmbus:422]:       Accuracy Decimals: 3
[C][wmbus:422]:       Icon: 'mdi:transmission-tower-export'
[D][api:102]: Accepted 192.168.178.48
[W][component:204]: Component api took a long time for an operation (0.28 s).
[W][component:205]: Components should block for at most 20-30ms.
[V][api.connection:1012]: Hello from client: 'Home Assistant 2023.7.3 (192.168.178.48)' | API Version 1.9
[D][api.connection:1031]: Home Assistant 2023.7.3 (192.168.178.48): Connected successfully
fpm close 3 
pm open,type:0 0
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'
[I][ota:113]: Boot seems successful, resetting boot loop counter.
wMBus-lib: Processing C1 A frame
 FullFrame: 0x54, 0xCD, 0x3C, 0x44, 0x93, 0x44, 0x20, 0x71, 0x41, 0x35, 0x33, 0x37, 0x75, 0xE3, 0x72, 0x46, 0x73, 0x63, 0x24, 0x93, 0x44, 0x33, 0x06, 0xF2, 0x00, 0x00, 0x20, 0x0C, 0x13, 0x85, 0xD5, 0x43, 0x30, 0x00, 0x00, 0x4C, 0x13, 0x17, 0x00, 0x00, 0x00, 0x42, 0x6C, 0xDF, 0x2C, 0xCC, 0x08, 0x13, 0xCE, 0xA2, 0x97, 0x25, 0x00, 0x00, 0xC2, 0x08, 0x6C, 0xFE, 0x26, 0x32, 0x6C, 0xFF, 0xFF, 0x04, 0x6D, 0x32, 0xC4, 0x04, 0x14, 0xF6, 0x27, 0x6E, 0xE6, 
   3C449344207141353337 8A1C [75E3]    724673632493443306F20000200C1385 2ABC [D543] 3000004C1317000000426CDF2CCC0813 315D [CEA2] 97250000C2086CFE26326CFFFF046D32 3BFB [C404]    14F627 9119 [6EE6] 
[I][wmbus:072]: Using driver 'qwater' for ID [0x35417120] RSSI: -86 dBm LQI: 137 Mode: C1 T: 3C449344207141353337724673632493443306F20000200C13853000004C1317000000426CDF2CCC081397250000C2086CFE26326CFFFF046D3214F627 (61)
[V][sensor:043]: '0x35417120 Wasser_OG': Received new state 3.085000
[D][sensor:093]: '0x35417120 Wasser_OG': Sending state 3.08500 m³ with 3 decimals of accuracy
[W][component:204]: Component wmbus took a long time for an operation (0.09 s).
[W][component:205]: Components should block for at most 20-30ms.
wMBus-lib: Processing T1 A frame
wMBus-lib:  Error during decoding '3 out of 6'
[D][sntp:077]: Synchronized time: 2023-07-22 21:49:28
wMBus-lib: Processing C1 A frame
 FullFrame: 0x54, 0xCD, 0x3C, 0x44, 0x93, 0x44, 0x11, 0x05, 0x21, 0x37, 0x23, 0x37, 0x67, 0x8E, 0x72, 0x07, 0x35, 0x03, 0x67, 0x93, 0x44, 0x23, 0x04, 0x48, 0x00, 0x00, 0x20, 0x0C, 0x05, 0x06, 0x4C, 0x18, 0x00, 0x00, 0x00, 0x4C, 0x05, 0x00, 0x00, 0x00, 0x00, 0x42, 0x6C, 0xDF, 0x2C, 0xCC, 0x08, 0x05, 0xA2, 0x9B, 0x06, 0x00, 0x00, 0x00, 0xC2, 0x08, 0x6C, 0xFE, 0x26, 0x32, 0x6C, 0xFF, 0xFF, 0x04, 0x6D, 0x2F, 0xFE, 0xC1, 0x13, 0xF6, 0x27, 0xAC, 0x7A, 
   3C449344110521372337 9871 [678E]    720735036793442304480000200C0506 B3E7 [4C18] 0000004C0500000000426CDF2CCC0805 5D64 [A29B] 06000000C2086CFE26326CFFFF046D2F 013E [FEC1]    13F627 5385 [AC7A] 
[I][wmbus:072]: Using driver 'qheat' for ID [0x37210511] RSSI: -55 dBm LQI: 128 Mode: C1 T: 3C449344110521372337720735036793442304480000200C05060000004C0500000000426CDF2CCC080506000000C2086CFE26326CFFFF046D2F13F627 (61)
[V][sensor:043]: '0x37210511 Heizung': Received new state 0.600000
[D][sensor:093]: '0x37210511 Heizung': Sending state 0.60000 kWh with 3 decimals of accuracy
[W][component:204]: Component wmbus took a long time for an operation (0.09 s).
[W][component:205]: Components should block for at most 20-30ms.````

closed 2023-07-24 by SzczepanLeon