SzczepanLeon/esphome-components #11: Missing rf_mbus.hpp file - compilation error

VivantSenior, 2023-03-24:
Hi,

unfortunately during compilation I got an error:
`src/esphome/components/wmbus/wmbus.h:14:10: fatal error: rf_mbus.hpp: No such file or directory`

Versions:
Home Assistant: 2023.3.5
ESPHome addon: 2023.3.1

Compilation logs:
```
INFO Reading configuration /config/esphome/watermeter-esp.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing watermeter-esp (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.15
INFO Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.15
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-e16r6t4u'...
Library Manager: wMbus-lib@0.9.15+sha.25dfb37 has been installed!
INFO wMbus-lib@0.9.15+sha.25dfb37 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing lsatan/SmartRC-CC1101-Driver-Lib @ ^2.5.7
INFO Installing lsatan/SmartRC-CC1101-Driver-Lib @ ^2.5.7
Unpacking  [####################################]  100%
Library Manager: SmartRC-CC1101-Driver-Lib@2.5.7 has been installed!
INFO SmartRC-CC1101-Driver-Lib@2.5.7 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/api_connection.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/api_frame_helper.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/api_pb2.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/api_pb2_service.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/api_server.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/list_entities.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/proto.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/subscribe_state.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/api/user_services.o
Compiling /data/watermeter-esp/.pioenvs/watermeter-esp/src/esphome/components/bluetooth_proxy/bluetooth_connection.o
[... 127 lines cut ...]
#     icon: "mdi:water-pump"
#     unit_of_measurement: m³
#     device_class: "water"
#     state_class: "total_increasing"
#     meter_id: yyyyyyyy
#     type: apator162
#     key: "00000000000000000000000000000000"

#   - platform: wmbus
#     name: "Garden water total usage"
#     icon: "mdi:water-pump"
#     unit_of_measurement: m³
#     device_class: "water"
#     state_class: "total_increasing"
#     meter_id: xxxxxxxx
#     type: apator162
#     key: "00000000000000000000000000000000"

  - platform: wifi_signal
    name: "${friendly_name} WiFi signal"
    update_interval: 10s
    
  - platform: uptime
    name: "${friendly_name} uptime"
```

SzczepanLeon, 2023-03-24:
Please do cleanup before compiling.

VivantSenior, 2023-03-24:
> Please do cleanup before compiling.

I’ve done cleanup before each compilation attempt.

SzczepanLeon, 2023-03-24:
Do You need **esp-idf** as framework?

If not then try:

```
esp32:
  board: esp32dev
  framework:
    type: arduino
```

VivantSenior, 2023-03-25:
I used `esp-df` because it's recommended in the ESPHome [bluetooth_proxy](https://esphome.io/components/bluetooth_proxy.html) documentation.

After switching to the `arduino` framework I was able to compile the code.

```
Successfully created esp32 image.
```

Thanks for spotting the issue :)

closed 2023-03-25 by VivantSenior