SzczepanLeon/esphome-components #12: [Question] Is it possible to get additional data like last update timestamp and RSSI value

VivantSenior, 2023-04-01:
Hi,

I would like to pass from ESPHome to my Home Assistant instance information regarding last data update and RSSI value (per meter). Is it possible?

I think last update data, RSSI and (calculated) signal strength should be in Diagnostic category. Formula for signal strength is as follows: `2 * (RSSI + 100)`.

This would be a configuration (based on a MQTT sensor):

Timestamp:
```
"device_class": "timestamp",
"icon": "mdi:clock-start"
"entity_category": "diagnostic"
```

RSSI:
```
"value_template": "{{ value_json.rssi_dbm }}",
"unit_of_measurement": "dBm",
"state_class": "measurement",
"icon": "mdi:antenna",
"entity_category": "diagnostic"
```

Signal strength:
```
"value_template": "{{ ((value_json.rssi_dbm|int + 100) * 2) }}",
"unit_of_measurement": "%",
"state_class": "measurement",
"icon": "mdi:wifi",
"entity_category": "diagnostic"
```

[image]

Best,
Viv

SzczepanLeon, 2023-04-01:
Currently only using this solution:
https://github.com/SzczepanLeon/esphome-components/blob/main/docs/wmbusgw.md

VivantSenior, 2023-04-01:
Thank you for the prompt response.

For the last sensor update timestamp I was also thinking to create template sensor and automation that will update its value when the sensor containing water usage value will change.

VivantSenior, 2023-04-08:
In terms of workaround for last update timestamp you can use [template sensor](https://www.home-assistant.io/integrations/template/) defined in your sensors.yaml or configuration.yaml:
```
- platform: template
  sensors:
    main_water_measurement_timstamp:
      friendly_name: Main water measurement timestamp
      unique_id: 13870c9e7d44209c910a5f5ae361cc08 #just a md5 hash of friendly_name: https://www.md5hashgenerator.com/
      icon_template: mdi:clock-start
      value_template: "{{ relative_time(states.sensor.main_water_total_usage.last_changed) }}" #name of the main sensor
```

closed 2023-04-08 by VivantSenior