SzczepanLeon/esphome-components #15: wmbusgw esphome run error -

babajun12, 2023-04-24:
Received an error while creating wmbusgw - is there an update needed?

**Error: Package version 0.9.15+sha.25dfb37 doesn't satisfy requirements 0.9.14 based on PackageMetaData <type=library name=wMbus-lib version=0.9.15+sha.25dfb37**

```
`C:\Users\Gerhard\.esphome>esphome run wmbus-gw.yaml
INFO Reading configuration wmbus-gw.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing wmbus-gw (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
INFO Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.14
INFO Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.14
git version 2.40.0.windows.1
Cloning into 'C:\Users\Gerhard\.platformio\.cache\tmp\pkg-installing-tkpqd1pc'...
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (17/17), done.
Receiving objects:  60% (12/20)sed 12 (delta 3), pack-reused 0
Receiving objects: 100% (20/20), 20.39 KiB | 5.10 MiB/s, done.
Resolving deltas: 100% (5/5), done.
Error: Package version 0.9.15+sha.25dfb37 doesn't satisfy requirements 0.9.14 based on PackageMetaData <type=library name=wMbus-lib version=0.9.15+sha.25dfb37 spec={'owner': None, 'id': None, 'name': 'wMbus-lib', 'requirements': '0.9.14', 'uri': 'git+https://github.com/SzczepanLeon/wMbus-lib'}`
```

SzczepanLeon, 2023-04-25:
Please update YAML:
```
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]
```

and recompile it.

babajun12, 2023-04-25:
Got a new error but will try to compile on a different computer later on.
PS: Did a `run clean` before

```

INFO Installing git+https://github.com/SzczepanLeon/wMbus-lib @ 0.9.14
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-9qct6i29'...
Error: Package version 1.0.3+sha.2a59805 doesn't satisfy requirements 0.9.14 based on PackageMetaData <type=library name=wMbus-lib version=1.0.3+sha.2a59805 spec={'owner': None, 'id': None, 'name': 'wMbus-lib', 'requirements': '0.9.14', 'uri': 'git+https://github.com/SzczepanLeon/wMbus-lib'}
```

babajun12, 2023-04-25:
Ok, found my error:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

**wmbusgw:  <<----- needs to be "wmbus" as well ;-)
  mosi_pin: GPIO13
  miso_pin: GPIO5
  clk_pin:  GPIO2
  cs_pin:   GPIO14
  gdo0_pin: GPIO15
  gdo2_pin: GPIO16
```

closed 2023-04-25 by SzczepanLeon