SzczepanLeon/esphome-components #17: Component wmbus was marked as failed

ArmenVanlasso, 2023-04-27:
Sprawdziłem okablowanie.
Wemos D1 Mini
Log:
`ld��|�l�<�d�c|����r�b�#��o'�log���cp��${l{lx�o��l��lco�|�Ǐ#��og�lćd`�gn�dg{���g{r�`p�'�{������co�<�#��og��l �g'�lgs�ۓosl`{ۓosl`��$l ��o�d[I][logger:259]: Log initialized
[C][ota:469]: There have been 2 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:039]:   Local MAC: D8:F1:5B:12:E3:B7
[D][wifi:387]: Starting scan...
[D][wifi:402]: Found networks:
[I][wifi:445]: - 'SYRION16262' [redacted]▂▄▆█
[D][wifi:447]:     Channel: 5
[D][wifi:448]:     RSSI: -63 d[W][wifi_esp8266:482]: Event: Disconnected ssid='SYRION16262' bssid=[redacted] reason='Auth Expired'
[W][wifi:550]: Error while connecting to network.
[W][wifi:586]: Restarting WiFi adapter...
[I][wifi:258]: WiFi Connecting to 'SYRION16262'...
[I][wifi:519]: WiFi Connected!
[C][wifi:363]:   Local MAC: D8:F1:5B:12:E3:B7
[C][wifi:364]:   SSID: [redacted]
[C][wifi:365]:   IP Address: 192.168.254.77
[C][wifi:366]:   BSSID: [redacted]
[C][wifi:368]:   Hostname: 'wodomierz'
[C][wifi:370]:   Signal strength: -65 dB ▂▄▆█
[C][wifi:374]:   Channel: 5
[C][wifi:375]:   Subnet: 255.255.255.0
[C][wifi:376]:   Gateway: 192.168.254.254
[C][wifi:377]:   DNS1: 192.168.254.254
[C][wifi:378]:   DNS2: 8.8.8.8
[D][wifi:528]: Disabling AP...
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.254.77:8266
[C][ota:097]:   Using Password.
[W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[C][api:025]: Setting up Home Assistant API server...
[C][sntp:028]: Setting up SNTP...

wMBus-lib: CC1101 initialization FAILED!
[E][component:113]: Component wmbus was marked as failed.
[E][wmbus:024]: CC1101 initialization failed.
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2023.4.1 compiled on Apr 27 2023, 18:13:00
[C][wifi:505]: WiFi:
[C][wifi:363]:   Local MAC: D8:F1:5B:12:E3:B7
[C][wifi:364]:   SSID: [redacted]
[C][wifi:365]:   IP Address: 192.168.254.77
[C][wifi:366]:   BSSID: [redacted]
[C][wifi:368]:   Hostname: 'wodomierz'
[C][wifi:370]:   Signal strength: -67 dB ▂▄▆█
[C][wifi:374]:   Channel: 5
[C][wifi:375]:   Subnet: 255.255.255.0
[C][wifi:376]:   Gateway: 192.168.254.254
[C][wifi:377]:   DNS1: 192.168.254.254
[C][wifi:378]:   DNS2: 8.8.8.8
[C][logger:294]: Logger:
[C][logger:295]:   Level: DEBUG
[C][logger:296]:   Log Baud Rate: 115200
[C][logger:297]:   Hardware UART: UART0
[C][wmbus_sensor:016]: wM-Bus Sensor 'My water from Apator'
[C][wmbus_sensor:016]:   Device Class: 'water'
[C][wmbus_sensor:016]:   State Class: 'total_increasing'
[C][wmbus_sensor:016]:   Unit of Measurement: 'm³'
[C][wmbus_sensor:016]:   Accuracy Decimals: 3
[C][wmbus_sensor:016]:   Icon: 'mdi:water'
[C][wmbus_sensor:022]:   Key: '00000000000000000000000000000000'
[C][wmbus_sensor:023]:   Type: iperl
[C][wmbus_sensor:024]:   ID: 606085152 [0x24202020]
[D][sntp:077]: Synchronized time: 2023-04-27 18:24:32
[D][main:215]: Synchronized sntp clock
[C][captive_portal:088]: Captive Portal:
[C][mdns:108]: mDNS:
[C][mdns:109]:   Hostname: wodomierz
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.254.77:8266
[C][ota:097]:   Using Password.
[W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[C][api:138]: API Server:
[C][api:139]:   Address: 192.168.254.77:6053
[C][api:141]:   Using noise encryption: YES
[C][sntp:053]: SNTP Time:
[C][sntp:054]:   Server 1: '0.pool.ntp.org'
[C][sntp:055]:   Server 2: '1.pool.ntp.org'
[C][sntp:056]:   Server 3: '2.pool.ntp.org'
[C][sntp:057]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[C][wmbus:312]: wM-Bus v1.5.1:
[C][wmbus:329]:   CC1101 SPI bus:
[C][wmbus:330]:     MOSI Pin: GPIO13
[C][wmbus:331]:     MISO Pin: GPIO12
[C][wmbus:332]:     CLK Pin:  GPIO14
[C][wmbus:333]:     CS Pin:   GPIO15
[C][wmbus:334]:     GDO0 Pin: GPIO5
[C][wmbus:335]:     GDO2 Pin: GPIO4
[C][wmbus:345]:   Check connection to CC1101!
[D][api:102]: Accepted 192.168.254.223
[D][api.connection:961]: Home Assistant 2023.4.6 (192.168.254.223): Connected successfully
[I][ota:113]: Boot seems successful, resetting boot loop counter.`
Config: [config](https://pastebin.com/aKGsjDi7)

SzczepanLeon, 2023-04-27:
CC1101 jest na 100% sprawny?

SzczepanLeon, 2023-04-27:
> wMBus-lib: CC1101 initialization FAILED!
> [E][component:113]: Component wmbus was marked as failed.
> [E][wmbus:024]: CC1101 initialization failed

Tak czy inaczej wygląda to na złe połączenie do CC1101 albo źle ustawiine ping do niego albo padnięty CC1101

closed 2023-04-29 by SzczepanLeon