SzczepanLeon/esphome-components #18: Adding driver for Topas meters

sanchosk, 2023-04-28:
Hi.
Arriving here from https://github.com/zibous/ha-watermeter.
I have trouble reporting my meter.
It seems on https://github.com/wmbusmeters/wmbusmeters/ there is already a driver for topas ES KR (topaseskr).
But it seems to be missing in your code.

Would it be, somehow, possible to add it?

Thanks in advance.

sanchosk, 2023-04-28:
Btw, might be a good idea to add something like ko-fi.com instead of the Polish site - I tried to pay using mBank, but I have different country, so it would not work. Wanted to buy you coffee, but I failed :( Perhaps github sponsoring?

SzczepanLeon, 2023-04-28:
> Wanted to buy you coffee, but I failed

There are two coffe pages.

sanchosk, 2023-04-28:
Thanks, I did not realize those are 2 logos, I thought it's one :)
Worked this time :+1:

SzczepanLeon, 2023-04-28:
topaseskr added

closed 2023-04-28 by SzczepanLeon

sanchosk, 2023-04-29:
That was fast, now to get home and test :)

sanchosk, 2023-04-29:
Not too much luck on my end :(
While compiling, I am getting this error:
```In file included from src/esphome.h:30,
                 from src/main.cpp:3:
src/esphome/components/wmbus/driver.h:11:34: error: 'map' is not a member of 'std'
   virtual esphome::optional<std::map<std::string, float>> get_values(std::vector<unsigned char> &telegram) = 0;
                                  ^~~
src/esphome/components/wmbus/driver.h:11:34: note: 'std::map' is defined in header '<map>'; did you forget to '#include <map>'?
src/esphome/components/wmbus/driver.h:1:1:
+#include <map>
 #pragma once
src/esphome/components/wmbus/driver.h:11:34:
   virtual esphome::optional<std::map<std::string, float>> get_values(std::vector<unsigned char> &telegram) = 0;```

SzczepanLeon, 2023-04-29:
Please share YAML

sanchosk, 2023-04-29:
```
substitutions:
  ## device settings
  device_name_short: "water-meter-esp"
  device_description: "Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon)"
  projectname: "OE9psj.wmbus-esp"
  appversion: "2.0.3"
  domain: !secret domain

  update_interval: 60s # Set to never to disable updates. Defaults to 60s.
  log_level: "VERBOSE" # use VERBOSE mode to find the meterId 0x43430778

  ## data service
  service_url: !secret service_host
  service_enabled: "false"
  service: "izar.watermeter"

  ## internal flag for optional sensors
  use_notall: "true"

esphome:
  name: ${device_name_short}
  comment: ${device_description}
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  platform: ESP32
  board: esp32dev

## ----------------------------------------------------------------
## External components
## ----------------------------------------------------------------
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
      priority: 0
  domain: !secret domain

captive_portal:

logger:
  id: appslogger
  level: ${log_level}
  baud_rate: 0 #disable logging over uart

ota:
  password: !secret ota_pswd
  on_error:
    then:
      - logger.log:
          level: INFO
          format: "OTA update error %d"
          args: ["x"]

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Berlin
    servers:
      - 10.10.20.254

    on_time_sync:
      then:
        - logger.log: "Synchronized sntp clock"

## ---------------------------------------------------
## WMBUS CC1101 --> ESP32
## https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/img/Esp32_CC1101.png
## ---------------------------------------------------
##
##                                        | 3.3V
##    - - - - - - - - - - - - - - - - - - x
##   |                                    |
## - | 5v           ESP32                 | -- ANT
##   |                                    |
##    - - - - x - x x x x x x - - - - - x x
##                | | | | |             | |
##                | | | | |             | | GND
##           GD00 | | | | |             |
##               GD02                  M0SI
##                   CSN
##                      SCK
##                        MISO
## -------------------------------------------------
wmbus:
  mosi_pin: GPIO23 ## SI:   braun
  miso_pin: GPIO19 ## SO:   grün
  clk_pin: GPIO18 ## SCLK: violett
  cs_pin: GPIO05 ## CSN:  orange
  gdo0_pin: GPIO16 ## GD00: gelb (rx)
  gdo2_pin: GPIO17 ## GD02: weiss (tx)

  ## -------------------------------------------------
  ## Using wmbusmeters i a pipe (optional)
  ## uncomment for using
  ## -------------------------------------------------
  # clients:
  #  - name: "wmbusmeters"
  #    ip_address: !secret wmbusmeter_host
  #    port: 7227
  #    format: rtlwMBus
  #    transport: UDP

sensor:
  - platform: wmbus
    meter_id: !secret watermeterId
    type: topaseskr
    lqi:
      name: "My lqi"
    rssi:
      name: "My RSSI"
    total_water_m3:
      name: "My cold water"
```

SzczepanLeon, 2023-04-29:
Try now (with clean).

sanchosk, 2023-04-29:
Now it compiles, I just have to find out how to log something :)
`ERROR No remote or local logging method configured (api/mqtt/logger)`

sanchosk, 2023-04-29:
So I get the logging now (forgot the api: part... - newbie...), but still getting the same error:
```
[00:48:54][I][wmbus:067]: Using driver 'topaseskr' for ID [0x69048692] RSSI: -30 dBm LQI: 128 Mode: T1 T: 4E44B40592860469F1077A3C0040059CD8604FD319E2899B85121A86C75B82FB8C545FFAF83B236B1E765DA3EB4FBE173AFDE44D479807C1B8FAF15029B0EB03C75E5F00D349CB1B2280625433A8DB (79)
[00:48:54][E][wmbus:115]: Can't get value from telegram for ID [0x69048692] 'topaseskr'
[00:48:54][E][wmbus:116]: T : 4E44B40592860469F1077A3C0040059CD8604FD319E2899B85121A86C75B82FB8C545FFAF83B236B1E765DA3EB4FBE173AFDE44D479807C1B8FAF15029B0EB03C75E5F00D349CB1B2280625433A8DB (79)
```

I've tried to put the ID in hexa and in the decimal form, no difference.

SzczepanLeon, 2023-04-29:
Payload is encrypted. Key is needed.

sanchosk, 2023-04-29:
:disappointed: Any idea how/where to get it? Is it proprietary and therefore bye-bye for me?

SzczepanLeon, 2023-04-29:
I don't know. For some types there is default key. Maybe meter owner can share that key with you?

sanchosk, 2023-04-29:
That's a local water company. I can try to ask them, but I doubt they will share it... But thanks a lot for the effort, for me it's well past the bed time today :)

zibous, 2023-04-30:
@sanchosk 

Try this with your watermeter telegram:
```
https://wmbusmeters.org/analyze/4E44B40592860469F1077A3C0040059CD8604FD319E2899B85121A86C75B82FB8C545FFAF83B236B1E765DA3EB4FBE173AFDE44D479807C1B8FAF15029B0EB03C75E5F00D349CB1B2280625433A8DB
```

Result w/o  Decryption key:

```
Auto driver  : topaseskr
Best driver  : unknown 00/00
Using driver : topaseskr 00/00
000   : 4e length (78 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b405 dll-mfct (AMT)
004   : 92860469 dll-id (69048692)
008   : f1 dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 3c tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
015 CE: 9CD8604FD319E2899B85121A86C75B82FB8C545FFAF83B236B1E765DA3EB4FBE173AFDE44D479807C1B8FAF15029B0EB03C75E5F00D349CB1B2280625433A8DB encrypted

{
    "media":"water",
    "meter":"topaseskr",
    "name":"",
    "id":"69048692",
    "timestamp":"1970-01-01T00:00:00Z"
}

Using: wmbusmeters: 1.12.0-28-g5e548ce
5e548ce3abadd7b84eca9dad3c4f87839e11c07
```

Telegram: **encrypted** (verschlüsselt)  ---> Key is needed.

SzczepanLeon, 2023-04-30:
Here https://github.com/wmbusmeters/wmbusmeters/issues/281 is nice discussion about keys. ;)

sanchosk, 2023-04-30:
Thanks, going to read now :) I'll try to call the company on Monday morning, explaining what I want, but I doubt I'll get to the key... Worth a try though.

zibous, 2023-04-30:
@sanchosk 

Try:  `51728910e66d83f851728910e66d83f8`

https://wmbusmeters.org/analyze/4e44B40592860469f1077a3c0040059cd8604fd319e2899B85121a86c75B82fB8c545ffaf83B236B1e765da3eB4fBe173afde44d479807c1B8faf15029B0eB03c75e5f00d349cB1B2280625433a8dB:auto:51728910e66d83f851728910e66d83f8

Result:
```
{
    "media":"water",
    "meter":"topaseskr",
    "name":"",
    "id":"69048692",
    "total_m3":27.863,
    "access_counter":140,
    "temperature_c":18.6,
    "current_flow_m3h":0,
    "volume_year_period_m3":0.2,
    "reverse_volume_year_period_m3":0,
    "meter_year_period_start_date":"2022-12-31",
    "volume_month_period_m3":13.296,
    "meter_month_period_start_datetime":"2023-03-31 23:59",
    "battery_y":12.911969,
    "timestamp":"2023-04-30T09:44:06Z"
}
```

sanchosk, 2023-04-30:
Man, you ROCK!
[11:49:09][D][sensor:110]: '0x69048692 My cold water': Sending state 27.94700 m³ with 3 decimals of accuracy

SzczepanLeon, 2023-04-30:
> Try: `51728910e66d83f851728910e66d83f8`

This is default key for that manufacture or somehow cracked?

zibous, 2023-04-30:
@SzczepanLeon 
Maybe the This is default key for that manufacture, found at:
https://github.com/wmbusmeters/wmbusmeters/issues/845