SzczepanLeon/esphome-components #20: Friendly_Name has WatermeterID

zibous, 2023-04-29:
Hi ,

Thanks, great job for the new version 2.0 - like this 💯 .

But the component add the watermeterid tot `total_water_m3` and i can not overwrite this.

[image]

Also on `lqi`and `rss`. I can use a workound (using template `copy` but this needs memory on the device.

```
## ---------------------------------------------------------------------------
## WMBUS METER az-delivery-devkit-v4 + CC1101 for Diehl IZAR RC 868 I R4 PL
##
## usage:  1. Using Homassistant API (discovery + devices)
##         2. Using HTTP Post to service server (optional)
##         3. Using wmbusmeters i a pipe (optional)
##         4. Reports free heap map on logging (optional)
##
## Project: https://github.com/zibous/ha-watermeter
## ---------------------------------------------------------------------------
substitutions:
  ## device settings
  device_name_short: "water-meter-esp"
  device_description: "Wasserzähler ESP32, CUL - CC1101, IZAR module (Diehl IZAR RC 868 I R4 PL (SzczepanLeon)"
  projectname: "OE9psj.wmbus-esp"
  domain: !secret domain

  update_interval: 60s # Set to never to disable updates. Defaults to 60s.
  log_level: "INFO" # use VERBOSE mode to find the meterId 0x43430778
  appversion: "1.1.7"

  ## data service
  service_url: !secret service_host
  service_enabled: "false"
  service: "izar.watermeter"

esphome:
  name: ${device_name_short}
  comment: ${device_description}
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}

  # ----------------------------------------------------------------
  # External components for wmbusmeter
  # ----------------------------------------------------------------
  on_boot:
    # At this priority, pretty much everything should already be initialized.
    priority: -100.0
    then:
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
  on_shutdown:
    priority: 700
    then:
      - logger.log:
          level: INFO
          tag: "BOOT"
          format: "BOOTMESSAGE:${device_name_short} is down!"

# ----------------------------------------------------------------
# az-delivery-devkit-v4
# ----------------------------------------------------------------
[... 813 lines cut ...]
  # curl -i http://water-meter-esp.local/text_sensor/watermeterdata
  - platform: template
    id: watermeterdata
    update_interval: ${update_interval}
    internal: true
    lambda: |-
      char buf[512];
      sprintf(buf, "%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%.3f|%s",
             id(waterdisplay).state,
             id(current_value),
             id(hour_value),
             id(daily_value),
             id(yesterday_value),
             id(week_value),
             id(month_value),
             id(lastmonth_value),
             id(year_value),
             id(systime).state.c_str()
             );
      std::string s = buf;
      return s;

# eof water-meter-izar.yaml

```

SzczepanLeon, 2023-04-29:
Hi

It is intended, but....
From version 2.0.3 You can use option **add_prefix**
```
sensor:
  - platform: wmbus
    meter_id: 0x78563412
    type: topaseskr
    use_prefix: False
    total_water_m3:
      name: "Water from topaseskr"
```

closed 2023-04-29 by SzczepanLeon

zibous, 2023-04-29:
thanks, perfect 👍