SzczepanLeon/esphome-components #21: Failed config

JarDol, 2023-04-29:
Prawdopodobnie po podniesieniu wersji esphome.
Nie mogę zwalidować a co za tym idzie zainstalować. Obecnie działa ale brak możliwości zmian.

sensor.wmbus: [source /config/esphome/wodomierz-master.yaml:59]
  platform: wmbus
  
  [id] is an invalid option for [sensor.wmbus]. Please check the indentation.
  id: wodomierz_master
  
  [name] is an invalid option for [sensor.wmbus]. Please check the indentation.
  name: Wodomierz - Master
  
  [filters] is an invalid option for [sensor.wmbus]. Please check the indentation.
  filters: 
    offset: 330.073

JarDol, 2023-04-29:
``esphome:
  name: esp-wodomierz-master
  comment: Licznik wody - apator-at-wmbus-16-2

esp8266:
  board: nodemcuv2

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
#  manual_ip:
#    static_ip: ${wifi_ip}
#    gateway: !secret wifi_gateway
#    subnet: !secret wifi_subnet
  ap:
    ssid: ${upper_devicename} Hotspot
    password: !secret wifi_ap_pass
#    use_address: ${wifi_ota_ip}

captive_portal:
debug:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pass

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

time:
  platform: sntp
  
# int ApatorID = 0x4829838;

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

sensor:
  - platform: wmbus
    id: wodomierz_master
    name: "Wodomierz - Master"
    filters:
      offset: +330.07300
    meter_id: 0x173629 
    type: apator162
    key: "00000000000000000000000000000000"

JarDol, 2023-04-29:
`
esphome:
  name: esp-wodomierz-master
  comment: Licznik wody - apator-at-wmbus-16-2

esp8266:
  board: nodemcuv2

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  ap:
    ssid: ${upper_devicename} Hotspot
    password: !secret wifi_ap_pass

captive_portal:
debug:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pass

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

time:
  platform: sntp
 
wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

sensor:
  - platform: wmbus
    id: wodomierz_master
    name: "Wodomierz - Master"
    filters:
      offset: +330.07300
    meter_id: 0x173629 
    type: apator162
    key: "00000000000000000000000000000000"
`

zibous, 2023-04-29:
hi

see readme: 

```
# this is not for version 2.0, you have to use the component Version 1.5.1
sensor:

    platform: wmbus
    id: wodomierz_master
    name: "Wodomierz - Master"
    filters:
    offset: +330.07300
    meter_id: 0x173629
    type: apator162
    key: "00000000000000000000000000000000"
```

https://github.com/SzczepanLeon/esphome-components#211-example-for-version-2x

```
## Version 2.0
external_components:
    source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]
```

```
## Version 1.5.1
external_components:
  - source: github://SzczepanLeon/esphome-components@1.5.1
    components: [ wmbus ]
```

SzczepanLeon, 2023-04-29:
So for version 1.x config is:
```
external_components:
    source: github://SzczepanLeon/esphome-components@1.5.1
    components: [ wmbus ]

sensor:
  - platform: wmbus
    meter_id: 0x12353648
    type: apator162
    key: "00000000000000000000000000000000"
    name: "Wodomierz - Master"
```

and for version 2.x
```
external_components:
    source: github://SzczepanLeon/esphome-components@master
    components: [ wmbus ]

sensor:
  - platform: wmbus
    meter_id: 0x12353648
    type: apator162
    key: "00000000000000000000000000000000"
    total_water_m3:
      name: "Wodomierz - Master"
```

In v2.x You can also add more sensor to one meter. For example RSSI:
```
sensor:
  - platform: wmbus
    meter_id: 0x12353648
    type: apator162
    key: "00000000000000000000000000000000"
    total_water_m3:
      name: "Wodomierz - Master"
    rssi:
      name: "My RSSI"
```

closed 2023-04-29 by SzczepanLeon

JarDol, 2023-04-29:
Po zmianie:
`sensor:
  - platform: wmbus
    total_water_m3:
      id: wodomierz_master
      name: "Wodomierz - Master"
      filters:
        offset: +330.07300
    meter_id: 0x173629 
    type: apator162
    key: "00000000000000000000000000000000"
    rssi:
      name: "Wodomierz - Master RSSI"`

Błędy przy kompilacji:

`In file included from /data/esp-wodomierz-master/.piolibdeps/esp-wodomierz-master/wMbus-lib/rf_mbus.cpp:3:
/data/esp-wodomierz-master/.piolibdeps/esp-wodomierz-master/wMbus-lib/rf_mbus.hpp:71:12: error: 'string' in namespace 'std' does not name a type
   71 | const std::string mode_to_string(WmBusMode mode);
      |            ^~~~~~
/data/esp-wodomierz-master/.piolibdeps/esp-wodomierz-master/wMbus-lib/rf_mbus.hpp:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /***********************************************************************************
/data/esp-wodomierz-master/.piolibdeps/esp-wodomierz-master/wMbus-lib/rf_mbus.cpp:10:12: error: 'string' in namespace 'std' does not name a type
   10 | const std::string mode_to_string(WmBusMode mode) {
      |            ^~~~~~
/data/esp-wodomierz-master/.piolibdeps/esp-wodomierz-master/wMbus-lib/rf_mbus.cpp:7:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
    6 | #include "tmode_rf_settings.hpp"
  +++ |+#include <string>
    7 | 
Compiling /data/esp-wodomierz-master/.pioenvs/esp-wodomierz-master/FrameworkArduino/FSnoop.cpp.o
*** [/data/esp-wodomierz-master/.pioenvs/esp-wodomierz-master/lib4eb/wMbus-lib/rf_mbus.cpp.o] Error 1`

SzczepanLeon, 2023-04-29:
Zmianie na co? Pełny config z odpowiednim formatowaniem poprosze.

JarDol, 2023-04-29:
`
esphome:
  name: esp-wodomierz-master
  comment: Licznik wody - apator-at-wmbus-16-2

esp8266:
  board: d1_mini

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  ap:
    ssid: ${upper_devicename} Hotspot
    password: !secret wifi_ap_pass

captive_portal:
debug:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pass

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

time:
  platform: sntp
  
wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

sensor:
  - platform: wmbus
    total_water_m3:
      id: wodomierz_master
      name: "Wodomierz - Master"
      filters:
        offset: +330.07300
    meter_id: 0x173629 
    type: apator162
    key: "00000000000000000000000000000000"
    rssi:
      name: "Wodomierz - Master RSSI"

`

SzczepanLeon, 2023-04-29:
Przeczytaj dokladnie moje poprzednie posty i popraw swój konfig.

JarDol, 2023-04-29:
Usunąłem folder z twoim kodem z: config\esphome\.esphome\external_components
Zassał na nowo i ruszyła kompilacja i zainstalowało się.