SzczepanLeon/esphome-components #22: LQI Confused - which is correct?

zibous, 2023-04-30:
Hi,

I try to understand LQI, but I get different statements about the interpretation of the values:

**electrodragon**
`a low value indicates a better link than what a high value does`

> LQI (Link Quality Indicator) is a metric of the current quality of the received signal. 
The LQI gives an estimate of how easily a received signal can be demodulated by accumulating 
the magnitude of the error between ideal constellations and the received signal over 
the 64 symbols immediately following the sync word. LQI is best used as a relative 
measurement of the link quality (a low value indicates a better link than what a high value does), 
since the value is dependent on the modulation format.
https://www.electrodragon.com/cc1101-rssi-and-lqi-test/

**wikipedia**
`0 stands for a very bad connection and 255 for a very good connection.`
> The Link Quality Indicator (LQI) is a measure of the strength or quality of a received 
(radio) packet for Wireless Personal Area Networks (WPAN).
In a radio connection, you want to know how likely it is that a receiver understands a message sent by a sender. 
This is particularly important when different transmission paths exist in order to select the 
path that most reliably transmits the message
LQI is defined in IEEE 802.15.4 Chapter 8.2.6[1] and is usually given as a number between 0 and 255.[2] 
0 stands for a very bad connection and 255 for a very good connection. The LQI should be determined for each 
individual package. It depends on the manufacturer how the LQI is determined. Some radio IC manufacturers 
simply use a scaled RSSI, others measure the signal-to-noise ratio.
https://de.wikipedia.org/wiki/Link_Quality_Indicator

Which one is correct for the wmbus meter component?

Thanks

SzczepanLeon, 2023-04-30:
LQI is directly copied from CC1101. See description in spec (page 48) https://www.ti.com/lit/gpn/cc1101

What interpretation is correct I don't know ;)

zibous, 2023-04-30:
Thanks,

> 17.6 Link Quality Indicator (LQI)
The Link Quality Indicator is a metric of the
current quality of the received signal. If
PKTCTRL1.APPEND_STATUS is enabled, the
value is automatically added to the last byte
appended after the payload. The value can
also be read from the LQI status register. The
LQI gives an estimate of how easily a received
signal can be demodulated by accumulating
the magnitude of the error between ideal
constellations and the received signal over the
64 symbols immediately following the sync
word. 

>LQI is best used as a relative
measurement of the link quality (a low value
indicates a better link than what a high value
does), since the value is dependent on the
modulation format.

closed 2023-04-30 by zibous