SzczepanLeon/esphome-components #23: Problem compiling due to the changes in version >2.0 - No clue what to change

cococheaf, 2023-05-02:
Good day,

i have problems compiling the sources after the esphome 2023.4.3 update.
I have read the documentation, you are mentioning a change in version > 2.0 regarding some yaml changes.
Problem is i have no cue and too less knowledge to correct the error.

Here's the esphome yaml configuration: 

`# ----------------------------------------------------------------
# Substitutions
# ----------------------------------------------------------------
substitutions:
  device_name_short: "wasserzaehler"
  update_interval: 10s
  log_level: "DEBUG"
  appversion: "1.1.1"
  device_description: "'Wemos D1 Mini' mit CC1101 und IZAR Modul (Diehl IZAR RC 868 I R4 PL)"

# ----------------------------------------------------------------
# ESP-Configuration
# ----------------------------------------------------------------
esphome:
  name: "wasserzaehler"
  friendly_name: Wasserzähler
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true
  arduino_version: recommended
  comment: ${device_description}
  project:
    name: "amgnet.wasserzaehler"
    version: ${appversion}
  build_path: ./build/${device_name_short}

# ----------------------------------------------------------------
# External components for wmbusmeter
# ----------------------------------------------------------------
  on_boot:
    priority: -500
    then:
      - logger.log:
          level: info
          format: "BOOTMESSAGE:${device_name_short} API is connected, Device ready!"
      - globals.set:
          id: boot_counter
          value: !lambda "return id(boot_counter)+=1;"

  on_shutdown:
    priority: 700
    then:
      - logger.log: ${device_name_short} is down!

# ----------------------------------------------------------------
# External components for wmbusmeter
# ----------------------------------------------------------------
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [wmbus]

# ----------------------------------------------------------------
# Preferences
# ----------------------------------------------------------------
preferences:
  flash_write_interval: 60min

# ----------------------------------------------------------------
# Global variables
# ----------------------------------------------------------------
globals:
  - id: boot_counter
[... 467 lines cut ...]
  accuracy_decimals: 3
  
  [on_value] is an invalid option for [sensor.wmbus]. Please check the indentation.
  on_value: 
    then: 
      - lambda: |-
          if (id(last_value) > 0.00 && (id(last_value)!=id(waterdisplay).state)) {
            ESP_LOGI("main", " -------  SET NEW VALUES !!!!!");
            id(current_value)=id(waterdisplay).state-id(last_value);
            id(watercurrent).publish_state(id(current_value));
            id(hour_value)+=id(waterdisplay).state-id(last_value);
            id(waterhour).publish_state(id(hour_value));
            id(daily_value)+=id(waterdisplay).state-id(last_value);
            id(waterday).publish_state(id(daily_value));
            id(week_value)+=id(waterdisplay).state-id(last_value);
            id(waterweek).publish_state(id(week_value));
            id(month_value)+=id(waterdisplay).state-id(last_value);
            id(watermonth).publish_state(id(month_value));
            id(year_value)+=id(waterdisplay).state-id(last_value);
            id(wateryear).publish_state(id(year_value));
          }
          id(last_value)=id(waterdisplay).state;`

Any help would be highly appreciated.
Thank you in advance.

SzczepanLeon, 2023-05-02:
Try this:
`
sensor:
  - platform: wmbus
    meter_id: 0x17740778
    type: izar
    add_prefix: False
    total_water_m3:
      name: "Wasseruhr Anzeige"
      id: "waterdisplay"
      on_value:
        then:
          - lambda: |-
              if (id(last_value) > 0.00 && (id(last_value)!=id(waterdisplay).state)) {
                ESP_LOGI("main", " -------  SET NEW VALUES !!!!!");
                id(current_value)=id(waterdisplay).state-id(last_value);
                id(watercurrent).publish_state(id(current_value));
                id(hour_value)+=id(waterdisplay).state-id(last_value);
                id(waterhour).publish_state(id(hour_value));
                id(daily_value)+=id(waterdisplay).state-id(last_value);
                id(waterday).publish_state(id(daily_value));
                id(week_value)+=id(waterdisplay).state-id(last_value);
                id(waterweek).publish_state(id(week_value));
                id(month_value)+=id(waterdisplay).state-id(last_value);
                id(watermonth).publish_state(id(month_value));
                id(year_value)+=id(waterdisplay).state-id(last_value);
                id(wateryear).publish_state(id(year_value));
              }
              id(last_value)=id(waterdisplay).state;
`

zibous, 2023-05-02:
@cococheaf 

For the new version see more changes here:
https://github.com/zibous/ha-watermeter/tree/master/esphome

cococheaf, 2023-05-02:
You nailed it! That was it.
Thank you very much. Can be closed!

closed 2023-05-02 by cococheaf