SzczepanLeon/esphome-components #24: Version 1.5.1 vs. Version 2.0.x - WatermeterID "0x00"

zibous, 2023-05-02:
Hi,

Something must have changed, because with the new version it is not possible to find the WatermeterID of my water meter with a WatermeterID 0x00. With version 1.5.1 it works, with the latest version I don't get any telegrams.

Am I doing something wrong with version 2.0.x or is this a bug?

**Version 1.5.1**

```yaml
 wmid: "0" 
 
external_components:
  - source: github://SzczepanLeon/esphome-components@1.5.1
    components: [ wmbus ]

sensor:
  - platform: wmbus
    meter_id: ${wmid}
    type: izar
    name: "Wasseruhr Anzeige"
    id: "waterdisplay"
    accuracy_decimals: 3
    on_value:
      then:
        - lambda: |-
            ESP_LOGI("SENSOR", "Water Display value: %f", id(waterdisplay).state);
 
RESULT:

16:32:27	[D]	[wmbus:158]	
Meter ID [0x43410778] RSSI: -67 dBm LQI: 128 not found in configuration T: 1944A511780741434418A211150013CC9E976D2F0A82C83AAB91 (26)
16:32:28	[D]	[wmbus:158]	
Meter ID [0x43430778] RSSI: -51 dBm LQI: 128 not found in configuration T: 1944A511780743434418A231150013CE5566CF1D9F1944EFB9C2 (26)
16:32:35	[D]	[wmbus:158]	
Meter ID [0x43410778] RSSI: -64 dBm LQI: 128 not found in configuration T: 1944A511780741434418A221150013CCA7E6B7D268F6A2578096 (26)
16:32:44	[D]	[wmbus:158]	
Meter ID [0x43410778] RSSI: -65 dBm LQI: 129 not found in configuration T: 1944A511780741434418A231150013CCB0C9FE7949257B8C9994 (26)
16:32:45	[D]	[wmbus:158]	
Meter ID [0x43430778] RSSI: -51 dBm LQI: 128 not found in configuration T: 1944A511780743434418A251150013CE27857AE75BF19035EFCD (26)
```

**Version 2.0**

```yaml
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

sensor:
  - platform: wmbus
    meter_id: ${wmid}
    type: izar
    add_prefix: true
    lqi:
      id: wmbus_cc1101_lqi
      name: "Wlan CC1101 LQI"
      entity_category: "diagnostic"
    rssi:
      name: "Wlan CC1101 RSSI"
      id: wmbus_cc1101_rssi
    total_water_m3:
      name: "Wasseruhr Anzeige"
      id: "waterdisplay"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      on_value:
        then:
          - lambda: |-
              ESP_LOGI("SENSOR", "Water Display value: %f", id(waterdisplay).state);

RESULT:
[16:40:31][C][wmbus:284]: wM-Bus v2.1.2:
[16:40:31][C][wmbus:301]:   CC1101 SPI bus:
[16:40:31][C][wmbus:302]:     MOSI Pin: GPIO13
[16:40:31][C][wmbus:303]:     MISO Pin: GPIO12
[16:40:31][C][wmbus:304]:     CLK Pin:  GPIO14
[16:40:31][C][wmbus:305]:     CS Pin:   GPIO15
[16:40:31][C][wmbus:306]:     GDO0 Pin: GPIO5
[16:40:31][C][wmbus:307]:     GDO2 Pin: GPIO4
[16:40:31][C][wmbus:314]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, compact5, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451
[16:40:31][C][wmbus:332]:   Meter:
[16:40:31][C][wmbus:333]:     ID: 0 [0x00000000]
[16:40:31][C][wmbus:334]:     Type: izar
[16:40:31][C][wmbus:335]:     Key: ''
[16:40:31][C][wmbus:337]:     Sensor '0x00 Wlan CC1101 LQI'
[16:40:31][C][wmbus:337]:       Device Class: 'signal_strength'
[16:40:31][C][wmbus:337]:       State Class: 'measurement'
[16:40:31][C][wmbus:337]:       Unit of Measurement: ''
---------------------------
No wmbus:158 LOGS

``` 

**RSSI and LQI**
Would it also be possible to display RSSI and LQI if no telegram is available? Would help to check if the CC1101 is working.

Thanks

SzczepanLeon, 2023-05-02:
Please add as **wmbus** component option:
  log_unknown (Optional): Show telegrams from not configured meters in log. Defaults to False.

zibous, 2023-05-02:
@SzczepanLeon,

Negativ, wmbus is allready present.
The examples are only abbreviated.

SzczepanLeon, 2023-05-02:
```
wmbus:
  log_unknown: True
```

closed 2023-05-02 by zibous

zibous, 2023-05-02:
> What You mean by "no telegram is available"? Do You want to print RSSI every X ms?

No when i use:

```
wmbus:
  log_unknown: True
and watermeterID: 0x00
```
[image]

zibous, 2023-05-02:
I would replace 0x00 with the WatermeterID evaluated from the telegram and set the associated values. Mariusz Woszczynski's variant does it this way. 

see:  MariuszWoszczynski /
[ESPhome-IZAR-meter-reader](https://github.com/MariuszWoszczynski/ESPhome-IZAR-meter-reader)

> So if meter with ID == 0 is defined then RSSI/LQI from all received should be transferred to HA.

No problem, they could be processed via a filter at Homeassistant. But it would be easier for finding the water meter. This would only happen if wmbus: log_unknown: True is set.

SzczepanLeon, 2023-05-02:
To be honest I don't know what You want to achieve.

With **log_unknown=true** telegrams for all undefined meters will be printed (debug log). From that log you can read ID, RSSI, LQI.

You don't need to define sensor with 0 ID to have that log.

SzczepanLeon, 2023-05-04:
Try version 2.1.3.
RSSI and LQI should be updated for meter with ID == 0
Also **last_month_total_water_m3** for izar was added.

SzczepanLeon, 2023-05-04:
What device_class, etc do You need for those sensors?

zibous, 2023-05-04:
> What device_class, etc do You need for those sensors?

Telegram: 
```
11:46:57 | [I] | [wmbus:063] | Using driver 'izar' for ID [0x43430778] RSSI: -50 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A221150013CE2F5786B6BECA9D34A0C0 (26)
```
MBUSMETERS Payload:
```
{
  "media":"water",
  "meter":"izar",
  "name":"watermeter",
  "id":"18444343",
  "total_m3":471.325,
  "last_month_total_m3":470.724,
  "last_month_measure_date":"2023-05-01",
  "remaining_battery_life_y":10.5,
  "current_alarms":"no_alarm",
  "previous_alarms":"no_alarm",
  "transmit_period_s":8,
  "current_month_total_m3":0.601,
  "timestamp":"2023-05-04T09:41:07Z",
  "device":"cul",
  "rssi_dbm":-53,
  "address":"Gemeinde",
  "city":"Hoechst",
  "service":"wmbusmeters nuc2023"
 }
 ```
device_class:

> remaining_battery_life_y: None: Generic sensor.
>previous_alarms: None: Generic sensor.
>previous_alarms: None: Generic sensor.
> last_month_measure_date: timestamp: Datetime object or timestamp string (ISO 8601)

SzczepanLeon, 2023-05-04:
Fields **remaining_battery_life_y** and **transmit_period_s** added in 2.1.4

For alarms & date I need more time.

zibous, 2023-05-04:
> Fields **remaining_battery_life_y** and **transmit_period_s** added in 2.1.4

Confirm, is working

> For alarms & date I need more time.

Alarms
```
        alarms.general_alarm = frame[11] >> 7;
        alarms.leakage_currently = frame[12] >> 7;
        alarms.leakage_previously = frame[12] >> 6 & 0x1;
        alarms.meter_blocked = frame[12] >> 5 & 0x1;
        alarms.back_flow = frame[13] >> 7;
        alarms.underflow = frame[13] >> 6 & 0x1;
        alarms.overflow = frame[13] >> 5 & 0x1;
        alarms.submarine = frame[13] >> 4 & 0x1;
        alarms.sensor_fraud_currently = frame[13] >> 3 & 0x1;
        alarms.sensor_fraud_previously = frame[13] >> 2 & 0x1;
        alarms.mechanical_fraud_currently = frame[13] >> 1 & 0x1;
        alarms.mechanical_fraud_previously = frame[13] & 0x1;

```
If it's easier, it doesn't have to be text, code would certainly be easier. This can then be translated in the ESPHome application:

Alarmcode example:

       GA =  alarms.general_alarm 
       CL = alarms.leakage_currently = frame[12] >> 7;
       PL = alarms.leakage_previously = frame[12] >> 6 & 0x1;
       MB = alarms.meter_blocked = frame[12] >> 5 & 0x1;
       BF = alarms.back_flow = frame[13] >> 7;
       UF = alarms.underflow = frame[13] >> 6 & 0x1;
       OF = alarms.overflow = frame[13] >> 5 & 0x1;
       AS = alarms.submarine = frame[13] >> 4 & 0x1;
       AF = alarms.sensor_fraud_currently = frame[13] >> 3 & 0x1;
       AP= alarms.sensor_fraud_previously = frame[13] >> 2 & 0x1;
       AMC = alarms.mechanical_fraud_currently = frame[13] >> 1 & 0x1;
       AMP alarms.mechanical_fraud_previously = frame[13] & 0x1;

SzczepanLeon, 2023-05-04:
We have 12 alarms so I can implement it as binary bitmap (this will be quick).

zibous, 2023-05-05:
hi,

Is this possible `get_current_month_total_water_m3` ?

```
esphome::optional<float> get_current_month_total_water_m3(std::vector<unsigned char> &telegram) {
      esphome::optional<float> ret_val{};
      uint8_t *decoded = reinterpret_cast<uint8_t*>(telegram.data());
      uint8_t decoded_len = telegram.size();
      uint8_t decrypted[64] = {0};
     if ((this->decrypt(decoded, decoded_len, decrypted)) > 0 ) {
       ret_val = (this->uintFromBytesLittleEndian(decrypted + 1)) * 0.001 - (this->uintFromBytesLittleEndian(decrypted + 5)) * 0.001;
     } 
  };
```

I can also do this with ESPHOME, but there I have to catch the error when booting, because at this point in time only `get_last_month_total_water_m3` but NOT `get_total_water_m3` is available. 

Don't understand why this is so. 

Therefore the implementation in your componente would be better.

i see that `get_*_alarms` is ready, perfekt work. 
Thank you.

SzczepanLeon, 2023-05-05:
> I can also do this with ESPHOME, but there I have to catch the error when booting, because at this point in time only `get_last_month_total_water_m3` but NOT `get_total_water_m3` is available.
> 
> Don't understand why this is so.
> 
> Therefore the implementation in your componente would be better.

It is for all telegrams? Do you have logs?
It is very strange.

zibous, 2023-05-05:
> Is it for all telegrams? Do you have logs?
> It is very strange.

```log
INFO Reading configuration /config/wm-esp32.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
INFO Generating C++ source...
INFO Backup config will take: 4483 bytes
INFO Compiling app...
Processing water-meter-esp (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Syslog @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- wMbus-lib @ 1.1.1+sha.da2558f
|   |-- SPI @ 2.0.0
|   |-- SmartRC-CC1101-Driver-Lib @ 2.5.7
Compiling .pioenvs/water-meter-esp/src/main.cpp.o
Linking .pioenvs/water-meter-esp/firmware.elf
RAM:   [=         ]   8.2% (used 43900 bytes from 532480 bytes)
Flash: [======    ]  60.8% (used 1115441 bytes from 1835008 bytes)
Building .pioenvs/water-meter-esp/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/water-meter-esp/firmware.bin"], [".pioenvs/water-meter-esp/firmware.elf"])
Wrote 0x121bc0 bytes to file /config/build/water-meter-esp/.pioenvs/water-meter-esp/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 41.96 seconds =========================
INFO Successfully compiled program.
INFO Uploading /config/./build/water-meter-esp/.pioenvs/water-meter-esp/firmware.bin (1121216 bytes)
Uploading: [============================================================] 100% Done...
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from water-meter-esp.local using esphome API
INFO Successfully connected to water-meter-esp.local
[10:16:12][I][app:102]: ESPHome version 2023.5.0-dev compiled on May  5 2023, 10:15:18
[10:16:12][I][app:104]: Project Diehl_IZAR_RC_868.Watermeter version 2.0.6

[10:16:15][I][wmbus:070]: Using driver 'izar' for ID [0x43430778] RSSI: -65 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A221150013CE325486B6BECA9D34A0C0 (26)

[10:16:24][I][SENSOR:706]: Current Month value:nan Water Display value: nan, last value: 470.724 m³
[10:16:32][I][SENSOR:706]: Current Month value:nan Water Display value: nan, last value: 470.724 m³

[10:16:41][I][wmbus:070]: Using driver 'izar' for ID [0x43430778] RSSI: -65 dBm LQI: 130 Mode: T1 T: 1944A511780743434418A251150013CE57987AE75BF19035EFCD (26)
[10:16:41][I][SENSOR:706]: Current Month value:nan Water Display value: nan, last value: 470.724 m³

[10:16:49][I][wmbus:070]: Using driver 'izar' for ID [0x43430778] RSSI: -66 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A261150013CE6EE9A01A3985FA58C4CA (26)
[10:16:49][I][SENSOR:706]: Current Month value:829.987 L Water Display value: 471.554 m³, last value: 470.724 m³

[10:16:58][I][wmbus:070]: Using driver 'izar' for ID [0x43430778] RSSI: -66 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A271150013CE79C6E9B118562383DDC8 (26)
[10:16:58][I][SENSOR:706]: Current Month value:829.987 L Water Display value: 471.554 m³, last value: 470.724 m³

```

SzczepanLeon, 2023-05-05:
Please add VERBOSE logs, to see more stuff related how ESPHome is handling sensor states.

Are You sure about this? Seems that **)** is missing after **month_value**
> id(month_value = (id(waterdisplay).state - id(waterdisplay_lastmonth).state) * 1000

zibous, 2023-05-05:
> Please add VERBOSE logs, to see more stuff related how ESPHome is handling sensor states.

Sorry,

```
last_month_total_water_m3:
      name: "Wasseruhr Anzeige letzter Monat"
      id: "waterdisplay_lastmonth"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
      on_value:
        then:
          - lambda: |-              
              id(month_value) = ( float(id(waterdisplay).state) - float(id(waterdisplay_lastmonth).state) ) * 1000.00;
              id(waterlastmonth).publish_state(id(month_value));
              ESP_LOGI("SENSOR", "Current Month value:%.3f L Water Display value: %.3f m³, last value: %.3f L", id(month_value), id(waterdisplay).state, id(waterdisplay_lastmonth).state);
```
but same log results.

SzczepanLeon, 2023-05-05:
I would guess that sensor **waterdisplay_lastmonth** is updated ass first and **waterdisplay** as second.

So when **waterdisplay_lastmonth** is updated and **on_value** code is executed there is no value for **waterdisplay**.

zibous, 2023-05-05:
> I would guess that sensor **waterdisplay_lastmonth** is updated first and **waterdisplay** second.

Yes, Hence my consideration to move this to your component, that would make it more correct.

```
esphome::optional<float> get_current_month_water_m3(std::vector<unsigned char> &telegram) {
      esphome::optional<float> ret_val{};
      uint8_t *decoded = reinterpret_cast<uint8_t*>(telegram.data());
      uint8_t decoded_len = telegram.size();
      uint8_t decrypted[64] = {0};
     if ((this->decrypt(decoded, decoded_len, decrypted)) > 0 ) {
       ret_val = (this->uintFromBytesLittleEndian(decrypted + 1)) * 0.001 - (this->uintFromBytesLittleEndian(decrypted + 5)) * 0.001;
     } 
  };
```

SzczepanLeon, 2023-05-05:
I agree with you. Please try 2.1.10 -- added **current_month_total_water_l**

zibous, 2023-05-05:
> I agree with you. Please try 2.1.10 -- added **current_month_total_water_l**

Perfect: optimized: *your code 5 lines* -  *esphome codelines: 20* 💯 
Confirm: is working:

[image]

zibous, 2023-05-07:
> For alarms & date I need more time.

`last_month_measure_date` is missing, but i think this is not so easy, 
 because a text sensor is needed for that (which doesn't exist). 

I've tried looking at wmbusmeters a bit, but I can't get the `text sensor` to work. 
Unfortunately, it is not possible to transmit a date as a sensor.

```
  // last_month_measure_date
  // see https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_izar.cc
  
  char ret_val[11]; // 0000-00-00
  uint16_t h0_year = ((telegram[10] & 0xF0) >> 1) + ((telegram[9] & 0xE0) >> 5);
  h0_year=(h0_year > 80)?h0_year += 1900:h0_year += 2000;  uint8_t h0_month=5;
  uint8_t h0_day = 17; // = telegram[9] & 0x1F;
  sprintf (ret_val, "%04u-%02u-%02u", h0_year, h0_month, h0_day);
  return ret_val;
  
```

Is there any change that is still available?

Thanks.