SzczepanLeon/esphome-components #26: ID in front of Enity name

MiG-41, 2023-05-07:
This new way , that in front of entity name there is automaticly added ID is something i don't need and like.
In configuration i have :
    ```
total_water_m3:
      name: "Wodomierz" 
```

And entity is named "0x123456 Wodomierz" .... 
If i need that i could add id my self in configuration ( either in strict way or by substitutions section...)

And finally ( if author find it right way ) how to get rid of this id :)

baterja, 2023-05-07:
I had the same impression at first :) I think the author decided to name entities like that because you can have multiple sensors with the same entity name and that could be confusing.

You can just rename the entity in its options as I did. Probably you won't be adding new water meters every day so renaming entities just once after the initial setup will fix it forever :)

SzczepanLeon, 2023-05-07:
Please check this:
https://github.com/SzczepanLeon/esphome-components/issues/20#issuecomment-1528708126

closed 2023-05-07 by SzczepanLeon

MiG-41, 2023-05-07:
Ok , thank you , havn't notice this....

So maybe consider at least     add_prefix: False as default instead oposite...

SzczepanLeon, 2023-05-07:
Cons and pros for True and False are equal for me.