SzczepanLeon/esphome-components #29: Wmbus component v2.x doesn't list telegrams in log

MarcinJanCiesielski, 2023-05-09:
Hi.
After update to version 2.X there is no sign of devices' telegrams.

Below is log from version 1.5.1 of wmbus
----------------------------------------------------------------------------------------------
[21:45:48][I][app:102]: ESPHome version 2023.4.4 compiled on May  9 2023, 16:25:16
[21:45:48][C][wifi:505]: WiFi:
[21:45:48][C][wifi:363]:   Local MAC: A0:20:A6:09:B4:77
[21:45:48][C][wifi:364]:   SSID: [redacted]
[21:45:48][C][wifi:365]:   IP Address: 192.168.1.160
[21:45:48][C][wifi:366]:   BSSID: [redacted]
[21:45:48][C][wifi:368]:   Hostname: '0_wmbus'
[21:45:48][C][wifi:370]:   Signal strength: -41 dB ▂▄▆█
[21:45:48][C][wifi:374]:   Channel: 8
[21:45:48][C][wifi:375]:   Subnet: 255.255.255.0
[21:45:48][C][wifi:376]:   Gateway: 192.168.1.1
[21:45:48][C][wifi:377]:   DNS1: 192.168.1.1
[21:45:48][C][wifi:378]:   DNS2: 0.0.0.0
[21:45:48][C][logger:294]: Logger:
[21:45:48][C][logger:295]:   Level: DEBUG
[21:45:48][C][logger:296]:   Log Baud Rate: 115200
[21:45:48][C][logger:297]:   Hardware UART: UART0
[21:45:48][C][uptime.sensor:031]: Uptime Sensor '0_wmbus_uptime'
[21:45:48][C][uptime.sensor:031]:   Device Class: 'duration'
[21:45:48][C][uptime.sensor:031]:   State Class: 'total_increasing'
[21:45:48][C][uptime.sensor:031]:   Unit of Measurement: 's'
[21:45:48][C][uptime.sensor:031]:   Accuracy Decimals: 0
[21:45:48][C][uptime.sensor:031]:   Icon: 'mdi:clock-outline'
[21:45:48][C][template.text_sensor:021]: Template Sensor '0_wmbus_id'
[21:45:48][C][template.text_sensor:021]:   Icon: 'mdi:chip'
[21:45:48][C][homeassistant.time:010]: Home Assistant Time:
[21:45:48][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[21:45:48][C][restart.button:017]: Restart Button '0_wmbus_restart'
[21:45:48][C][safe_mode.button:022]: Safe Mode Button '0_wmbus_safemode'
[21:45:48][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[21:45:48][C][wmbus_sensor:016]: wM-Bus Sensor '0_wmbus_water'
[21:45:48][C][wmbus_sensor:016]:   Device Class: 'water'
[21:45:48][C][wmbus_sensor:016]:   State Class: 'total'
[21:45:48][C][wmbus_sensor:016]:   Unit of Measurement: 'l'
[21:45:48][C][wmbus_sensor:016]:   Accuracy Decimals: 0
[21:45:48][C][wmbus_sensor:016]:   Icon: 'mdi:water-sync'
[21:45:48][C][wmbus_sensor:022]:   Key: '00000000000000000000000000000000'
[21:45:48][C][wmbus_sensor:023]:   Type: mkradio4
[21:45:48][C][wmbus_sensor:024]:   ID: 2492994353 [0x94981331]
[21:45:48][C][version.text_sensor:021]: Version Text Sensor '0_wmbus_version'
[21:45:48][C][version.text_sensor:021]:   Icon: 'mdi:code-json'
[21:45:48][C][wifi_info:013]: WifiInfo Mac Address '0_wmbus_mac'
[21:45:48][C][wifi_info:013]:   Icon: 'mdi:lan'
[21:45:48][C][captive_portal:088]: Captive Portal:
[21:45:48][C][web_server:151]: Web Server:
[21:45:48][C][web_server:152]:   Address: 0_wmbus.local:80
[21:45:48][C][mdns:108]: mDNS:
[21:45:48][C][mdns:109]:   Hostname: 0_wmbus
[21:45:48][C][ota:093]: Over-The-Air Updates:
[21:45:48][C][ota:094]:   Address: 0_wmbus.local:8266
[21:45:48][C][ota:097]:   Using Password.
[21:45:48][C][api:138]: API Server:
[21:45:48][C][api:139]:   Address: 0_wmbus.local:6053
[21:45:48][C][api:141]:   Using noise encryption: YES
[21:45:48][C][wifi_signal.sensor:009]: WiFi Signal '0_wmbus_linkquality'
[21:45:48][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:45:48][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:45:48][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:45:48][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:45:48][C][debug:048]: Debug component:
[21:45:48][D][debug:060]: ESPHome version 2023.4.4
[21:45:48][D][debug:064]: Free Heap Size: 15544 bytes
[21:45:48][D][debug:092]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[21:45:48][D][debug:248]: Chip ID: 0x0009B477
[21:45:48][D][debug:249]: SDK Version: 2.2.2-dev(38a443e)
[... 108 lines cut ...]
  #     name: ${hostname}_rssi
  #   total_water_m3:
  #     name: ${hostname}_water
  #     accuracy_decimals: 0
  #     force_update: true
  #     icon: "mdi:water-sync"
  #     unit_of_measurement: "l"
  #     state_class: "total"
  #     device_class: "water"
  #     filters:
  #       - multiply: 1000
# v1
  - platform: wmbus
    name: ${hostname}_water
    meter_id: "0x94981331"
    type: "mkradio4"
    key: "00000000000000000000000000000000"
    accuracy_decimals: 0
    force_update: true
    icon: "mdi:water-sync"
    unit_of_measurement: "l"
    state_class: "total"
    device_class: "water"
    filters:
      - multiply: 1000

MarcinJanCiesielski, 2023-05-09:
You was right - after adjust configuration telegrams are listed.
Now I also get 'T1 not found in configuration'. Is it because the telegram is probably C1 and wmbus cannot cope with it yet?

SzczepanLeon, 2023-05-09:
> Now I also get 'T1 not found in configuration'. Is it because the telegram is probably C1 and wmbus cannot cope with it yet?

No, it is T1 but you don't have sensor with that ID defined.

MarcinJanCiesielski, 2023-05-09:
OK. Thank you.

closed 2023-05-09 by MarcinJanCiesielski