SzczepanLeon/esphome-components #33: Forwarding telegrams to Wmbusmeters addon stops working beyond 1.5.1

convicte, 2023-05-17:
Telegrams are succesfully collected by the CC1101, but they are never received and/or decoded by Wmbusmeters:

```
wMBus-lib: Error during decoding '3 out of 6'[D][wmbus:158]: Meter ID [0x4113D74A] RSSI: -77 dBm LQI: 129 not found in configuration T: 1944304C4AD71341CC01A271150013C4B2AF258E358E2121FC1D (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -55.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -72 dBm LQI: 133 not found in configuration T: 1944304C3D504121D401A2013C0013DA0DD2E59A18BFFB2A5667 (26)
[D][wmbus:158]: Meter ID [0x4113D74A] RSSI: -77 dBm LQI: 128 not found in configuration T: 1944304C4AD71341CC01A201150013C4D763D9DFD0B52C20B310 (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -48.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -75 dBm LQI: 128 not found in configuration T: 1944304C3D504121D401A2113C0013DA1AFDAC31396C22F14F65 (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -77 dBm LQI: 130 not found in configuration T: 1944304C4AD241419801A2311C0013B456010AD1A92F6E3FB08F (26)
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -75 dBm LQI: 128 not found in configuration T: 1944304C3D504121D401A2213C0013DA238C76CC5B18489C6462 (26)
[D][sensor:093]: 'Reku temp monitor uptime': Sending state 778.60999 s with 0 decimals of accuracy
wMBus-lib: Error during decoding '3 out of 6'[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4113D74A] RSSI: -77 dBm LQI: 129 not found in configuration T: 1944304C4AD71341CC01A231150013C4EE120322B2C1464D9817 (26)
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -76 dBm LQI: 131 not found in configuration T: 1944304C4AD241419801A2511C0013B424E2BF2B6DC7BAE5E680 (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4113D74A] RSSI: -73 dBm LQI: 139 not found in configuration T: 1944304C4AD71341CC01A241150013C48BDEFF7357FA4B4CD71A (26)
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -76 dBm LQI: 131 not found in configuration T: 1944304C3D504121D401A2513C0013DA46408A9DBE23459D2B6F (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -48.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -75 dBm LQI: 129 not found in configuration T: 1944304C3D504121D401A2613C0013DA7F315060DC572FF00068 (26)
wMBus-lib: Error during decoding '3 out of 6'[D][wmbus:158]: Meter ID [0x4113D74A] RSSI: -76 dBm LQI: 128 not found in configuration T: 1944304C4AD71341CC01A261150013C4A5806C25145DF8FAE51F (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -51.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -76 dBm LQI: 128 not found in configuration T: 1944304C3D504121D401A2713C0013DA681E19CBFD84F62B196A (26)
wMBus-lib: Error during decoding '3 out of 6'[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -77 dBm LQI: 129 not found in configuration T: 1944304C4AD241419801A2111C0013B4785F9987EA88DD89828A (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][sensor:093]: 'Reku temp monitor uptime': Sending state 838.60999 s with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -78 dBm LQI: 129 not found in configuration T: 1944304C4AD241419801A2311C0013B456010AD1A92F6E3FB08F (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -76 dBm LQI: 130 not found in configuration T: 1944304C4AD241419801A2411C0013B433CDF6804C14633EFF82 (26)
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -73 dBm LQI: 132 not found in configuration T: 1944304C3D504121D401A2313C0013DA34A33F677ACB91477D60 (26)
[D][sensor:093]: 'Reku temp WiFi signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[D][wmbus:158]: Meter ID [0x4141D24A] RSSI: -77 dBm LQI: 128 not found in configuration T: 1944304C4AD241419801A2511C0013B424E2BF2B6DC7BAE5E680 (26)
[D][wmbus:158]: Meter ID [0x2141503D] RSSI: -75 dBm LQI: 128 not found in configuration T: 1944304C3D504121D401A2413C0013DA516FC3369FF09C46326D (26)
```

My current config for this ESP:

```yaml

esphome:
  name: "reku-temp-monitor"
  platform: ESP8266
  board: d1_mini
  
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: XXXXXXXX

wifi:
  reboot_timeout: 2min
  fast_connect: true
  power_save_mode: none
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    channel: 11
    hidden: true
    manual_ip:
      static_ip: XXXXXX
      gateway: XXXXXX
      subnet: XXXXXX
      dns1: XXXXXXX
      dns2: XXXXXX

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Water-Tank-Monitor"
    password: !secret wifi_password

captive_portal:

# i2c:
#   sda: D4
#   scl: D5
#   scan: true

button:
  - platform: restart
    name: "reku-temp-monitor restart"

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@1.5.1
    refresh: 0s
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO4
  gdo2_pin: GPIO5
  
  clients:
    - name: "wmbusmeters_udp"
      ip_address: XXXXXXXXX
      port: 7011
      format: RTLWMBUS
      transport: UDP

sensor:

  - platform: wifi_signal
    id: wifi_water_meter
    name: "Reku temp WiFi signal"
    update_interval: 10s

  - platform: uptime
    name: "Reku temp monitor uptime"
    
 ```

Changing from `github://SzczepanLeon/esphome-components@1.5.1` to `github://SzczepanLeon/esphome-components@main` and thus to version 2.X.X, will prevent the information from the meter reaching HA, and the sensors will no longer update, until I move back to `1.5.1`, which I learned the hard way today.

It's pretty easy to guess when I updated the node to the latest ESP version 2023.5.0 today, and while doing so update this component, since the link was set to `main`.
[image]

I'd be happy to test things out of you have an idea where to start poking.

SzczepanLeon, 2023-05-17:
Please switch to TCP.

convicte, 2023-05-17:
Tried both TCP and UDP on 2.X.X, to no effect.
Is there any particular reason to switch to TCP as a best practice?
I've been running UDP, since this was recommended about a year ago.

SzczepanLeon, 2023-05-17:
I have latest version (both wmbus and ESPHome) with TCP and everything is OK.
What wmbus version do you have?

convicte, 2023-05-17:
[image]

SzczepanLeon, 2023-05-17:
Not wmbusmeters but **wmbus** component for ESPHome.

convicte, 2023-05-17:
Is that what you mean?
[image]

SzczepanLeon, 2023-05-17:
Yes, so 1.5.1 is working but 2. is not?
What exacly 2. you have?

convicte, 2023-05-17:
I'll have to ask that we come back to it tomorrow.
I've already moved the board back to it's enclosure in the garage and what may be another piece of useful information, the OTA when running on 2.X.X. version is successful, but the unit will not reconnect the WiFi until updated via USB again with the same exact code.

SzczepanLeon, 2023-06-02:
So partial working. i.e. data are transferred but without telegram.
Fix is coming.

SzczepanLeon, 2023-06-05:
Fixed in 2.1.17

closed 2023-06-05 by SzczepanLeon