SzczepanLeon/esphome-components #35: izar - remaining_battery_life_y not calculated correctly

zibous, 2023-05-25:
In the latest current version, the battery life is not calculated correctly. The decimal places are missing

see:
<https://wmbusmeters.org/analyze/1944A511780731434418A261150013A9E04CC69B52B2C274E984> 

```
{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"18444331",
    "total_m3":361.125,
    "last_month_total_m3":356.27,
    "last_month_measure_date":"2023-05-01",
    "remaining_battery_life_y":10.5,    <----- 
    "current_alarms":"no_alarm",
    "previous_alarms":"no_alarm",
    "transmit_period_s":8,
    "timestamp":"2023-05-25T05:18:49Z"
}
```

[image]

```yaml
    # get the battery life time (wM-Bus v2.1.4)
    remaining_battery_life_y:
      name: "Wasseruhr Batterielebensdauer"
      id: "watermeter_batterie"
      accuracy_decimals: 2
      entity_category: "diagnostic"
      icon: mdi:battery
```

pepsonEL, 2023-05-25:
BTW
What is the webserver ? It is no default from ESPHome. Thanks

zibous, 2023-05-25:
Local webserver:

zibous, 2023-05-25:
> WHat you mean ?

The web_server component creates a simple web server on the node that can be accessed through any browser and a simple REST API. In order to save memory, I don't use the ESP WEBserver, but my own version that I have installed on my server.

SzczepanLeon, 2023-05-25:
I have correct values on latest version:
```
Using driver 'izar' for ID [0x43310778] RSSI: -58 dBm LQI: 128 T: 1944A511780731434418A261150013A9E04CC69B52B2C274E984 (26)
  current_alarms = 0.000
  last_month_total_water_m3 = 356.270
  previous_alarms = 0.000
  remaining_battery_life_y = 10.500
  total_water_m3 = 361.125
  transmit_period_s = 8.000
```

zibous, 2023-05-25:
@SzczepanLeon 

Strange, i (think) use the last version. However, I only get 10.00 years.

```
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    components: [wmbus]

ESPHome version 2023.5.4
wM-Bus v2.1.13:

[11:30:53][C][wmbus:349]:     Sensor '0x43430778 Watermeter battery life'
[11:30:53][C][wmbus:349]:       State Class: 'measurement'
[11:30:53][C][wmbus:349]:       Unit of Measurement: ''
[11:30:53][C][wmbus:349]:       Accuracy Decimals: 2
[11:30:53][C][wmbus:349]:       Icon: 'mdi:battery'

[11:30:59][I][wmbus:065]: Using driver 'izar' for ID [0x43430778] RSSI: -39 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A221140013D9E66D5F63F7A15B103FC2 (26)
[11:30:59][V][sensor:043]: '0x43430778 Watermeter battery life': Received new state 10.000000
```

Even if I download the latest version from Github and use that local. 

What can I do to find the error?

SzczepanLeon, 2023-05-25:
Please turn on very verbose traces.
I was checking this on some dev branch, but resoult should be similar than master. Later I will check master branch.

pepsonEL, 2023-05-25:
for apator162 also this info is available remaining_battery_life_y ?

SzczepanLeon, 2023-05-25:
> for apator162 also this info is available remaining_battery_life_y ?

No.

zibous, 2023-05-25:
> Please turn on very verbose traces.

I use:
```
## ---------------------------------------------------
## LOGGER COMPONENT
## ---------------------------------------------------
logger:
  id: appslogger
  level: VERY_VERBOSE
  baud_rate: 0
  logs:
    wmbus: VERY_VERBOSE
    wMBus-lib: VERY_VERBOSE
```

But I don't see any more information than DEBUG level.

```
[11:59:16][I][wmbus:065]: Using driver 'izar' for ID [0x43430778] RSSI: -39 dBm LQI: 128 Mode: T1 T: 1944A511780743434418A211140013D9DB1C859E95D5317D14C5 (26)
[11:59:16][V][sensor:043]: '0x43430778 CC1101 LQI': Received new state 128.000000
[11:59:16][D][sensor:093]: '0x43430778 CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 88 bytes
[11:59:16][V][sensor:043]: '0x43430778 CC1101 RSSI': Received new state -39.000000
[11:59:16][D][sensor:093]: '0x43430778 CC1101 RSSI': Sending state -39.00000 dBm with 0 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 88 bytes
[11:59:16][V][sensor:043]: '0x43430778 Watermeter current alarm': Received new state 0.000000
[11:59:16][D][sensor:093]: '0x43430778 Watermeter current alarm': Sending state 0.00000  with 0 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 96 bytes
[11:59:16][V][sensor:043]: '0x43430778 Water current month': Received new state 5433.000000
[11:59:16][D][sensor:093]: '0x43430778 Water current month': Sending state 5433.00000 L with 2 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 96 bytes
[11:59:16][V][sensor:043]: '0x43430778 Water last month': Received new state 470.723999
[11:59:16][D][sensor:093]: '0x43430778 Water last month': Sending state 470.72400 m³ with 3 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 96 bytes
[11:59:16][V][sensor:043]: '0x43430778 Watermeter previous alarm': Received new state 0.000000
[11:59:16][D][sensor:093]: '0x43430778 Watermeter previous alarm': Sending state 0.00000  with 0 decimals of accuracy
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 96 bytes
----
[11:59:16][V][sensor:043]: '0x43430778 Watermeter battery life': Received new state 10.000000
[11:59:16][D][sensor:093]: '0x43430778 Watermeter battery life': Sending state 10.00000  with 2 decimals of accuracy
---
[11:59:16][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:59:16][V][json:056]: Size after shrink 96 bytes
[11:59:16][V][sensor:043]: '0x43430778 Watermeter display': Received new state 476.157013
[11:59:16][D][sensor:093]: '0x43430778 Watermeter display': Sending state 476.15701 m³ with 3 decimals of accuracy
[11:59:16][V][sensor:043]: '0x43430778 Watermeter transmit periode': Received new state 8.000000
[11:59:16][D][sensor:093]: '0x43430778 Watermeter transmit periode': Sending state 8.00000 s with 0 decimals of accuracy
```

SzczepanLeon, 2023-05-25:
Version 2.1.15 should be.

zibous, 2023-05-25:
> Please try at least 2.1.15

Negativ get wM-Bus v2.1.16 ?:
```
Library Manager: Installing git+https://github.com/SzczepanLeon/wMbus-lib#1.1.1
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- wMbus-lib @ 1.1.1+sha.da2558f
|   |-- SPI @ 1.0
|   |-- SmartRC-CC1101-Driver-Lib @ 2.5.7
|-- wmbus-drivers @ 0.0.0+20230525121551.sha.8c88db3

[12:18:31][C][wmbus:297]: wM-Bus v2.1.16:
[12:18:31][C][wmbus:314]:   CC1101 SPI bus:
[12:18:31][C][wmbus:315]:     MOSI Pin: GPIO13
[12:18:31][C][wmbus:316]:     MISO Pin: GPIO12
[12:18:31][C][wmbus:317]:     CLK Pin:  GPIO14
[12:18:31][C][wmbus:318]:     CS Pin:   GPIO15
[12:18:31][C][wmbus:319]:     GDO0 Pin: GPIO5
[12:18:31][C][wmbus:320]:     GDO2 Pin: GPIO4
[12:18:31][C][wmbus:327]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, elf, evo868, fhkvdataiii, hydrocalm3, hydrus, iperl, itron, izar, mkradio3, mkradio4, qheat, qwater, sharky774, topaseskr, ultrimis, unismart, vario451

[12:18:31][C][logger:301]: Logger:
[12:18:31][C][logger:302]:   Level: DEBUG
[12:18:31][C][logger:303]:   Log Baud Rate: 0
[12:18:31][C][logger:305]:   Hardware UART: UART0
[12:18:31][C][logger:309]:   Level for 'wmbus': DEBUG
[12:18:31][C][logger:309]:   Level for 'wMBus-lib': DEBUG

[12:18:42][D][sensor:093]: '0x43430778 Watermeter battery life': Sending state 10.00000  with 2 decimals of accuracy
```

zibous, 2023-05-25:
Now with version 2.1.16:
```
[12:38:15][VV][wmbus:106]: Publishing 'remaining_battery_life_y' = 10.0000
[12:38:15][V][sensor:043]: '0x43430778 Watermeter battery life': Received new state 10.000000
[12:38:15][D][sensor:093]: '0x43430778 Watermeter battery life': Sending state 10.00000  with 2 decimals of accuracy
[12:38:15][VV][wmbus:106]: Publishing 'total_water_m3' = 476.1680
```

SzczepanLeon, 2023-05-25:
OK, later I will check where fault is.

zibous, 2023-05-25:
> Maybe sometimes it is transmited 10.5 and sometimes 10.0?

Yes, but i have checked this with my NanoCUL + wbmusmeters, also there sometime it is 10.00 or 10.50.
So all is o.k 👍  , sorry.

Please can you check
https://github.com/SzczepanLeon/esphome-components/issues/34