SzczepanLeon/esphome-components #39: Decryption problem when using qwater driver

Andes1984, 2023-07-19:
I'm using the C1 branch to read a Zenner Minomess meter. I know that there is no driver for it yet, but all I need is "total_water_m3", so I'm abusing the "qwater" driver.

But there is a decryption problem with the telegram:

[10:23:47][I][wmbus:071]: Using driver 'qwater' for ID [0x55228840] RSSI: -32 dBm LQI: 128 Mode: C1 T: 6644496A4088225518377253251425496A0106C80050050C35DD5BDABAD41BE760D033EEB4B867EB5544305D97153E031430AB2A7D7BC77712C437EE6C530C60CDBBCBBF23DD01C88125ADD3DACE621AA5564E83845C6EFD92CC2048A7EB833DB18F820D4D02F5 (103)
[10:23:47][E][wmbus:091]: Something was not OK during decrypting telegram for ID [0x55228840] 'qwater' key: 'BEFE9863E28D564A873A2EB9B87451A4'
[10:23:47][E][wmbus:092]: T : 6644496A4088225518377253251425496A0106C80050050C35DD5BDABAD41BE760D033EEB4B867EB5544305D97153E031430AB2A7D7BC77712C437EE6C530C60CDBBCBBF23DD01C88125ADD3DACE621AA5564E83845C6EFD92CC2048A7EB833DB18F820D4D02F5 (103)
[10:23:47][E][wmbus:093]: T': 6644496A40882255183772532514259C800AD34E865F1D3B27DAB25180209B5E5FF8C33D88AC6F2E96B4608543CC8521992F7767261BA8B070E0E76A366F9760CDBBCBBF23DD01C88125ADD3DACE621AA5564E83845C6EFD92CC2048A7EB833DB18F820D4D02F5 (103)
[10:23:47][E][wmbus:117]: Can't get value from telegram for ID [0x55228840] 'qwater'
[10:23:48][E][wmbus:118]: T : 6644496A40882255183772532514259C800AD34E865F1D3B27DAB25180209B5E5FF8C33D88AC6F2E96B4608543CC8521992F7767261BA8B070E0E76A366F9760CDBBCBBF23DD01C88125ADD3DACE621AA5564E83845C6EFD92CC2048A7EB833DB18F820D4D02F5 (103)

Do you have a hint?

SzczepanLeon, 2023-07-19:
Hi

I have two information, good and bad ;-)

Finally I have logs from C1 where telegram is OK. So it seems code for C1 reception is working.
Can You collect logs again, but not via web but directly via USB/RS232??

Bad one is that I have bug (or rather hardcoded decryption) in code. I think it should be fixed soon (maybe tomorrow).

SzczepanLeon, 2023-07-19:
Please try new C1 version - 2.2.7

Andes1984, 2023-07-20:
Thanks for checking and fixing so quickly :-)
Decryption error is gone now, but now (without me changing anything except new C1 version) the meter ID is not found in configuration:

`[06:02:45]wMBus-lib: Processing C1 A frame
[06:02:45] L=119 l=102 ll=102 lll=117
[06:02:46] FullFrame: 0x54, 0xCD, 0x66, 0x44, 0x49, 0x6A, 0x40, 0x88, 0x22, 0x55, 0x18, 0x37, 0xC7, 0xEB, 0x72, 0x53, 0x25, 0x14, 0x25, 0x49, 0x6A, 0x01, 0x06, 0x51, 0x00, 0x50, 0x05, 0x6D, 0x4E, 0xD0, 0x0E, 0x8B, 0xE1, 0xBA, 0x62, 0x8A, 0x5F, 0xD9, 0x45, 0xA7, 0x19, 0xA4, 0xC0, 0x7C, 0x80, 0x40, 0x58, 0x7C, 0xEF, 0xD7, 0xBF, 0x98, 0x9A, 0xCA, 0x10, 0x39, 0x66, 0xE3, 0xEC, 0xEE, 0x93, 0x98, 0xFB, 0xE4, 0x0B, 0x6E, 0x2F, 0x65, 0xF6, 0xE0, 0xE2, 0x2F, 0x02, 0xAF, 0x78, 0xEC, 0xC9, 0xC5, 0xEB, 0x5F, 0x44, 0x90, 0x80, 0x0A, 0x2A, 0xF0, 0xE7, 0x66, 0x70, 0x69, 0x47, 0x55, 0xA9, 0x3C, 0x80, 0x12, 0x42, 0x8F, 0xB5, 0x8A, 0x2D, 0x72, 0x52, 0x1E, 0x6B, 0x93, 0x82, 0x38, 0xC9, 0x7E, 0x44, 0xB5, 0x5A, 0x5D, 0xF5, 0x19, 0xE6, 0x14, 0x1D, 
[06:02:47] Frame: 0x66, 0x44, 0x49, 0x6A, 0x40, 0x88, 0x22, 0x55, 0x18, 0x37, 0xC7, 0xEB, 0x72, 0x53, 0x25, 0x14, 0x25, 0x49, 0x6A, 0x01, 0x06, 0x51, 0x00, 0x50, 0x05, 0x6D, 0x4E, 0xD0, 0x0E, 0x8B, 0xE1, 0xBA, 0x62, 0x8A, 0x5F, 0xD9, 0x45, 0xA7, 0x19, 0xA4, 0xC0, 0x7C, 0x80, 0x40, 0x58, 0x7C, 0xEF, 0xD7, 0xBF, 0x98, 0x9A, 0xCA, 0x10, 0x39, 0x66, 0xE3, 0xEC, 0xEE, 0x93, 0x98, 0xFB, 0xE4, 0x0B, 0x6E, 0x2F, 0x65, 0xF6, 0xE0, 0xE2, 0x2F, 0x02, 0xAF, 0x78, 0xEC, 0xC9, 0xC5, 0xEB, 0x5F, 0x44, 0x90, 0x80, 0x0A, 0x2A, 0xF0, 0xE7, 0x66, 0x70, 0x69, 0x47, 0x55, 0xA9, 0x3C, 0x80, 0x12, 0x42, 0x8F, 0xB5, 0x8A, 0x2D, 0x72, 0x52, 0x1E, 0x6B, 0x93, 0x82, 0x38, 0xC9, 0x7E, 0x44, 0xB5, 0x5A, 0x5D, 0xF5, 0x19, 0xE6, 
[06:02:47]   6644496A408822551837 3814 [C7EB]    7253251425496A0106510050056D4ED0 F174 [0E8B] E1BA628A5FD945A719A4C07C8040587C 1028 [EFD7] BF989ACA103966E3ECEE9398FBE40B6E D09A [2F65] F6E0E22F02AF78ECC9C5EB5F4490800A D50F [2AF0] E76670694755A93C8012428FB58A2D72 ADE1 [521E]    6B938238C97E44B55A5DF519E6 EBE2 [141D] 
[06:02:48]    CRC not OK Frame: 0x66, 0x44, 0x49, 0x6A, 0x40, 0x88, 0x22, 0x55, 0x18, 0x37, 0xC7, 0xEB, 0x72, 0x53, 0x25, 0x14, 0x25, 0x49, 0x6A, 0x01, 0x06, 0x51, 0x00, 0x50, 0x05, 0x6D, 0x4E, 0xD0, 0x0E, 0x8B, 0xE1, 0xBA, 0x62, 0x8A, 0x5F, 0xD9, 0x45, 0xA7, 0x19, 0xA4, 0xC0, 0x7C, 0x80, 0x40, 0x58, 0x7C, 0xEF, 0xD7, 0xBF, 0x98, 0x9A, 0xCA, 0x10, 0x39, 0x66, 0xE3, 0xEC, 0xEE, 0x93, 0x98, 0xFB, 0xE4, 0x0B, 0x6E, 0x2F, 0x65, 0xF6, 0xE0, 0xE2, 0x2F, 0x02, 0xAF, 0x78, 0xEC, 0xC9, 0xC5, 0xEB, 0x5F, 0x44, 0x90, 0x80, 0x0A, 0x2A, 0xF0, 0xE7, 0x66, 0x70, 0x69, 0x47, 0x55, 0xA9, 0x3C, 0x80, 0x12, 0x42, 0x8F, 0xB5, 0x8A, 0x2D, 0x72, 0x52, 0x1E, 0x6B, 0x93, 0x82, 0x38, 0xC9, 0x7E, 0x44, 0xB5, 0x5A, 0x5D, 0xF5, 0x19, 0xE6, 0x14, 0x1D, 
[06:02:48] ll=103
[06:02:48][D][wmbus:136]: Meter ID [0x55228840] RSSI: -32 dBm LQI: 128 Mode: C1 not found in configuration T: 6644496A4088225518377253251425496A0106510050056D4ED0E1BA628A5FD945A719A4C07C8040587CBF989ACA103966E3ECEE9398FBE40B6EF6E0E22F02AF78ECC9C5EB5F4490800AE76670694755A93C8012428FB58A2D726B938238C97E44B55A5DF519E6 (103)
`

Here's my yaml:

`substitutions:

  ## logger settings
  log_level: "VERY_VERBOSE"  # note that logging need memory, so in production mode use "WARN"
  log_wmbus: "VERY_VERBOSE" # Loglevel for wmbus meters component
  log_baudrate: "9600" # 0 disable uart logger messages

## ----------------------------------------------------------------
## HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
## ----------------------------------------------------------------
esp32:
  board: wemos_d1_mini32

preferences:
  flash_write_interval: 1min

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: wmbus

external_components:
  - source: github://SzczepanLeon/esphome-components@wmbus_c1
    refresh: 0s
    components: [wmbus]

wifi:
  networks:
    - ssid: "***"
      password: "***"
      priority: 0
    - ssid: "***"
      password: "***"
      priority: 1

captive_portal:

logger:
  id: appslogger
  level: ${log_level}
  baud_rate: ${log_baudrate}
  logs:
    wmbus: ${log_wmbus}
    wMBus-lib: ${log_wmbus}

ota:
  on_error:
    then:
      - logger.log:
          level: WARN
          tag: "system"
          format: "OTA update error %d"
          args: ["x"]

[... 45 lines cut ...]

  # Wifi quality RSSI in percentage
  # Received Signal Strength (RSSI) is a measure of incoherent (raw) RF power in a channel.
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "Device WLAN RSSI"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

  - platform: wmbus
    meter_id: 0x55228840
    type: qwater
    add_prefix: false
    key: "BEFE9863E28D564A873A2EB9B87451A4"
    total_water_m3:
      name: "***"
      id: ***
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
`

SzczepanLeon, 2023-07-20:
Fix was not perfect. It was fast with lots of praying ;-)

Please try new version 2.2.11   - also lots of hard-coded stuff, but ...

Andes1984, 2023-07-21:
Thanks, decryption works fine now!

FYI: the meter sends telegrams every 180s, but in the log i often see this message in between:

```
[08:37:26]wMBus-lib: Processing C1 B frame -- NOT supported yet
[08:37:26] Error during decoding 'unknown'
```

Not sure if this is relevant, I'm pretty sure that there is no other wMbus device in range of my receiver currently.

SzczepanLeon, 2023-07-21:
Pleases be aware that this is still c1 beta branch. I will merge C1 changes to main after weekend.

New version 2.2.14 is available. Please check it and collect traces (for long time period).
I've added debug traces around that C1 B-type frame.
Lots of "**wMBus-lib: Processing C1 B frame**" should be OK.

Also there is one small change around meter ID. If frame with long TPL is received then meter ID is taken from TPL header not from DLL.

So ID can change from 0x55228840 to 0x25142553

btw
What ID you have on sticker/meter?

Andes1984, 2023-07-21:
I will, thanks!

The full meter serial number looks like this:

9ZRI0125142553_EZRI1855228840

But the barcode sticker on the meter only shows EZRI1855228840, and the received telegram only has 855228840 as ID.

SzczepanLeon, 2023-07-21:
You can switch to main branch. From version 2.2.22 C1 is available (still experimental).

closed 2023-07-24 by SzczepanLeon