SzczepanLeon/esphome-components #48: Support for kamwater?

anhorbc, 2023-08-23:
Would it be possible to add support for kamwater?

SzczepanLeon, 2024-02-02:
Do you have example telegrams? Is it supported by wmbusmeters?

SzczepanLeon, 2024-02-08:
Do you have sample telegrams?

anhorbc, 2024-02-10:
> Do you have sample telegrams?
Something like this? I can't promise that the frame is from the actual kamwater device. 
[22:46:14][D][mbus:035]: Processing T1 A frame
[22:46:14][V][mbus:039]: Frame: 3939562793A2F2FABAB78354094289C51E2F510BF55959C07DC41537BF037E00FE04F62B80B79AF90E1CE1EEEE26F2EFC22BFA0E4CD8489E84AFAF8BA7A653EF42FF025FCD (69) [RAW]

At the moment I am using wmbusmeters on a raspberry pi to acquire the data. Would it help by some logs from that?

anhorbc, 2024-02-10:
More captured frames:
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D6D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFAB4
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D3D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E4242256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A315A898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BCBC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22B4B0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A625757D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF84A7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD6978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE89CC31EF47EE26F2EFC22BFAB44CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF4747EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB4CCEE11ABE82E42256D5E1A2757D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4C
[... cut ...] FC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE89883131EF47EE26F2EFC22BFA0E
6ACCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8A8AEFF94A6B5A31EE898831EF47EE26F2EFC22BFA0E4CD8489E84AF8BA7A653EF42FF025F5FCDCCB46AEE11ABE82E42256D5E1A6257D389EFB3FEF62CD7978F08BC8AEFF94A6B5AB4EE898831EF47EE26F2EFC22BFA0E

anhorbc, 2024-02-11:
Never mind. The reason for requesting kamwater was to get flowiq2200 working. I see now that support has been added since the original request.