SzczepanLeon/esphome-components #50: unable to compile

Woodpeckercz, 2023-09-13:
Hi,

I'm having issues compiling a watermeter firmware for wemos d1 mini that uses wmbus.
I'm getting this error:

In file included from /data/esp-wmbus/.piolibdeps/esp-wmbus/wMbus-lib/rf_mbus.cpp:6:
/data/esp-wmbus/.piolibdeps/esp-wmbus/wMbus-lib/tmode_rf_settings.hpp:8:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

**********************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:ELECHOUSE_CC1101_SRC_DRV.h
*
**********************************************************************************

    8 | #include <ELECHOUSE_CC1101_SRC_DRV.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/esp-wmbus/.pioenvs/esp-wmbus/lib523/wMbus-lib/rf_mbus.cpp.o] Error 1
========================= [FAILED] Took 56.43 seconds =========================

the wmbus library is being pulled from the repository so it looks like the ELECHOUSE_CC1101_SRC_DRV.h file is missing ?
or could it be something on my end ?

Regards.

ondras76, 2023-10-28:
I have the same issue
@SzczepanLeon 
[image]

SzczepanLeon, 2023-10-28:
Warning is not error.

Woodpeckercz, 2023-10-28:
Fyi: on my end it ended up being an issue with the esphome addon in home assistant.
I switched to compiling the firmware on raspberry pi and it worked fine.

closed 2023-10-28 by Woodpeckercz