SzczepanLeon/esphome-components #52: Amiplus not able to read all sensors

portexst, 2023-10-07:
Hello,
I am trying to configure the amiplus meter but for some reason I can read only:
- lqi
- rssi
- total_energy_consumption_kwh
- total_energy_production_kwh:

I cannot read current_power_production_kw, current_power_consumption_kw, voltage_at_phase_1_v, voltage_at_phase_2_v, voltage_at_phase_3_v sesnors.

I have checked https://wmbusmeters.org/ and when I put telegram and my decryption key I am able to read all information:

```
{
    "media":"electricity",
    "meter":"amiplus",
    "name":"",
    "id":"xxxxxxxxxxx",
    "current_power_consumption_kw":0.373,
    "current_power_production_kw":0,
    "total_energy_consumption_kwh":595.686,
    "total_energy_consumption_tariff_1_kwh":595.686,
    "total_energy_consumption_tariff_2_kwh":0,
    "total_energy_consumption_tariff_3_kwh":0,
    "total_energy_production_kwh":0.242,
    "total_energy_production_tariff_1_kwh":0.242,
    "total_energy_production_tariff_2_kwh":0,
    "total_energy_production_tariff_3_kwh":0,
    "voltage_at_phase_1_v":235,
    "voltage_at_phase_2_v":234,
    "voltage_at_phase_3_v":235,
    "timestamp":"2023-10-07T21:12:32Z"
}
``` 
In the esphome log I have found this:
```
23:08:34 | [I] | [wmbus:078] | Using driver 'amiplus' for ID [xxxxxxxxxxx] RSSI: -83 dBm LQI: 128 Mode: T1 T: 7E4401062839695601027ABD107005967F85132614508AD632A9A034C5272566E7CCCADE0E3B7221E81D06B658108AD5F883A723DF494187D6EBFF0241AA290DD6D16A420F6FC5CF3A2FED00C6B4A5BAB1A31B4232CA1BF5C49BD7081B8D3F20F0BD16051E801F725959CEEB9382C7D6FE2AF5E7FD3AF9418E3AE05942F076 (127)
23:08:34 | [D] | [wmbus:090] | Decrypted T : 7E4401062839695601027ABD1070052F2F0E038656590000008E10038656590000008E20030000000000008E30030000000000000E833C4202000000008E100DD6D16A420F6FC5CF3A2FED00C6B4A5BAB1A31B4232CA1BF5C49BD7081B8D3F20F0BD16051E801F725959CEEB9382C7D6FE2AF5E7FD3AF9418E3AE05942F076 (127)
23:08:34 | [D] | [sensor:093] | '0x56693928 Energy meter lqi': Sending state 128.00000  with 0 decimals of accuracy
23:08:34 | [D] | [sensor:093] | '0x56693928 Energy meter rssi': Sending state -83.00000 dBm with 0 decimals of accuracy
23:08:34 | [D] | [sensor:093] | '0x56693928 Energy meter total energy consumption kwh': Sending state 595.68597 kwh with 3 decimals of accuracy
23:08:34 | [D] | [sensor:093] | '0x56693928 Energy meter total energy production': Sending state 0.24200 kwh with 3 decimals of accuracy
23:08:34 | [W] | [component:204] | Component wmbus took a long time for an operation (0.10 s).
23:08:34 | [W] | [component:205] | Components should block for at most 20-30ms.
``` 

Here is my configuration:
```
esphome:
  name: ${device_name}
      
esp8266:
  board: d1_mini_pro
  
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]
    refresh: 0d
  
logger: 
  level: DEBUG

api:
  encryption:
    key: "${encryptionpass}"
  
ota:
  
web_server:
  port: 80
  
wifi:
  ssid: ${ssid}
  password: ${password}
  fast_connect: True
  reboot_timeout: ${reboot_time}
  ap:
    ssid: "${device_name}_Hotspot"
    password: "${appassword}"

time:
  - platform: sntp
    id: time_sntp

captive_portal:
  
wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO4
  gdo2_pin: GPIO5
  clients:
    - name: "wmbusmeters_udp"
      ip_address: "192.168.1.32"
      port: 9011
      format: RTLWMBUS
      transport: UDP
  
sensor:
  - platform: wmbus
    meter_id: xxxxxxxxxxxx
    type: amiplus
    mode: T1C1
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    lqi:
      name: "Energy meter lqi"
    rssi:
      name: "Energy meter rssi"
    total_energy_consumption_kwh:
      name: "Energy meter total energy consumption kwh"
      state_class: total_increasing
      device_class: energy
      unit_of_measurement: kwh
    current_power_consumption_kw:
      name: "Energy meter current power consumption"
      device_class: power
      state_class: measurement
      unit_of_measurement: kw
    total_energy_production_kwh:
      name: "Energy meter total energy production"
      state_class: total_increasing
      device_class: energy
      unit_of_measurement: kwh
    current_power_production_kw:
      name: "Energy meter current power production"
      device_class: power
      state_class: measurement
      unit_of_measurement: kw
    voltage_at_phase_1_v:
      name: "Energy meter voltage at phase 1"
      device_class: voltage
      state_class: measurement
      unit_of_measurement: V
    voltage_at_phase_2_v:
      name: "Energy meter voltage at phase 2"
      device_class: voltage
      state_class: measurement
      unit_of_measurement: V
    voltage_at_phase_3_v:
      name: "Energy meter voltage at phase 3"
      device_class: voltage
      state_class: measurement
      unit_of_measurement: V
      
  - platform: wmbus
    meter_id: xxxxxxxxxxxx
    type: apator162
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    lqi:
      name: "Cold Water lqi"
    rssi:
      name: "Cold Water rssi"
    total_water_m3:
      name: "Cold Water m3"
      state_class: total_increasing
      device_class: water
      unit_of_measurement: m3
  
  - platform: wmbus
    meter_id xxxxxxxxxxxx
    type: apator162
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    lqi:
      name: "Hot Water lqi"
    rssi:
      name: "Hot Water rssi"
    total_water_m3:
      name: "Hot Water m3"
      state_class: total_increasing
      device_class: water
      unit_of_measurement: m3
  
  # Uptime sensor
  - platform: uptime
    name: ${device_name} Uptime
    icon: mdi:clock-outline
  
  # WiFi Signal sensor
  - platform: wifi_signal
    name: ${device_name} Wifi Signal
    update_interval: 30s
    icon: mdi:wifi

text_sensor:
  - platform: version
    name: "${device_name} ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${device_name} ip"
    ssid:
      name: "${device_name} ssid"
  - platform: wmbus
    name: "Text debug for Apator 16-2"
``` 

Could you help me with this issue?

SzczepanLeon, 2023-10-07:
Here I see partial data:

https://wmbusmeters.org/analyze/7E4401062839695601027ABD1070052F2F0E038656590000008E10038656590000008E20030000000000008E30030000000000000E833C4202000000008E100DD6D16A420F6FC5CF3A2FED00C6B4A5BAB1A31B4232CA1BF5C49BD7081B8D3F20F0BD16051E801F725959CEEB9382C7D6FE2AF5E7FD3AF9418E3AE05942F076

It seems that not all frames are with voltage etc

Please capture frame with voltage.

portexst, 2023-10-07:
Hello,

https://wmbusmeters.org/analyze/7E4401062839695601027A181070059887A3F804E4A19584254C3D5E075120E3EAB018A93ABB17F53009EA9174B6ABEBF3325DA7808CF44186215F41C0EBA284902E5B1ADFDC01F7CDAF64A6AE62CFE6C4904371AC1DFE928085303309A03B48074FC5AFCAEF3D2DC07F211489197B025A7A7C70A0E67C1A8B63AF8EDDFB9D:auto:643838373435646238396A6734356679

```
23:54:04 | [I] | [wmbus:078] | Using driver 'amiplus' for ID [0x56693928] RSSI: -86 dBm LQI: 128 Mode: T1 T: 7E4401062839695601027A181070059887A3F804E4A19584254C3D5E075120E3EAB018A93ABB17F53009EA9174B6ABEBF3325DA7808CF44186215F41C0EBA284902E5B1ADFDC01F7CDAF64A6AE62CFE6C4904371AC1DFE928085303309A03B48074FC5AFCAEF3D2DC07F211489197B025A7A7C70A0E67C1A8B63AF8EDDFB9D (127)
23:54:04 | [D] | [wmbus:090] | Decrypted T : 7E4401062839695601027A181070052F2F0E037460590000008E10037460590000008E20030000000000008E30030000000000000E833C4202000000008E1084902E5B1ADFDC01F7CDAF64A6AE62CFE6C4904371AC1DFE928085303309A03B48074FC5AFCAEF3D2DC07F211489197B025A7A7C70A0E67C1A8B63AF8EDDFB9D (127)
23:54:04 | [D] | [sensor:093] | '0x56693928 Energy meter lqi': Sending state 128.00000  with 0 decimals of accuracy
23:54:04 | [D] | [sensor:093] | '0x56693928 Energy meter rssi': Sending state -86.00000 dBm with 0 decimals of accuracy
23:54:04 | [D] | [sensor:093] | '0x56693928 Energy meter total energy consumption kwh': Sending state 596.07397 kwh with 3 decimals of accuracy
23:54:04 | [D] | [sensor:093] | '0x56693928 Energy meter total energy production': Sending state 0.24200 kwh with 3 decimals of accuracy
23:54:04 | [W] | [component:204] | Component wmbus took a long time for an operation (0.10 s).
23:54:04 | [W] | [component:205] | Components should block for at most 20-30ms.
```

ficueu, 2024-01-17:
Hello, same issue for me. I think this is connected with decription bug because last 16 bytes in frame was not decrypted.

```
Received frame frame: 4E44B6101373250000027A0B0040050FBB06B505DDB02FB3CDB7968762BEDFCE0816D585F8991CF2819482F210B48ABCE44BEA5629AF9C0D607A6873E4F1C7 854FF18EECD57D41BC871D9123BE6FF5
Decrypted frame by wmbus: 4E44B6101373250000027A0B0040052F2F0C7814981300066D0072131131400E039582590500000E833C0000000000000B2B0203000BAB3C0000000AFDC9FC 854FF18EECD57D41BC871D9123BE6FF5
Decrypted frame by wmbusmeters: 4E44B6101373250000027A0B0040052F2F0C7814981300066D0072131131400E039582590500000E833C0000000000000B2B0203000BAB3C0000000AFDC9FC 0145020BABC8FC100000002F2F2F2F2F
```
I put space before last of 16 bytes in each frame to visualise it.

SzczepanLeon, 2024-01-17:
Great findings. I will check it.

ficueu, 2024-01-17:
@SzczepanLeon it was related to _tpl_num_encr_blocks_ in **wmbus-lib/utils.cpp**.
If I commented one if function, it works fine:
```
// if (tpl_num_encr_blocks) {
//   num_bytes_to_decrypt = tpl_num_encr_blocks*16;
// }
```

```
[21:09:59][VV][wmbus:126]: Publishing 'voltage_at_phase_1_v' = 245.0000
[21:09:59][V][sensor:043]: '0x257313 Tauron L1 Voltage': Received new state 245.000000
[21:09:59][D][sensor:094]: '0x257313 Tauron L1 Voltage': Sending state 245.00000 V with 0 decimals of accuracy
[21:09:59][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[21:09:59][V][json:058]: Size after shrink 88 bytes
[21:09:59][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 4033323085
  state: 245
  missing_state: NO
}
```

SzczepanLeon, 2024-01-17:
> @SzczepanLeon it was related to tpl_num_encr_blocks in wmbus-lib/utils.cpp

Probably bug is here:
https://github.com/SzczepanLeon/wMbus-lib/blob/main/utils.cpp#L53

ficueu, 2024-01-17:
> > @SzczepanLeon it was related to tpl_num_encr_blocks in wmbus-lib/utils.cpp
> 
> Probably bug is here: https://github.com/SzczepanLeon/wMbus-lib/blob/main/utils.cpp#L53
> 
> I will fix it tomorrow.

I fix it based on code from wmbusmetes, but i'm not sure if it is good solution.

```
  uint16_t tpl_cfg = ((uint32_t)frame[14] << 8) | ((uint32_t)frame[13]);
  uint8_t tpl_num_encr_blocks = (tpl_cfg >> 4) & 0x0f;
  if (tpl_num_encr_blocks) {
    num_bytes_to_decrypt = tpl_num_encr_blocks*16;
  }
```

SzczepanLeon, 2024-01-17:
> but i'm not sure if it is good solution

Should be OK.

ficueu, 2024-01-17:
Not working - same issue with encrypted and decrypted telegram:
```
Received: 4E44B6101373250000027AD70040052F8171BF536389B8E1A215177DEE0777AA46CCA722A65B32AFEF4548F4CF48FFD068AB8818B56B898A8802418F397D95 22ECCDCC64FB6337F07459C8C109AF8F
Decrypted: 4E44B6101373250000027AD70040052F2F0C7814981300066D004E171131400E036593590500000E833C0000000000000B2B6702000BAB3C0000000AFDC9FC 22ECCDCC64FB6337F07459C8C109AF8F
```
Compilation with code posted https://github.com/SzczepanLeon/esphome-components/issues/52#issuecomment-1896662841 works great.

SzczepanLeon, 2024-01-18:
Great

closed 2024-01-18 by SzczepanLeon