SzczepanLeon/esphome-components #53: Support for Frame B

cenobitedk, 2023-10-21:
Hi! I just found your repo and wanted to test it out, despite no specific driver for my Multical 21 meter. There's other solutions (MQTT) available bot none that works with esphome, so I wanted to test it.

I have set up the config with the required values, and let `type` default to `unknown`.

I get a lot these errors though:
```
...
[23:11:13]wMBus-lib: Processing C1 B frame -- NOT supported yet
[23:11:13] FullFrame: 0x54, 0x3D, 0x25, 0x44, 0x2D, 0x2C, 0x16, 0x30, 0x59, 0x57, 0x1B, 0x16, 0x8D, 0x20, 0x14, 0x91, 0x0A, 0xD5, 0xD5, 0x24, 0xA3, 0x8F, 0xF7, 0x34, 0x95, 0xD4, 0x71, 0x89, 0x78, 0x7C, 0x8B, 0x8D, 0x58, 0xF2, 0x02, 0x11, 0xDF, 0xA1, 0x3F, 0x18,
[23:11:13]wMBus-lib:  Error during decoding 'unknown'
...
```

The meter is broadcasting every 16 seconds and every 96 seconds a long frame. That looks a bit different and the length varies. e.g. 
```
...
[23:09:45]wMBus-lib: Processing C1 B frame -- NOT supported yet
[23:09:45] FullFrame: 0x54, 0x3D, 0x5E, 0x44, 0x2D, 0x2C, 0x94, 0x07, 0x52, 0x61, 0x19, 0x0C, 0x8D, 0x20, 0x5A, 0xC0, 0xC7, 0x62, 0x24, 0x24, 0x77, 0xA1, 0x78, 0x4B, 0xF5, 0xDB, 0xF2, 0x63, 0x8C, 0x99, 0x87, 0x28, 0x05, 0x2D, 0xEC, 0xC4, 0xBC, 0x18, 0xA4, 0x6E, 0xCA, 0x84, 0x80, 0x8D, 0x69, 0xAC, 0xA7, 0xBB, 0xED, 0xF3, 0x3E, 0x7B, 0x23, 0xC7, 0x84, 0x08, 0xB4, 0xF0, 0x82, 0xF7, 0x25, 0x76, 0x02, 0x9F, 0x7B, 0x54, 0x86, 0x3D, 0x5E, 0x44, 0x2D, 0x2C, 0x94, 0x07, 0x52, 0x61, 0x19, 0x0C, 0x8D, 0x20, 0x5A, 0xC0, 0xC7, 0x62, 0x24, 0x7E, 0x3A, 0xB7, 0x2B, 0x4E, 0x6B, 0x70, 0x74, 0xDE, 0x3A, 0x28, 0x05,
[23:09:45]wMBus-lib:  Error during decoding 'unknown'
[23:09:45][W][component:204]: Component wmbus took a long time for an operation (0.05 s).
[23:09:45][W][component:205]: Components should block for at most 20-30ms.
...
```

Generally it looks like the frame length varies, which confuses me, here's a screenshot:

[image]

Are you planning to work on supporting Frame B?

cenobitedk, 2023-10-24:
I realise now from the samples I have input from 2 different devices. So that helps clearing up some of the confusion.

I working on my own branch, but I can't work out the CRC. For B frames it "should" be fairly straight forward to calculate the CRC, but I can't find a single online tester where the CRC is matching the last 2 bytes of the full frame. Any ideas?

SzczepanLeon, 2023-10-24:
> I working on my own branch, but I can't work out the CRC. 

I have the same headache with A frame in C1. Seems that there is something special (or maybe I have bug in that part) in C1 reception. But in other hand somebody received C1 A frame but coded in higher layer and after decoding everything was OK.

Anyway I don't have time to fight with that CRC for C1 :(

cenobitedk, 2023-10-24:
> I don't see any catch in TI spec/pdf.

My neither. I also looked in other spec/pdf, but I haven't been able to find the original EN13757-4 spec document to verify anything. The CC1101 wasn't much help either (https://www.ti.com/lit/ds/symlink/cc1101.pdf).

> I don't have time to fight with that CRC for C1 :(

Okay, thanks for the reply.

SzczepanLeon, 2024-02-02:
Please try version _pre_ 3:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@pre_v3.0
    refresh: 0d
    components: [ wmbus ]
```

SzczepanLeon, 2024-02-06:
Added in version 3.0.0

closed 2024-02-06 by SzczepanLeon

cenobitedk, 2024-02-07:
Cool, I'll test it as soon as possible :)

SzczepanLeon, 2024-02-07:
Please remember about _mode_ configuration variable. Default is T1, I recommend to set it to _both_.

cenobitedk, 2024-02-07:
Roger that :)

Good work, I can see you made a lot of changes. I was trying to do my own fork, but didn't get past decrypting the ELL layer before doing CRC check, so great to see all the great work you have done! 💪

SzczepanLeon, 2024-02-07:
We will see. For now it is tested on one meter C1 B. And seems to be OK:
https://github.com/SzczepanLeon/esphome-components/issues/62