SzczepanLeon/esphome-components #54: Wired mbus support

skupu, 2023-11-06:
@SzczepanLeon @rafmaz @nepozs 
Hi,

Is there any chance to get the wired mbus protocol support?

I have few pieces of bmeters wired adapters and would like to use it, but until now no success.

Maybe there is a way to modify the code to make the wired version work?

Thank you!

chobott, 2023-11-28:
@SzczepanLeon @rafmaz @nepozs
For me too :) MBus is simillar datagram like wMBus i have success make M-Bus to UART and i can read datagram in RAW 

Thank you :-)

chobott, 2023-11-29:
YES! :-)
For UART this board - https://taaralabs.eu/m-bus-uart-converter/
convert MBus to UART

at now have ESP32 Devkit v1 with Tasmota and compiled SML support right this
https://tasmota.github.io/docs/Smart-Meter-Interface/

with this meter
https://tasmota.github.io/docs/Smart-Meter-Interface/#inepro-metering-pro380-mb-m-bus

chobott, 2023-12-12:
@SzczepanLeon is there any chance for support wired MBus?

SzczepanLeon, 2023-12-12:
I don't have any plans to add wired Mbus support.
But You can add code and create pull request.

closed 2024-02-02 by SzczepanLeon