SzczepanLeon/esphome-components #58: Reboooting - not getting data

evilmumi, 2023-11-26:
Hi,

i have different esp32 from az-delivery (germany) - all with same problems:

yaml:

```
esphome:
  name: water
  friendly_name: water

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

# Enable logging
logger: 
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "otapw"

wifi:
  ssid: ssid
  password: pw

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Water Fallback Hotspot"
    password: "pw"

captive_portal:

web_server:
  port: 80
  version: 2
  auth:
    username: user
    password: password   

time:
  - platform: sntp
    id: time_sntp

wmbus:
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO2
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

sensor:
  - platform: wmbus
    meter_id: 0
```

log:

```
[C][wifi:400]:   Hostname: 'water'
[C][wifi:402]:   Signal strength: -60 dB ▂▄▆█
[C][wifi:406]:   Channel: 6
[C][wifi:407]:   Subnet: 255.255.255.0
[C][wifi:408]:   Gateway: 192.168.0.1
[C][wifi:409]:   DNS1: 192.168.0.1
[C][wifi:410]:   DNS2: 0.0.0.0
[D][wifi:582]: Disabling AP...
[C][web_server:123]: Setting up web server...
[C][sntp:028]: Setting up SNTP...
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: water.local:3232
[C][ota:101]:   Using Password.
[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[C][api:025]: Setting up Home Assistant API server...

wMBus-lib: CC1101 version '79'
wMBus-lib: CC1101 initialized
[I][app:062]: setup() finished successfully!
E (24765) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (24765) task_wdt:  - loopTask (CPU 1)
E (24765) task_wdt: Tasks currently running:
E (24765) task_wdt: CPU 0: IDLE
E (24765) task_wdt: CPU 1: loopTask
E (24765) task_wdt: Aborting.

abort() was called at PC 0x400fdafc on core 0

Backtrace:0x400838d9:0x3ffbe9cc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

zibous, 2023-12-06:
@evilmumi 

How do you flash the firmeware ?
I use also esp32 from az-delivery...

```
## ----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4
##          ESP32 240MHz, 520KB RAM, 4MB Flash
## ----------------------------------------------------------------
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino
```

You can try the test firmware from my testcases:
https://github.com/zibous/ha-watermeter/tree/master/esphome/testcases

evilmumi, 2023-12-06:
> @evilmumi
>
> How do you flash the firmeware ? I use also esp32 from az-delivery...
>
> ```
> ## ----------------------------------------------------------------
> ## HARDWARE az-delivery-devkit-v4 [...]

So i have tried different ways. I try to describe one by one.

1. Download https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/wm-esp32-izar-test.yaml
2. copy it via winscp on my docker esphome container share
3. change content to log_baudrate: "115200" 
4. save
5. install
6. manual download
7. modern format
8. download
9. open https://web.esphome.io/?dashboard_install
10. connect and flash the builded file via web

logs:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[I][logger:326]: Log initialized
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[W][system:313]: BOOTMESSAGE:water-meter-esp32-izar-test API is connected, Device ready!
[V][esp32.preferences:059]: nvs_get_blob('88491487'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:044]: Enabling AP.
[    57][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[V][wifi_esp32:417]: Event: WiFi ready
[   220][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  1559][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[V][wifi_esp32:512]: Event: WiFi AP start[  1563][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped

[  1571][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
[  1572][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[V][wifi_esp32:516]: Event: WiFi AP stop[  1586][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started

[  1595][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[V][wifi_esp32:512]: Event: WiFi AP start
[  1612][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[V][wifi_esp32:428]: Event: WiFi STA start
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: water-meter-esp32-izar-test.local:3232
[C][ota:101]:   Using Password.

wMBus-lib: CC1101 version '60'
wMBus-lib: CC1101 initialized
[I][app:062]: setup() finished successfully!
E (13642) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13642) task_wdt:  - loopTask (CPU 1)
E (13642) task_wdt: Tasks currently running:
E (13642) task_wdt: CPU 0: IDLE
E (13642) task_wdt: CPU 1: loopTask
E (13642) task_wdt: Aborting.

abort() was called at PC 0x400ff548 on core 0

Backtrace:0x400838d9:0x3ffbeadc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

zibous, 2023-12-06:
> 6\. flash via ota the file https://github.com/zibous/ha-watermeter/blob/master/esphome/testcases/water-meter-esp32-izar-test.bin

Pinlayout 
```
## ------------------------------------------------------------------
##           WMBUS CC1101 --> ESP32 az-delivery-devkit-v4
##       https://www.az-delivery.de/products/esp-32-dev-kit-c-v4
## ------------------------------------------------------------------
##
##
##                                                               o 1 (3.3V)
##                                                               |
##   ╭――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――o―╮
##   |                                                             |
##   |                                                             |
## - | 5v               az-delivery-devkit-v4                      | -- ANT
##   |                                                             |
##   |                          16 17 5  18 19               23    |
##   ╰――x――x――x――x――x――x――x――x――o――x――o――o――o――o――o――o――o――o――o――o―╯
##                              |  |  |  |  |                 |   |
##                              o  |  |  o  |                 |   ╰-o - 2 (GND)
##                              7  o  |  4  o                 o
##                            GDO0 6  | CLK 5                 3
##                               GD02 o    MISO              M0SI
##                                    8
##                                   CSN
##
## ------------------------------------------------------------------
wmbus:
  mosi_pin: GPIO23  ## SI:   braun   3: MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19  ## SO:   grün    5: MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18   ## SCLK: violett 4: SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05    ## CSN:  orange  8: CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16  ## GD00: gelb    7: RX Clock output. High Impedance !
  gdo2_pin: GPIO17  ## GD02: weiss   6: TX FIFO status signals. High Impedance !
```

`water-meter-esp32-izar-test.bin` uses:
gdo0_pin: GPIO16  (not GPIO21) !
gdo2_pin: GPIO17  (not GPIO22) !

evilmumi, 2023-12-06:
it doesn't make any difference.

if it is ok for you we can also do it in german :)

Would you be so kind an build a new one should I attach pictures of cabling?

zibous, 2023-12-07:
> perhaps you use other wifi versions? i think there could be a problem with fritzbox mesh but not sure

Yes is use a own `2.4GHz Network Only` with unifi ap.
But you can try:

> ESPs can handle mesh networking by enabling enable_rrm in the WiFi setting. See https://esphome.io/components/wifi.html However, you need to configure it with framework esp-idf, which can bring some drawbacks. See https://esphome.io/components/esp32.html?highlight=framework

> Would you be so kind an build a new one should I attach pictures of cabling?

see: https://github.com/zibous/ha-watermeter/tree/master/esphome

evilmumi, 2023-12-07:
So I will give it a try with esp-idf and enable_rrm . But I think there is another problem because i tested it on a mobile phone hotspot with same problems ...

cabling is fine.

SzczepanLeon, 2024-02-08:
Please try version 3.x.x

closed 2024-02-08 by SzczepanLeon