SzczepanLeon/esphome-components #60: No incmoming data when MQTT section enabled

kregiel, 2023-12-06:
Refers to:
- meters: apator-8 and apator 16-2
- hardware: ESP32 + CC1101
- sensor configuration: no sensors defined

Problem:
When I activate an MQTT section (with valid data entered), entries regarding incoming data no longer appear in the logs. Commenting out the MQTT section causes the logs to appear again.
Since I receive messages with the MQTT section disabled, I assume there is no hardware problem, but a software problem.

SzczepanLeon, 2024-02-02:
Probably MQTT component consume lots of time and we have CC1101 Rx FIFO overflow. With current implementation there is no solution for that, maybe interrupt driven Rx read can help.

SzczepanLeon, 2024-02-07:
Please try version 3.1.1

closed 2024-02-07 by SzczepanLeon