SzczepanLeon/esphome-components #62: Support for flowiq2200

evilmumi, 2023-12-15:
Hi,

is it possible to add flowiq2200 support?
I found something here but dont know how to implement:
https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_flowiq2200.cc

Log is like:
```

wMBus-lib: Processing C1 B frame -- NOT supported yet
 FullFrame: 0x54, 0x3D, 0x34, 0x44, 0x37, 0x2C, 0x37, 0x79, 0x71, 0x23, 0x3A, 0x16, 0x8D, 0x20, 0xA0, 0xA0, 0x81, 0x12, 0x3D, 0x48, 0x20, 0x86, 0xA0, 0xA2, 0x57, 0x93, 0x83, 0x8C, 0xD2, 0xDF, 0xDB, 0x41, 0x45, 0xE8, 0x5B, 0x12, 0x1D, 0x72, 0x38, 0x71, 0x82, 0x91, 0x00, 0xF1, 0x8A, 0xA3, 0x9D, 0xD5, 0x67, 0x4F, 0xC3, 0xCF, 0x45, 0xAF, 0xFE, 
wMBus-lib:  Error during decoding 'unknown'
```
```

wMBus-lib: Processing C1 B frame -- NOT supported yet
 FullFrame: 0x54, 0x3D, 0x9A, 0x22, 0x1B, 0x96, 0x1B, 0xBC, 0xB8, 0xD8, 0xDE, 0x84, 0xA3, 0x48, 0x29, 0x29, 0x8A, 0xE1, 0xBF, 0x44, 0x93, 0x82, 0x18, 0x64, 0xA8, 0xFA, 0xD9, 0x9B, 0xF4, 0x2C, 0x3B, 0xE4, 0x19, 0x23, 0xBE, 0x10, 0xB9, 0xB9, 0x0B, 0xD1, 0xE1, 0x9D, 0x54, 0x09, 0x7B, 0x3C, 0x42, 0xF3, 0x5B, 0x5B, 0xE3, 0xC0, 0x13, 0x05, 0x8D, 0x21, 0xBF, 0x2D, 0xFF, 0x9C, 0x56, 0x61, 0x36, 0x11, 0xBD, 0xB7, 0x3D, 0x9A, 0x22, 0x1B, 0x96, 0x1B, 0xBC, 0xB8, 0xD8, 0xDE, 0x84, 0xA3, 0x48, 0x29, 0xBC, 0x47, 0x65, 0xA4, 0x2C, 0x5F, 0xFF, 0x81, 0xBB, 0x29, 0xDF, 0x9B, 0xF4, 0x2C, 0x3B, 0xE4, 0x19, 0x23, 0xBE, 0x10, 0xB9, 0xB9, 0x0B, 0xD1, 0xE1, 0x9D, 0x54, 0x09, 0x7B, 0x3C, 0x42, 0xF3, 0x5B, 0x5B, 0xE3, 0xC0, 0x13, 0x05, 0x8D, 0x21, 0xBF, 0x2D, 0xFF, 0x9C, 0x56, 0x61, 0x36, 0x11, 0xBD, 0xB7, 0x3D, 0x9A, 0x22, 0x1B, 0x96, 0x1B, 0xBC, 0xB8, 0xD8, 0xDE, 0x84, 0xA3, 0x48, 0x29, 0xBC, 0x47, 0x65, 0xA4, 0x2C, 0x5F, 0xFF, 0x81, 0xBB, 0x29, 0xDF, 0xC9, 0x7E, 
wMBus-lib:  Error during decoding 'unknown'
```

SzczepanLeon, 2023-12-19:
Yes it is possible, but I don't have plans to add B frame support.

evilmumi, 2023-12-19:
> Yes it is possible, but I don't have plans to add B frame support.

Is there any chance to "bribe" you?

SzczepanLeon, 2023-12-19:
It is not possible. I don't have time to read about B frame structure etc.

But if You really want that feature You can try to find in internet/github code for that B frame (how to calculate CRC and how to cut CRC from frame).

evilmumi, 2023-12-20:
> It is not possible. I don't have time to read about B frame structure etc.
> 
> But if You really want that feature You can try to find in internet/github code for that B frame (how to calculate CRC and how to cut CRC from frame).

I am not a big coder.
I found 2 things
https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_flowiq2200.cc
and
https://github.com/chester4444/esp-multical21/tree/master/src

is this helpful that you integrate it?

SzczepanLeon, 2024-02-02:
Please try version pre 3:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@pre_v3.0. refresh: 0d
components: [ wmbus ]
```

evilmumi, 2024-02-04:
Thanks. Now I am getting this:

`[17:29:12][W][component:214]: Component debug took a long time for an operation (0.10 s).
[17:29:12][W][component:215]: Components should block for at most 20-30ms.
[17:29:12][E][rxLoop:157]: Length problem: req(53) != rx(51)
[17:29:12][D][rxLoop:159]: Have 51 bytes from CC1101 Rx
[17:29:12][D][mbus:024]: Processing C1 B frame
[17:29:12][V][mbus:028]: Frame: 343D34000044372C347971233A168D202F6060253F212A014752D819915FB658B5C2128F20FEB43F9084041B18D2B3EBD80D26 (51) [with CRC]
[17:29:12][V][mbus:153]: Package (51) too short for packet Length: 52
[17:29:12][V][mbus:154]:   53 > 51`

SzczepanLeon, 2024-02-04:
> Now I am getting this:

Thanks for traces. I see couple places with bug in my code - will be fixed probably today.

evilmumi, 2024-02-04:
If I can help - just pls say me how

SzczepanLeon, 2024-02-04:
> If I can help - just pls say me how

Yes You can. I don't have meter transmiting in C1 B, so I need logs.
Please wait for next version with fix.

evilmumi, 2024-02-05:
so i should change sensor platform wmbus type flowiq2000 to type evo868 ?

SzczepanLeon, 2024-02-05:
> so i should change sensor platform wmbus type flowiq2000 to type evo868 ?

For now - yes. It should be like:
```
sensor:
  - platform: wmbus
    meter_id: 0x23717934
    type: evo868
    key: "key_key_key"
    total_water_m3:
      name: "test meter"
```

Driver for flowiq2000 should be ready soon, but first we should get correct C1 B frame.

SzczepanLeon, 2024-02-05:
Please include text logs.

evilmumi, 2024-02-05:
here you go - fine with it?
`[11:28:40][V][sensor:043]: 'Heap Free': Received new state 14056.000000
[11:28:40][D][sensor:093]: 'Heap Free': Sending state 14056.00000 B with 0 decimals of accuracy
[11:28:40][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:40][V][json:058]: Size after shrink 76 bytes
[11:28:40][V][sensor:043]: 'Heap Max Block': Received new state 7552.000000
[11:28:40][D][sensor:093]: 'Heap Max Block': Sending state 7552.00000 B with 0 decimals of accuracy
[11:28:40][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:40][V][json:058]: Size after shrink 80 bytes
[11:28:40][V][sensor:043]: 'Heap Fragmentation': Received new state 18.000000
[11:28:40][D][sensor:093]: 'Heap Fragmentation': Sending state 18.00000 % with 1 decimals of accuracy
[11:28:40][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:40][V][json:058]: Size after shrink 84 bytes
[11:28:40][V][sensor:043]: 'Application Loop Time': Received new state 342.000000
[11:28:40][D][sensor:093]: 'Application Loop Time': Sending state 342.00000 ms with 0 decimals of accuracy
[11:28:40][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:40][V][json:058]: Size after shrink 84 bytes
[11:28:40][W][component:214]: Component debug took a long time for an operation (0.10 s).
[11:28:40][W][component:215]: Components should block for at most 20-30ms.
[11:28:45][V][sensor:043]: 'Heap Free': Received new state 14056.000000
[11:28:45][D][sensor:093]: 'Heap Free': Sending state 14056.00000 B with 0 decimals of accuracy
[11:28:45][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:45][V][json:058]: Size after shrink 76 bytes
[11:28:45][V][sensor:043]: 'Heap Max Block': Received new state 7552.000000
[11:28:45][D][sensor:093]: 'Heap Max Block': Sending state 7552.00000 B with 0 decimals of accuracy
[11:28:45][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:45][V][json:058]: Size after shrink 80 bytes
[11:28:45][V][sensor:043]: 'Heap Fragmentation': Received new state 18.000000
[11:28:45][D][sensor:093]: 'Heap Fragmentation': Sending state 18.00000 % with 1 decimals of accuracy
[11:28:45][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:45][V][json:058]: Size after shrink 84 bytes
[11:28:45][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[11:28:45][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[11:28:45][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:45][V][json:058]: Size after shrink 84 bytes
[11:28:45][W][component:214]: Component debug took a long time for an operation (0.10 s).
[11:28:45][W][component:215]: Components should block for at most 20-30ms.
[11:28:50][V][sensor:043]: 'Heap Free': Received new state 14056.000000
[11:28:50][D][sensor:093]: 'Heap Free': Sending state 14056.00000 B with 0 decimals of accuracy
[11:28:50][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:50][V][json:058]: Size after shrink 76 bytes
[11:28:50][V][sensor:043]: 'Heap Max Block': Received new state 7552.000000
[11:28:50][D][sensor:093]: 'Heap Max Block': Sending state 7552.00000 B with 0 decimals of accuracy
[11:28:50][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:50][V][json:058]: Size after shrink 80 bytes
[11:28:50][V][sensor:043]: 'Heap Fragmentation': Received new state 21.000000
[11:28:50][D][sensor:093]: 'Heap Fragmentation': Sending state 21.00000 % with 1 decimals of accuracy
[11:28:50][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:50][V][json:058]: Size after shrink 84 bytes
[11:28:50][V][sensor:043]: 'Application Loop Time': Received new state 118.000000
[11:28:50][D][sensor:093]: 'Application Loop Time': Sending state 118.00000 ms with 0 decimals of accuracy
[11:28:50][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:50][V][json:058]: Size after shrink 84 bytes
[11:28:50][W][component:214]: Component debug took a long time for an operation (0.10 s).
[11:28:50][W][component:215]: Components should block for at most 20-30ms.
[11:28:55][V][sensor:043]: 'Heap Free': Received new state 14200.000000
[11:28:55][D][sensor:093]: 'Heap Free': Sending state 14200.00000 B with 0 decimals of accuracy
[11:28:55][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:55][V][json:058]: Size after shrink 76 bytes
[11:28:55][V][sensor:043]: 'Heap Max Block': Received new state 7552.000000
[11:28:55][D][sensor:093]: 'Heap Max Block': Sending state 7552.00000 B with 0 decimals of accuracy
[11:28:55][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:55][V][json:058]: Size after shrink 80 bytes
[11:28:55][V][sensor:043]: 'Heap Fragmentation': Received new state 22.000000
[11:28:55][D][sensor:093]: 'Heap Fragmentation': Sending state 22.00000 % with 1 decimals of accuracy
[11:28:55][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:28:55][V][json:058]: Size after shrink 84 bytes
[11:28:55][V][sensor:043]: 'Application Loop Time': Received new state 120.000000
[11:28:55][D][sensor:093]: 'Application Loop Time': Sending state 120.00000 ms with 0 decimals of accuracy
[11:28:55][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[... 296 lines cut ...]
[... cut ...]
[11:30:15][V][json:058]: Size after shrink 84 bytes
[11:30:15][V][sensor:043]: 'Application Loop Time': Received new state 118.000000
[11:30:15][D][sensor:093]: 'Application Loop Time': Sending state 118.00000 ms with 0 decimals of accuracy
[11:30:15][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:30:15][V][json:058]: Size after shrink 84 bytes
[11:30:15][W][component:214]: Component debug took a long time for an operation (0.10 s).
[11:30:15][W][component:215]: Components should block for at most 20-30ms.`

SzczepanLeon, 2024-02-05:
Please try this version and collect VERBOSE:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@pre_v3.0_noCrc
    refresh: 0d
    components: [ wmbus ]
```

evilmumi, 2024-02-05:
here we go again.
do you need more lines?
`[12:04:09][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[12:04:09][D][mbus:022]: Processing C1 B frame
[12:04:09][V][mbus:026]: Frame: 3444372C347971233A168D208503034D47E4267B29F54D2C0DBAFEABFBFD637A980568CC43188EAFA09126B45AD56753CF785DDB64 (53) [with CRC]
[12:04:09][V][mbus:159]: Validating CRC for Block1 + Block2
[12:04:09][I][crc:035]:   calculated: 0x7E01, read: 0xDB64  !!!
[12:04:09][D][mbus:056]: Frame: 3244372C347971233A168D208503034D47E4267B29F54D2C0DBAFEABFBFD637A980568CC43188EAFA09126B45AD56753CF785D (51) [without CRC]
[12:04:09][D][wmbus:195]: Meter ID [0x23717934] RSSI: -95 dBm LQI: 159 Frame: C1 B not found in configuration T: 3244372C347971233A168D208503034D47E4267B29F54D2C0DBAFEABFBFD637A980568CC43188EAFA09126B45AD56753CF785D (51)
[12:05:42][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[12:05:42][D][mbus:022]: Processing C1 B frame
[12:05:42][V][mbus:026]: Frame: 34443728347971233A168D208BA2A2B51F90DA26D6D3003A3E1AFBAF2B69D5D768B2921C8BBA984B6479423CE27BD81D060DDB92D6 (53) [with CRC]
[12:05:42][V][mbus:159]: Validating CRC for Block1 + Block2
[12:05:42][I][crc:035]:   calculated: 0x3310, read: 0x92D6  !!!
[12:05:42][D][mbus:056]: Frame: 32443728347971233A168D208BA2A2B51F90DA26D6D3003A3E1AFBAF2B69D5D768B2921C8BBA984B6479423CE27BD81D060DDB (51) [without CRC]
[12:05:42][D][wmbus:195]: Meter ID [0x23717934] RSSI: -96 dBm LQI: 155 Frame: C1 B not found in configuration T: 32443728347971233A168D208BA2A2B51F90DA26D6D3003A3E1AFBAF2B69D5D768B2921C8BBA984B6479423CE27BD81D060DDB (51)
`

SzczepanLeon, 2024-02-05:
Please add sensor configuration with popper key.

```
sensor:
  - platform: wmbus
    meter_id: 0x23717934
    type: evo868
    key: "key_key_key"
    total_water_m3:
      name: "test meter"
```

evilmumi, 2024-02-05:
here we go

`[12:24:19][C][wmbus:440]:   Meter:
[12:24:19][C][wmbus:441]:     ID: 594639156 [0x23717934]
[12:24:19][C][wmbus:442]:     Type: evo868
[12:24:19][C][wmbus:443]:     Mode: T1
[12:24:19][C][wmbus:444]:     Key: '05234353253523342434234432(FAKE)F3B3'
[12:24:19][C][wmbus:446]:     Sensor '0x23717934 test meter'
[12:24:19][C][wmbus:446]:       Device Class: 'water'
[12:24:19][C][wmbus:446]:       State Class: 'total_increasing'
[12:24:19][C][wmbus:446]:       Unit of Measurement: 'm³'
[12:24:19][C][wmbus:446]:       Accuracy Decimals: 3
[12:24:19][C][wmbus:446]:       Icon: 'mdi:water'`

after that i got no information about any meter in log.
only
this stuff
`[12:27:28][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[12:27:28][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[12:27:28][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:27:28][V][json:058]: Size after shrink 84 bytes
[12:27:28][W][component:214]: Component debug took a long time for an operation (0.10 s).
[12:27:28][W][component:215]: Components should block for at most 20-30ms.
[12:27:33][V][sensor:043]: 'Heap Free': Received new state 17160.000000
[12:27:33][D][sensor:093]: 'Heap Free': Sending state 17160.00000 B with 0 decimals of accuracy
[12:27:33][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:27:33][V][json:058]: Size after shrink 76 bytes
[12:27:33][V][sensor:043]: 'Heap Max Block': Received new state 10376.000000
[12:27:33][D][sensor:093]: 'Heap Max Block': Sending state 10376.00000 B with 0 decimals of accuracy
[12:27:33][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:27:33][V][json:058]: Size after shrink 80 bytes
[12:27:33][V][sensor:043]: 'Heap Fragmentation': Received new state 16.000000
[12:27:33][D][sensor:093]: 'Heap Fragmentation': Sending state 16.00000 % with 1 decimals of accuracy
[12:27:33][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:27:33][V][json:058]: Size after shrink 84 bytes
[12:27:33][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[12:27:33][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[12:27:33][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:27:33][V][json:058]: Size after shrink 84 bytes
[12:27:33][W][component:214]: Component debug took a long time for an operation (0.10 s).
[12:27:33][W][component:215]: Components should block for at most 20-30ms.
`

is the id wrong inteprated? hex / binary?

SzczepanLeon, 2024-02-05:
Strange -- lets downgrade logs to DEBUG.

evilmumi, 2024-02-05:
no useful output in log since 8 minutes - only repeating stuff:
`12:43:43][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[12:43:43][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[12:43:43][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:43:43][V][json:058]: Size after shrink 84 bytes
[12:43:43][W][component:214]: Component debug took a long time for an operation (0.10 s).
[12:43:43][W][component:215]: Components should block for at most 20-30ms.
[12:43:48][V][sensor:043]: 'Heap Free': Received new state 16992.000000
[12:43:48][D][sensor:093]: 'Heap Free': Sending state 16992.00000 B with 0 decimals of accuracy
[12:43:48][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:43:48][V][json:058]: Size after shrink 76 bytes
[12:43:48][V][sensor:043]: 'Heap Max Block': Received new state 10360.000000
[12:43:48][D][sensor:093]: 'Heap Max Block': Sending state 10360.00000 B with 0 decimals of accuracy
[12:43:48][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:43:48][V][json:058]: Size after shrink 80 bytes
[12:43:48][V][sensor:043]: 'Heap Fragmentation': Received new state 15.000000
[12:43:48][D][sensor:093]: 'Heap Fragmentation': Sending state 15.00000 % with 1 decimals of accuracy
[12:43:48][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:43:48][V][json:058]: Size after shrink 84 bytes
[12:43:48][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[12:43:48][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[12:43:48][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:43:48][V][json:058]: Size after shrink 84 bytes
[12:43:48][W][component:214]: Component debug took a long time for an operation (0.10 s).
[12:43:48][W][component:215]: Components should block for at most 20-30ms.`

evilmumi, 2024-02-05:
postioned it nearer to the watermeter:

`[12:48:25][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[12:48:25][D][mbus:022]: Processing C1 B frame
[12:48:25][V][mbus:026]: Frame: 3444372C347971233A168D202AD16C3F2121019F00A219D2AF992B490B4ECF84A1DEA0DFA04E456E6F7864A9BF00BFA9051E86FAC0 (53) [with CRC]
[12:48:25][V][mbus:159]: Validating CRC for Block1 + Block2
[12:48:25][I][crc:035]:   calculated: 0xA936, read: 0xFAC0  !!!
[12:48:25][D][mbus:056]: Frame: 3244372C347971233A168D202AD16C3F2121019F00A219D2AF992B490B4ECF84A1DEA0DFA04E456E6F7864A9BF00BFA9051E86 (51) [without CRC]
[12:48:25][D][wmbus:195]: Meter ID [0x23717934] RSSI: -87 dBm LQI: 136 Frame: C1 B not found in configuration T: 3244372C347971233A168D202AD16C3F2121019F00A219D2AF992B490B4ECF84A1DEA0DFA04E456E6F7864A9BF00BFA9051E86 (51)`

`[12:48:57][D][rxLoop:159]: Have 74 bytes from CC1101 Rx
[12:48:57][D][mbus:022]: Processing C1 B frame
[12:48:57][V][mbus:026]: Frame: 4944372C347971233A168D202CD36C3F3F4D2904B1502C07371EC0A83FE653CBAD684C3CD82959BECD869A709FDBE74597696BB8E688686FF4101D10DB4B5CA93D4944372C347971233A (74) [with CRC]
[12:48:57][V][mbus:159]: Validating CRC for Block1 + Block2
[12:48:57][I][crc:035]:   calculated: 0x5C6B, read: 0x233A  !!!
[12:48:57][D][mbus:056]: Frame: 4744372C347971233A168D202CD36C3F3F4D2904B1502C07371EC0A83FE653CBAD684C3CD82959BECD869A709FDBE74597696BB8E688686FF4101D10DB4B5CA93D4944372C347971 (72) [without CRC]
[12:48:58][D][wmbus:195]: Meter ID [0x23717934] RSSI: -86 dBm LQI: 135 Frame: C1 B not found in configuration T: 4744372C347971233A168D202CD36C3F3F4D2904B1502C07371EC0A83FE653CBAD684C3CD82959BECD869A709FDBE74597696BB8E688686FF4101D10DB4B5CA93D4944372C347971 (72)
[12:48:58][W][component:214]: Component wmbus took a long time for an operation (0.07 s).`

SzczepanLeon, 2024-02-05:
With sensor section?

evilmumi, 2024-02-05:
with 
## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:
  - platform: wmbus
    meter_id: "0x23717934"
    mode: T1C1
    type: evo868
    key: "05NUMBERS3B3"
    total_water_m3:
      name: "test meter"

evilmumi, 2024-02-05:
[13:06:43][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[13:06:43][D][mbus:022]: Processing C1 B frame
[13:06:43][V][mbus:026]: Frame: 3444372C347971233A168D206FF26DBF235391EBC6C6B708F5B941EAFFB022E23097FCD8D875A086D7B18E954EAB566ADF4B2112EF (53) [with CRC]
[13:06:43][V][mbus:159]: Validating CRC for Block1 + Block2
[13:06:43][I][crc:035]:   calculated: 0x2748, read: 0x12EF  !!!
[13:06:43][D][mbus:056]: Frame: 3244372C347971233A168D206FF26DBF235391EBC6C6B708F5B941EAFFB022E23097FCD8D875A086D7B18E954EAB566ADF4B21 (51) [without CRC]
[13:06:43][D][wmbus:195]: Meter ID [0x23717934] RSSI: -87 dBm LQI: 139 Frame: C1 B not found in configuration T: 3244372C347971233A168D206FF26DBF235391EBC6C6B708F5B941EAFFB022E23097FCD8D875A086D7B18E954EAB566ADF4B21 (51)

[13:07:45][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[13:07:45][D][mbus:022]: Processing C1 B frame
[13:07:45][V][mbus:026]: Frame: 3444372C347971233A168D2073026E3F217DA8A845C21B9F88AA5244D503B12B625855FCAC21EA0E7C5309347C9556BA7CF57CE68B (53) [with CRC]
[13:07:45][V][mbus:159]: Validating CRC for Block1 + Block2
[13:07:45][I][crc:035]:   calculated: 0x9B42, read: 0xE68B  !!!
[13:07:45][D][mbus:056]: Frame: 3244372C347971233A168D2073026E3F217DA8A845C21B9F88AA5244D503B12B625855FCAC21EA0E7C5309347C9556BA7CF57C (51) [without CRC]
[13:07:45][D][wmbus:195]: Meter ID [0x23717934] RSSI: -86 dBm LQI: 137 Frame: C1 B not found in configuration T: 3244372C347971233A168D2073026E3F217DA8A845C21B9F88AA5244D503B12B625855FCAC21EA0E7C5309347C9556BA7CF57C (51)

SzczepanLeon, 2024-02-05:
Please remove " from Meter ID.

evilmumi, 2024-02-05:
changed to
  ## all watermeters wmid: "0", your watermeter: wmid: !secret watermeterId
  ## wmid: "0"
  wmid: 23717934

## ---------------------------------------------------
## SENSORS
## ---------------------------------------------------
sensor:
  - platform: wmbus
    meter_id: $wmid
    mode: T1C1
    type: evo868
    key: "05B9424286A9F419E037D5B84B5CF3B3"
    total_water_m3:
      name: "test meter"

evilmumi, 2024-02-05:
Must be string, got <class 'esphome.helpers.EInt'>. did you forget putting quotes around the value?.
  wmid: 23717934

SzczepanLeon, 2024-02-05:
This is HEX so 0x at the beginning should be.

evilmumi, 2024-02-05:
changed to wmid: "0x23717934"

evilmumi, 2024-02-05:
`[13:17:53][D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[13:17:53][D][mbus:022]: Processing C1 B frame
[13:17:53][V][mbus:026]: Frame: 3444372C347971233A168D209AA26E6E3F213B42131EBD78EC7E29118E58815E0754503644F3FC5876E6D8DD15AA90CB6BCA482A5A (53) [with CRC]
[13:17:53][V][mbus:159]: Validating CRC for Block1 + Block2
[13:17:53][I][crc:035]:   calculated: 0x4EE9, read: 0x2A5A  !!!
[13:17:53][D][mbus:056]: Frame: 3244372C347971233A168D209AA26E6E3F213B42131EBD78EC7E29118E58815E0754503644F3FC5876E6D8DD15AA90CB6BCA48 (51) [without CRC]
[13:17:53][I][wmbus:090]: Using driver 'evo868' for ID [0x23717934] RSSI: -87 dBm LQI: 133 Mode: X T: 3244372C347971233A168D209AA26E6E3F213B42131EBD78EC7E29118E58815E0754503644F3FC5876E6D8DD15AA90CB6BCA48 (51)
[13:17:53][V][utils:109]: (ELL)  CI: 8D  offset: 17
[13:17:53][V][utils:110]: (ELL)  IV: 372C347971233A1620A26E6E3F000000
[13:17:53][V][utils:115]: (ELL) KEY: 05B9424286A9F419E037D5B84B5CF3B3 (16)
[13:17:53][D][utils:124]: (ELL) AES_CTR decrypting: 213B42131EBD78EC7E29118E58815E0754503644F3FC5876E6D8DD15AA90CB6BCA48 (34)
[13:17:53][D][utils:152]: (ELL) AES_CTR  decrypted: B1DBF085A732476A5825D441372F20E3C8D9D639652BE393437CE21B8F375D17467F (34)
[13:17:53][V][wmbus:302]: Validating CRC for ELL payload
[13:17:53][I][crc:051]:   calculated: 0x2295, read: 0xDBB1  !!!
[13:17:53][E][wmbus:115]: Something was not OK during decrypting telegram for ID [0x23717934] 'evo868' key: '05B9424286A9F419E037D5B84B5CF3B3'
[13:17:53][E][wmbus:119]: T : 3244372C347971233A168D209AA26E6E3F213B42131EBD78EC7E29118E58815E0754503644F3FC5876E6D8DD15AA90CB6BCA48 (51)
[13:17:53][E][wmbus:120]: T': 3244372C347971233A168D209AA26E6E3FB1DBF085A732476A5825D441372F20E3C8D9D639652BE393437CE21B8F375D17467F (51)`

SzczepanLeon, 2024-02-05:
It is number.

evilmumi, 2024-02-05:
`
[13:18:09][W][component:214]: Component wmbus took a long time for an operation (0.14 s).
[13:18:09][W][component:215]: Components should block for at most 20-30ms.
[13:18:13][V][sensor:043]: 'Heap Free': Received new state 17144.000000
[13:18:13][D][sensor:093]: 'Heap Free': Sending state 17144.00000 B with 0 decimals of accuracy
[13:18:13][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:13][V][json:058]: Size after shrink 76 bytes
[13:18:13][V][sensor:043]: 'Heap Max Block': Received new state 11184.000000
[13:18:13][D][sensor:093]: 'Heap Max Block': Sending state 11184.00000 B with 0 decimals of accuracy
[13:18:13][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:13][V][json:058]: Size after shrink 80 bytes
[13:18:13][V][sensor:043]: 'Heap Fragmentation': Received new state 9.000000
[13:18:13][D][sensor:093]: 'Heap Fragmentation': Sending state 9.00000 % with 1 decimals of accuracy
[13:18:13][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:13][V][json:058]: Size after shrink 80 bytes
[13:18:13][V][sensor:043]: 'Application Loop Time': Received new state 163.000000
[13:18:13][D][sensor:093]: 'Application Loop Time': Sending state 163.00000 ms with 0 decimals of accuracy
[13:18:13][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:13][V][json:058]: Size after shrink 84 bytes
[13:18:13][W][component:214]: Component debug took a long time for an operation (0.10 s).
[13:18:13][W][component:215]: Components should block for at most 20-30ms.
[13:18:18][V][sensor:043]: 'Heap Free': Received new state 17144.000000
[13:18:18][D][sensor:093]: 'Heap Free': Sending state 17144.00000 B with 0 decimals of accuracy
[13:18:18][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:18][V][json:058]: Size after shrink 76 bytes
[13:18:18][V][sensor:043]: 'Heap Max Block': Received new state 11184.000000
[13:18:18][D][sensor:093]: 'Heap Max Block': Sending state 11184.00000 B with 0 decimals of accuracy
[13:18:18][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:18][V][json:058]: Size after shrink 80 bytes
[13:18:18][V][sensor:043]: 'Heap Fragmentation': Received new state 9.000000
[13:18:18][D][sensor:093]: 'Heap Fragmentation': Sending state 9.00000 % with 1 decimals of accuracy
[13:18:18][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:18][V][json:058]: Size after shrink 80 bytes
[13:18:18][V][sensor:043]: 'Application Loop Time': Received new state 118.000000
[13:18:18][D][sensor:093]: 'Application Loop Time': Sending state 118.00000 ms with 0 decimals of accuracy
[13:18:18][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:18][V][json:058]: Size after shrink 84 bytes
[13:18:18][W][component:214]: Component debug took a long time for an operation (0.10 s).
[13:18:18][W][component:215]: Components should block for at most 20-30ms.
[13:18:20][D][esp8266.preferences:238]: Saving preferences to flash...
[13:18:23][V][sensor:043]: 'Heap Free': Received new state 17144.000000
[13:18:23][D][sensor:093]: 'Heap Free': Sending state 17144.00000 B with 0 decimals of accuracy
[13:18:23][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:23][V][json:058]: Size after shrink 76 bytes
[13:18:23][V][sensor:043]: 'Heap Max Block': Received new state 11184.000000
[13:18:23][D][sensor:093]: 'Heap Max Block': Sending state 11184.00000 B with 0 decimals of accuracy
[13:18:23][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:23][V][json:058]: Size after shrink 80 bytes
[13:18:23][V][sensor:043]: 'Heap Fragmentation': Received new state 9.000000
[13:18:23][D][sensor:093]: 'Heap Fragmentation': Sending state 9.00000 % with 1 decimals of accuracy
[13:18:23][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:23][V][json:058]: Size after shrink 80 bytes
[13:18:23][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[13:18:23][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[13:18:23][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:23][V][json:058]: Size after shrink 84 bytes
[13:18:23][W][component:214]: Component debug took a long time for an operation (0.10 s).
[13:18:23][W][component:215]: Components should block for at most 20-30ms.
[13:18:24][V][sensor:043]: 'Boot counter': Received new state 22.000000
[13:18:24][D][sensor:093]: 'Boot counter': Sending state 22.00000  with 0 decimals of accuracy
[13:18:24][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:24][V][json:058]: Size after shrink 72 bytes
[13:18:24][D][rxLoop:159]: Have 74 bytes from CC1101 Rx
[13:18:24][D][mbus:022]: Processing C1 B frame
[13:18:24][V][mbus:026]: Frame: 4944372C347971233A168D209CB06E3F21E55959232A9B1FB0BFAF2AFC406CC38FF07603BF841B9E0C26356DB8EA43A453553D0F5DE98B3A580B2ACF2DE864046E3D4944372C34797123 (74) [with CRC]
[13:18:24][V][mbus:159]: Validating CRC for Block1 + Block2
[13:18:24][I][crc:035]:   calculated: 0x82A7, read: 0x7123  !!!
[13:18:24][D][mbus:056]: Frame: 4744372C347971233A168D209CB06E3F21E55959232A9B1FB0BFAF2AFC406CC38FF07603BF841B9E0C26356DB8EA43A453553D0F5DE98B3A580B2ACF2DE864046E3D4944372C3479 (72) [without CRC]
[... cut ...]
[13:18:38][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:38][V][json:058]: Size after shrink 84 bytes
[13:18:38][V][sensor:043]: 'Application Loop Time': Received new state 117.000000
[13:18:38][D][sensor:093]: 'Application Loop Time': Sending state 117.00000 ms with 0 decimals of accuracy
[13:18:38][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[13:18:38][V][json:058]: Size after shrink 84 bytes
[13:18:38][W][component:214]: Component debug took a long time for an operation (0.10 s).
`

SzczepanLeon, 2024-02-05:
As you can see below frame C1 B is received, but CRC check is failed (I don't know why - other frames that I have are OK):
[image]

Seems that received frame is corrupted, so next step decrypting is also failing:
[image]

evilmumi, 2024-02-05:
my other solution from https://github.com/pthalin/esp32-multical21 is still working - maybe you can grab there some information in code?

SzczepanLeon, 2024-02-05:
Can you provide me logs from that other solution?

evilmumi, 2024-02-05:
atm i dont know how to get logs - will give it a try

evilmumi, 2024-02-05:
yeah - but i only got `ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8`

and it is not connecting to wlan

SzczepanLeon, 2024-02-05:
Can you decrypt this with propper key:
https://wmbusmeters.org/analyze/4944372C347971233A168D209CB06E3F21E55959232A9B1FB0BFAF2AFC406CC38FF07603BF841B9E0C26356DB8EA43A453553D0F5DE98B3A580B2ACF2DE864046E3D4944372C34797123
??

evilmumi, 2024-02-05:
i found some old stuff: 
[image]

SzczepanLeon, 2024-02-05:
Great - do you have that in txt format?

evilmumi, 2024-02-05:
found old here ` StateOperating:
Payload: 44372c347971233a168d20b260262421855f9737ba8cab271cac23ab626a30f68d1907e                                                  ec34dcb5831dd91766de5fde8e2d55cbe
Data: 6eed790509059a00000000f57803002105065000066000e12b11140000838d030010
calc_crc: 0xed6e
read_crc: 0xed6e
CRC: OK
total: 2029322.2 m▒ - target: 86048.771 m▒ -  6 ▒C - 80 ▒C

StateOperating:
Payload: 44372c347971233a168d20bb82262421366199761da2c0ba71d59e730878273ab495f809203def2520579dd3a8bec879da40a9cd
Data: 6eed790509059a00000000f57803002105065000066000e12b11140000838d030010
calc_crc: 0xed6e
read_crc: 0xed6e
CRC: OK
total: 2029322.2 m▒ - target: 86048.771 m▒ -  6 ▒C - 80 ▒C

StateOperating:
Payload: 44372c347971233a168d201db2702421c794afac403e72f56807d5051aa039e276be0cb3b825381237dfe80b46a2e4c0eb759b71
Data: 4a477905090d5c00000000f57803002105055000056000e12b111400004d8e030010
calc_crc: 0x474a
read_crc: 0x474a
CRC: OK
total: 2029322.2 m▒ - target: 86048.771 m▒ -  5 ▒C - 80 ▒C

StateConnected:
MQTT-SEND: watermeter/0/online True
MQTT-SEND: watermeter/0/ipaddr 192.168.0.181
StateOperating:
Payload: 44372c347971233a168d20311071242196508919b2b811f9c911fb84f7a696c291272a6eaa1a9e39b68df4a26fb84a7138eebc36
Data: 4a477905090d5c00000000f57803002105055000056000e12b111400004d8e030010
calc_crc: 0x474a
read_crc: 0x474a
CRC: OK
total: 2029322.2 m▒ - target: 86048.771 m▒ -  5 ▒C - 80 ▒C
`

evilmumi, 2024-02-05:
> Can you decrypt this with propper key: https://wmbusmeters.org/analyze/4944372C347971233A168D209CB06E3F21E55959232A9B1FB0BFAF2AFC406CC38FF07603BF841B9E0C26356DB8EA43A453553D0F5DE98B3A580B2ACF2DE864046E3D4944372C34797123 ??

no failed decryption. Wrong key?

evilmumi, 2024-02-05:
from now: `StateOperating:
Payload: 44372c347971233a168d2030b2793f21430559765f701639ab46f8b02fa7a14a7624a4170ebd7c3298658b383dcf34f98ef301cb
Data: e7947905095875000000007ce203007c0506600006d00001320b1100001de703000f
calc_crc: 0x94e7
read_crc: 0x94e7
CRC: OK
total: 255.773 m▒ - target: 92012.547 m▒ -  6 ▒C - 96 ▒C
`

only total is correct - rest of values are wrong

SzczepanLeon, 2024-02-05:
https://wmbusmeters.org/analyze/4744372c347971233a168d209cB06e3f21e55959232a9B1fB0Bfaf2afc406cc38ff07603Bf841B9e0c26356dB8ea43a453553d0f5de98B3a580B2acf2de864046e3d4944372c3479

https://wmbusmeters.org/analyze/4944372C347971233A168D209CB06E3F21E55959232A9B1FB0BFAF2AFC406CC38FF07603BF841B9E0C26356DB8EA43A453553D0F5DE98B3A580B2ACF2DE864046E3D4944372C34797123

but with key that you have - is it OK?

evilmumi, 2024-02-05:
> https://wmbusmeters.org/analyze/4744372c347971233a168d209cB06e3f21e55959232a9B1fB0Bfaf2afc406cc38ff07603Bf841B9e0c26356dB8ea43a453553d0f5de98B3a580B2acf2de864046e3d4944372c3479 [...]

both links failed decryption. Wrong key?

evilmumi, 2024-02-05:
one guy told me to change source code of his to get it work with me:

`void WMBusFrame::printMeterInfo(uint8_t *data, size_t len)
{
    // init positions for compact frame
  // int pos_tt = 9; // total consumption, 9, 10, 11, 12    his
  int pos_tt = 29;`

and
`  if(data[2] == 0x79)  //compact frame
  {
    pos_tt = 29;
    pos_tg = 13;
    //pos_ic = 7;
    pos_ft = 17;
    pos_at = 18;
  }`

in https://github.com/pthalin/esp32-multical21/blob/master/src/WMBusFrame.cpp

not sure if helpful or nothing do to with encrypting

SzczepanLeon, 2024-02-05:
And frame:
https://wmbusmeters.org/analyze/3444372c347971233a168d2030b2793f21430559765f701639ab46f8b02fa7a14a7624a4170ebd7c3298658b383dcf34f98ef301cb

is OK, and can be decrypted?

evilmumi, 2024-02-05:
now grabbed: 

`Payload: 44372c347971233a168d20b9f27b3f2164fa16c9ff63f85cb8b050296dffcb6f858f928afd1daa57a8ea16fbb6d79695f80852c1
Data: e7947905095875000000007ce203007c0506600006d00001320b1100001de703000f
calc_crc: 0x94e7
read_crc: 0x94e7
CRC: OK
total: 255.773 m▒ - target: 92012.547 m▒ -  6 ▒C - 96 ▒C

Payload: 44372c347971233a168d20bd027c3f21352dc34eb25afe9131557e7e9a2a2c0417648a48eef85e90973c5a1806f4c3a2394219be
Data: e7947905095875000000007ce203007c0506600006d00001320b1100001de703000f
calc_crc: 0x94e7
read_crc: 0x94e7
CRC: OK
total: 255.773 m▒ - target: 92012.547 m▒ -  6 ▒C - 96 ▒C

`

SzczepanLeon, 2024-02-05:
It is strange for me.
For sure CRC calculation in my code is OK (both CRC, for frame and for payload).
There is something strange in Rx from CC1101. T1 A mode is working, C1 A mode also should be OK (I don't have that meter but I have feedback).

Only that C1 B mode is not OK - but logic is the same as for C1 A (almost).
Maybe that almost is key to my riddle :-(

evilmumi, 2024-02-05:
> It is strange for me. For sure CRC calculation in my code is OK (both CRC, for frame and for payload). There is something strange in Rx from CC1101. T1 A mode is working, C1 A mode also should be OK (I don't have that meter but I have feedback).
> 
> Only that C1 B mode is not OK - but logic is the same as for C1 A (almost). Maybe that almost is key to my riddle :-(
> 
> btw You run my soft and that other on the same hardware?

atm yours on an 8266 d1 mini and the other on an esp32dev ;az-delivery-devkit-v4

do you think - i should test yours on esp32 also?

SzczepanLeon, 2024-02-05:
One week ago I would say __it is does not matters__, but today - yes please test it on the same HW.

evilmumi, 2024-02-05:
so i took also esp32 and same antenna:

`[D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 3444372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFBFBF1 (53) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x31A6, read: 0xFBF1  !!!
[D][mbus:056]: Frame: 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -65 dBm LQI: 132 Frame: C1 B T: 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51)
[V][utils:109]: (ELL)  CI: 8D  offset: 17
[V][utils:112]: (ELL)  IV: 372C347971233A162003843F21000000
[V][utils:115]: (ELL) KEY: keykeykey(16)
[D][utils:124]: (ELL) AES_CTR decrypting: A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (34)
[D][utils:152]: (ELL) AES_CTR  decrypted: FDC37905099CE6000000007CE203007C0506600006D00001320B11000079E703000F (34)
[V][wmbus:303]: Validating CRC for ELL payload
[I][crc:047]:   calculated: 0xC3FD, read: 0xC3FD
[E][wmbus:119]: Something was not OK during decrypting telegram for ID [0x23717934] 'evo868' key: '05B9424286A9F419E037D5B84B5CF3B3'
[E][wmbus:120]: T : 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51)
[E][wmbus:121]: T': 3244372C347971233A168D209D03843F21FDC37905099CE6000000007CE203007C0506600006D00001320B11000079E703000F (51)`

`[D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 3444372C377971233A168D20B3D1843F21AA77373ED72D2AFC0CAA521E667DE53753C538E2B1BCE2733B418889EABECDAC4A6464F1 (53) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x283C, read: 0x64F1  !!!
[D][mbus:056]: Frame: 3244372C377971233A168D20B3D1843F21AA77373ED72D2AFC0CAA521E667DE53753C538E2B1BCE2733B418889EABECDAC4A64 (51) [without CRC]
[D][wmbus:202]: Meter ID [0x23717937] RSSI: -88 dBm LQI: 136 Frame: C1 B not found in configuration T: 3244372C377971233A168D20B3D1843F21AA77373ED72D2AFC0CAA521E667DE53753C538E2B1BCE2733B418889EABECDAC4A64 (51)`

`[D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 3444372C347971233A168D209F11843F211F09E5F8CAA5DE0B9879D41726086FA1BA4597D4419A403816E22668B9124A32942F2FF1 (53) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0xFF8E, read: 0x2FF1  !!!
[D][mbus:056]: Frame: 3244372C347971233A168D209F11843F211F09E5F8CAA5DE0B9879D41726086FA1BA4597D4419A403816E22668B9124A32942F (51) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -66 dBm LQI: 129 Frame: C1 B T: 3244372C347971233A168D209F11843F211F09E5F8CAA5DE0B9879D41726086FA1BA4597D4419A403816E22668B9124A32942F (51)
[V][utils:109]: (ELL)  CI: 8D  offset: 17
[V][utils:112]: (ELL)  IV: 372C347971233A162011843F21000000
[V][utils:115]: (ELL) KEY: keykeky(16)
[D][utils:124]: (ELL) AES_CTR decrypting: 1F09E5F8CAA5DE0B9879D41726086FA1BA4597D4419A403816E22668B9124A32942F (34)
[D][utils:152]: (ELL) AES_CTR  decrypted: FDC37905099CE6000000007CE203007C0506600006D00001320B11000079E703000F (34)
[V][wmbus:303]: Validating CRC for ELL payload
[I][crc:047]:   calculated: 0xC3FD, read: 0xC3FD
[E][wmbus:119]: Something was not OK during decrypting telegram for ID [0x23717934] 'evo868' key: 'keykey'
[E][wmbus:120]: T : 3244372C347971233A168D209F11843F211F09E5F8CAA5DE0B9879D41726086FA1BA4597D4419A403816E22668B9124A32942F (51)
[E][wmbus:121]: T': 3244372C347971233A168D209F11843F21FDC37905099CE6000000007CE203007C0506600006D00001320B11000079E703000F (51)`

`[D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 3444372C377971233A168D20B5D3843F212519285C5745E744DA2C8513E0AFC05EB2DDF22DE86E493685F9A932374088AFC8F7F7F1 (53) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x1BFD, read: 0xF7F1  !!!
[D][mbus:056]: Frame: 3244372C377971233A168D20B5D3843F212519285C5745E744DA2C8513E0AFC05EB2DDF22DE86E493685F9A932374088AFC8F7 (51) [without CRC]
[D][wmbus:202]: Meter ID [0x23717937] RSSI: -87 dBm LQI: 131 Frame: C1 B not found in configuration T: 3244372C377971233A168D20B5D3843F212519285C5745E744DA2C8513E0AFC05EB2DDF22DE86E493685F9A932374088AFC8F7 (51)
`

evilmumi, 2024-02-05:
with this yaml:

`## ------------------------------------------------------------------
## WMBUS METER D1 MINI + CC1101 for Diehl IZAR RC 868 I R4 PL
## HARDWARE: D1 MINI ESP8266 80MHz, 80KB RAM, 4MB Flash
##
## CC1101 TESTCASE FOR ALL WATERMETERS
## DO NOT USE THIS CONFIGURATION FOR PRODUCTION
##
##   1. uses esphome-components latest version from github
##   2. uses syslog component  (optional)
##      tail -f /var/log/syslog | grep "water-meter"
##
## This version shows data from the all watermeters.
## Use this for frist test to find your watermeterID.
##
## ------------------------------------------------------------------
substitutions:

  ## device settings
  device_name_short: "water-meter-esp32"
  device_description: "TEST: SzczepanLeon - find watermeters"
  projectname: "Diehl IZAR RC 868.Watermeter"
  appversion: "2.0.8"

  ## all watermeters wmid: "0", your watermeter: wmid: !secret watermeterId
  ## wmid: "0"
  wmid: "0x23717934"

  ## logger settings, use DEBUG or VERBOSE to see the telegram data
  log_level: "VERBOSE"    # In production mode use "WARN"
  log_wmbus: "VERBOSE"    # Loglevel for wmbus meters component
  log_baudrate: "115200"     # Disable uart logger messages log baudrate=0

## ----------------------------------------------------------------
## HARDWARE: WEMOS D1 MINI ESP8266 80MHz, 80KB RAM, 4MB Flash
## ----------------------------------------------------------------

# wemmos D1 Mini  = d1_mini
# D1 mit externer Antenne = d1_mini_pro
# esp01 mit 1MB Speicher = esp01_1m

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino  

preferences:
  flash_write_interval: 1min

## ---------------------------------------------------
## ESPHOME Configuration
## ---------------------------------------------------
esphome:
  name: ${device_name_short}
  comment: ${device_description}
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false
  project:
    name: ${projectname}
    version: ${appversion}
  build_path: ./build/${device_name_short}
  on_boot:
    - priority: -100.00
      then:
        - globals.set:
            id: boot_counter
            value: !lambda "return id(boot_counter)+=1;"
        - logger.log:
[... 312 lines cut ...]
    icon: mdi:lock-reset
    name: Restart factory default

## ---------------------------------------------------
## TEXT SENSORS
## ---------------------------------------------------
text_sensor:
  - platform: debug
    # Reports the device information:
    device:
      name: "Device Info"
      entity_category: "diagnostic"
    # Reports the last reboot reason in a human-readable form.
    reset_reason:
      name: "Reset Reason"
      entity_category: "diagnostic"

  # Last boot timestamp
  - platform: template
    name: "Last Boot timestamp"
    disabled_by_default: true
    id: device_lastBoot_time
    icon: mdi:clock-start`

from here https://github.com/zibous/ha-watermeter/tree/master

SzczepanLeon, 2024-02-05:
Strange ;-)
Frames from "other sw" have frame CRC OK, and for frames from my sw frame CRC is not OK:
```
[D][rxLoop:159]: Have 53 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 3444372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFBFBF1 (53) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x31A6, read: 0xFBF1  !!!
[D][mbus:056]: Frame: 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -65 dBm LQI: 132 Frame: C1 B T: 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51)
[V][utils:109]: (ELL)  CI: 8D  offset: 17
[V][utils:112]: (ELL)  IV: 372C347971233A162003843F21000000
[V][utils:115]: (ELL) KEY: keykeykey(16)
[D][utils:124]: (ELL) AES_CTR decrypting: A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (34)
[D][utils:152]: (ELL) AES_CTR  decrypted: FDC37905099CE6000000007CE203007C0506600006D00001320B11000079E703000F (34)
[V][wmbus:303]: Validating CRC for ELL payload
[I][crc:047]:   calculated: 0xC3FD, read: 0xC3FD
[E][wmbus:119]: Something was not OK during decrypting telegram for ID [0x23717934] 'evo868' key: 'kexkkeykey'
[E][wmbus:120]: T : 3244372C347971233A168D209D03843F21A6C0F65FA962803F1FF7D3F89A54E701D33740F84D5C09DB2B6D486D49C4C626CAFB (51)
[E][wmbus:121]: T': 324
```

evilmumi, 2024-02-05:
now i got:
```
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 4944372C347971233A168D206CC28B3F21A687CAA1488C8F6694BE3E87377D3A09C57F21DA53E9886A9CC48CD32314A510E43E0688EC3DC27FD095512EBC6114EE2D4C2125B11C22223A (74) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0xDFFF, read: 0x223A  !!!
[D][mbus:056]: Frame: 4744372C347971233A168D206CC28B3F21A687CAA1488C8F6694BE3E87377D3A09C57F21DA53E9886A9CC48CD32314A510E43E0688EC3DC27FD095512EBC6114EE2D4C2125B11C22 (72) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -86 dBm LQI: 130 Frame: C1 B T: 4744372C347971233A168D206CC28B3F21A687CAA1488C8F6694BE3E87377D3A09C57F21DA53E9886A9CC48CD32314A510E43E0688EC3DC27FD095512EBC6114EE2D4C2125B11C22 (72)
```
https://wmbusmeters.org/analyze/4744372C347971233A168D206CC28B3F21A687CAA1488C8F6694BE3E87377D3A09C57F21DA53E9886A9CC48CD32314A510E43E0688EC3DC27FD095512EBC6114EE2D4C2125B11C22

`Auto driver  : flowiq2200
Best driver  : flowiq2200 20/27
Using driver : flowiq2200 00/00
000   : 47 length (71 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 372c dll-mfct (KAW)
004   : 34797123 dll-id (23717934)
008   : 3a dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 6c ell-acc
013   : c28b3f21 sn (AES_CTR)
017   : 86e2 payload crc (calculated 86e2 OK)
019   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : FF vif (Manufacturer specific)
022   : 23 vife (?)
023 C!: 00000000 ("status":"OK")
027   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
028   : 13 vif (Volume l)
029 C!: 7CE20300 ("target_m3":254.588)
033   : 52 dif (16 Bit Integer/Binary Maximum value storagenr=1)
034   : 3B vif (Volume flow l/h)
035 C!: 7C05 ("max_flow_m3h":1.404)
037   : 06 dif (48 Bit Integer/Binary Instantaneous value)
038   : FF vif (Manufacturer specific)
039   : 1B vife (?)
040 C?: 06600006D000
046   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
047   : 6C vif (Date type G)
048 C!: 0132 ("target_date":"2024-02-01")
050   : 61 dif (8 Bit Integer/Binary Minimum value storagenr=1)
051   : 67 vif (External temperature °C)
052 C!: 0B ("min_external_temperature_c":11)
053   : 51 dif (8 Bit Integer/Binary Maximum value storagenr=1)
054   : 67 vif (External temperature °C)
055 C!: 11 ("max_external_temperature_c":17)
056   : 02 dif (16 Bit Integer/Binary Instantaneous value)
057   : 3B vif (Volume flow l/h)
058 C!: 0000 ("flow_m3h":0)
060   : 04 dif (32 Bit Integer/Binary Instantaneous value)
061   : 13 vif (Volume l)
062 C!: B6E70300 ("total_m3":255.926)
066   : 81 dif (8 Bit Integer/Binary Instantaneous value)
067   : 01 dife (subunit=0 tariff=0 storagenr=2)
068   : E7 vif (External temperature °C)
069   : FF combinable vif (MfctSpecific)
070   : 0F combinable extension vife
071 C?: 0F

{
    "media":"cold water",
    "meter":"flowiq2200",
    "name":"",
    "id":"23717934",
    "flow_m3h":0,
    "max_external_temperature_c":17,
    "max_flow_m3h":1.404,
    "min_external_temperature_c":11,
    "target_m3":254.588,
    "total_m3":255.926,
    "status":"OK",
    "target_date":"2024-02-01",
    "timestamp":"2024-02-05T20:04:57Z"
}
`

evilmumi, 2024-02-05:
so long(74) are "sometimes" working - short (53) not my conclusion
but only without CRC
```

[D][rxLoop:159]: Have 74 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 4944372C347971233A168D207C038C3F2144DA6679CD7F37A3528D4B6E82A1081D130A8B36CF6C17BFFC72AFE93287C46E7D34077534024AD808E8FE1F115E3557A9B1F2F2B4A369693A (74) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x2EB2, read: 0x693A  !!!
[D][mbus:056]: Frame: 4744372C347971233A168D207C038C3F2144DA6679CD7F37A3528D4B6E82A1081D130A8B36CF6C17BFFC72AFE93287C46E7D34077534024AD808E8FE1F115E3557A9B1F2F2B4A369 (72) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -85 dBm LQI: 129 Frame: C1 B T: 4744372C347971233A168D207C038C3F2144DA6679CD7F37A3528D4B6E82A1081D130A8B36CF6C17BFFC72AFE93287C46E7D34077534024AD808E8FE1F115E3557A9B1F2F2B4A369 (72)
```

evilmumi, 2024-02-05:
sometimes even long is not decryptable:
4744372c347971233a168d2094708c3f01316cBB1860B223a35ae42eB5c8289f67a665BB3508c79fBce99ddc4a9c7da3698d18328254d994e5c6B15c6486d7dc854de934Ba5e618d

AND value is not getting into homeassisant perhaps - why?

`[D][rxLoop:159]: Have 74 bytes from CC1101 Rx
[D][mbus:022]: Processing C1 B frame
[V][mbus:026]: Frame: 4944372C347971233A168D20D4808D3F21EB3F13F7A008346D102F4DE1CD3165BBF90D3BCCFBD96DF5D542F3FA22A2748786C3EFB9C15D3674B1A2CB70473364D84519C301E4591E1E3A (74) [with CRC]
[V][mbus:159]: Validating CRC for Block1 + Block2
[E][crc:035]:   calculated: 0x57B4, read: 0x1E3A  !!!
[D][mbus:056]: Frame: 4744372C347971233A168D20D4808D3F21EB3F13F7A008346D102F4DE1CD3165BBF90D3BCCFBD96DF5D542F3FA22A2748786C3EFB9C15D3674B1A2CB70473364D84519C301E4591E (72) [without CRC]
[I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -87 dBm LQI: 131 Frame: C1 B T: 4744372C347971233A168D20D4808D3F21EB3F13F7A008346D102F4DE1CD3165BBF90D3BCCFBD96DF5D542F3FA22A2748786C3EFB9C15D3674B1A2CB70473364D84519C301E4591E (72)
[V][utils:109]: (ELL)  CI: 8D  offset: 17
[V][utils:112]: (ELL)  IV: 372C347971233A1620808D3F21000000`

`Auto driver  : flowiq2200
Best driver  : flowiq2200 20/27
Using driver : flowiq2200 00/00
000   : 47 length (71 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 372c dll-mfct (KAW)
004   : 34797123 dll-id (23717934)
008   : 3a dll-version
009   : 16 dll-type (Cold water meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : d4 ell-acc
013   : 808d3f21 sn (AES_CTR)
017   : 86e2 payload crc (calculated 86e2 OK)
019   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
020   : 04 dif (32 Bit Integer/Binary Instantaneous value)
021   : FF vif (Manufacturer specific)
022   : 23 vife (?)
023 C!: 00000000 ("status":"OK")
027   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
028   : 13 vif (Volume l)
029 C!: 7CE20300 ("target_m3":254.588)
033   : 52 dif (16 Bit Integer/Binary Maximum value storagenr=1)
034   : 3B vif (Volume flow l/h)
035 C!: 7C05 ("max_flow_m3h":1.404)
037   : 06 dif (48 Bit Integer/Binary Instantaneous value)
038   : FF vif (Manufacturer specific)
039   : 1B vife (?)
040 C?: 06600006D000
046   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
047   : 6C vif (Date type G)
048 C!: 0132 ("target_date":"2024-02-01")
050   : 61 dif (8 Bit Integer/Binary Minimum value storagenr=1)
051   : 67 vif (External temperature °C)
052 C!: 0B ("min_external_temperature_c":11)
053   : 51 dif (8 Bit Integer/Binary Maximum value storagenr=1)
054   : 67 vif (External temperature °C)
055 C!: 11 ("max_external_temperature_c":17)
056   : 02 dif (16 Bit Integer/Binary Instantaneous value)
057   : 3B vif (Volume flow l/h)
058 C!: 0000 ("flow_m3h":0)
060   : 04 dif (32 Bit Integer/Binary Instantaneous value)
061   : 13 vif (Volume l)
062 C!: B6E70300 ("total_m3":255.926)
066   : 81 dif (8 Bit Integer/Binary Instantaneous value)
067   : 01 dife (subunit=0 tariff=0 storagenr=2)
068   : E7 vif (External temperature °C)
069   : FF combinable vif (MfctSpecific)
070   : 0F combinable extension vife
071 C?: 0F

{
    "media":"cold water",
    "meter":"flowiq2200",
    "name":"",
    "id":"23717934",
    "flow_m3h":0,
    "max_external_temperature_c":17,
    "max_flow_m3h":1.404,
    "min_external_temperature_c":11,
    "target_m3":254.588,
    "total_m3":255.926,
    "status":"OK",
    "target_date":"2024-02-01",
    "timestamp":"2024-02-05T20:32:16Z"
}`

can you explain your question with CRC and ELL :)

SzczepanLeon, 2024-02-05:
> can you explain your question with CRC and ELL :)

You mentioned that sometimes it is OK. So I'm curious it is related to correct/wrong CRC in ELL calculation.

Anyway, **please try next version**. It is with small changes how I count bytes to receive in CC1101.

evilmumi, 2024-02-06:
looks like a good thing, or?

`[07:20:18][E][rxLoop:157]: Length problem: req(76) != rx(78)
[07:20:18][D][rxLoop:159]: Have 78 bytes from CC1101 Rx
[07:20:18][D][mbus:024]: Processing C1 B frame
[07:20:18][V][mbus:028]: Frame: 4944372C347971233A168D207450B23F215977E7024D5C84E3F522DAC84830F762589A46414BAA83F6C0C82A334FF4C9EC07A1D4AAFD6ADF88316D460C92903FD8221DEDFE70D55F7D1C1C8D (76) [with CRC]
[07:20:18][V][mbus:161]: Validating CRC for Block1 + Block2
[07:20:18][I][crc:031]:   calculated: 0x7D1C, read: 0x7D1C
[07:20:18][D][mbus:058]: Frame: 4744372C347971233A168D207450B23F215977E7024D5C84E3F522DAC84830F762589A46414BAA83F6C0C82A334FF4C9EC07A1D4AAFD6ADF88316D460C92903FD8221DEDFE70D55F (72) [without CRC]
[07:20:18][I][wmbus:097]: Using driver 'evo868' for ID [0x23717934] RSSI: -69 dBm LQI: 130 Frame: C1 B T: 4744372C347971233A168D207450B23F215977E7024D5C84E3F522DAC84830F762589A46414BAA83F6C0C82A334FF4C9EC07A1D4AAFD6ADF88316D460C92903FD8221DEDFE70D55F (72)
[07:20:18][V][utils:109]: (ELL)  CI: 8D  offset: 17
[07:20:18][V][utils:112]: (ELL)  IV: 372C347971233A162050B23F21000000
[07:20:18][V][utils:115]: (ELL) KEY: keykeykkey(16)
[07:20:18][D][utils:124]: (ELL) AES_CTR decrypting: 5977E7024D5C84E3F522DAC84830F762589A46414BAA83F6C0C82A334FF4C9EC07A1D4AAFD6ADF88316D460C92903FD8221DEDFE70D55F (55)
[07:20:18][D][utils:152]: (ELL) AES_CTR  decrypted: 23907804FF230000000044137CE20300523B7C0506FF1B06700006D000426C013261670B516711023B00000413DFE703008101E7FF0F0E (55)
[07:20:18][V][wmbus:303]: Validating CRC for ELL payload
[07:20:18][I][crc:047]:   calculated: 0x9023, read: 0x9023
[07:20:19][D][wmbus:104]: Decrypted T : 4744372C347971233A168D207450B23F2123907804FF230000000044137CE20300523B7C0506FF1B06700006D000426C013261670B516711023B00000413DFE703008101E7FF0F0E (72)
[07:20:19][I][wmbus:743]: Reset current value to: 0.000, Waterdisplay value: 255.967, last value: 0.000
[07:20:19][D][wmbus:757]: Diff millisecond is: 29824
[07:20:19][D][wmbus:535]: cc1101 state message: Warte auf Daten, error code: 1
[07:20:30][D][wmbus:535]: cc1101 state message: Warte auf Daten, error code: 1`

SzczepanLeon, 2024-02-06:
Yes, I see progres.

btw
What fields from telegram do you need to have?

For sure total water m3.

evilmumi, 2024-02-06:
Great. Thanks for your doing!

fields. i am not sure whats possible.
possible fields are these listed below?
i think this is last value of last minute in last month? target_m3
total_m3 -> yes
status is the leckage? would be good
i dont think this is needed for me:
"flow_m3h":0,
"max_external_temperature_c":17,
"max_flow_m3h":1.404,
"min_external_temperature_c":11,

i am not sure whats possible 

https://documentation.kamstrup.com/docs/flowIQ_2200_EU/en-GB/Data_sheet/CONT07BEC5F79AF846B2AD153D44DEEBA501/

[17 more replies]