SzczepanLeon/esphome-components #63: Support for TTGO T3 V1.6

kubert96, 2023-12-20:
Hello, 

its posible to get this working on TTGO T3 V1.6? 
[image]

SzczepanLeon, 2023-12-20:
No. Currently only CC1101 is supported.

But it should be possible:
https://github.com/xaelsouth/Semtech-wMBusReceiver

kubert96, 2023-12-20:
I know that curently onlu CC1101 :D 

Can you guys implement TTGO? :D 

Thanks :) 

I ordered some CC1101 but had TTGO laing around for some time and this would be perfect usage :D I coud do the screan imlementation. 👍

SzczepanLeon, 2023-12-20:
> Can you guys implement TTGO? :D

I don't have that TTGO on my desk :-(

kubert96, 2023-12-20:
Yeah no problem, can test ist any time :)

closed 2024-02-02 by SzczepanLeon