SzczepanLeon/esphome-components #66: driver fhkvdataiii  read temperature

jawilczek, 2024-01-06:
Hi

Is it possible to supplement the "fhkvdataiii" driver with the ability to read the room temperature?
frame analysis shows that such data is there.

telegram=|33446850405045109480A2_0FFF2E000060020000918D0873080000000000000000000000000000000000000000000000000000000|

Auto driver : fhkvdataiii
Best driver: fhkvdataiii 12/53
Using driver: fhkvdataiii 00/00
000 : 33 length (51 bytes)
001 : 44 dll-c (from meter SND_NR)
002 : 6850 dll-mfct (TCH)
004 : 40504510 dll-id (10455040)
008 : 94 dll-version
009 : 80 dll-type (Heat Cost Allocator)
010 : a2 tpl-ci-field (Mfct specific)
011 C?: 0FFF2E000060020000918D08730800000000000000000000000000000000000000000000000000000 mfct specific
012 C!: *** ff2e previous_date = 2023-07-31T02:00:00Z
014 C!: *** 0000 previous_hca = 0
016 C!: *** 6002 current_date = 2024-01-06T02:00:00Z
018 C!: *** 0000 current_hca = 0
021 C!: *** 8d08 temp_room_c = 21.890000000000001
023 C!: *** 7308 temp_radiator_c = 21.629999999999999

{
     "media":"heat cost allocator",
     "meter":"fhkvdataiii",
     "name":"",
     "id":"10455040",
     "current_hca":0,
     "previous_hca":0,
     "temp_radiator_c":21.63,
     **"temp_room_c":21.89,**
     "current_date":"2024-01-06T02:00:00Z",
     "previous_date":"2023-07-31T02:00:00Z",
     "timestamp":"2024-01-06T12:33:34Z"
}

--
Jacek

SzczepanLeon, 2024-01-06:
Please try version **2.4.10**

jawilczek, 2024-01-07:
Thank you very much, it works as it should :)

[06:18:55][I][wmbus:087]: Using driver 'fhkvdataiii' for ID [0x10360161] RSSI: -41 dBm LQI: 128 Mode: T1 T: 33446850610136109480A20FFF2E0000700200001132082B08000000000000000000000000000000000000000000000000000000 (52)
[06:18:55][D][sensor:094]: 'BlueRoom C.O LQI': Sending state 128.00000  with 0 decimals of accuracy
[06:18:55][D][sensor:094]: 'BlueRoom C.O RSSI': Sending state -41.00000 dBm with 0 decimals of accuracy
[06:18:55][D][sensor:094]: 'BlueRoom C.O aktualnie': Sending state 0.00000  with 0 decimals of accuracy
[06:18:55][D][sensor:094]: 'BlueRoom C.O poprzednio': Sending state 0.00000  with 0 decimals of accuracy
[06:18:55][D][sensor:094]: 'BlueRoom C.O temp.': Sending state 20.98000 °C with 2 decimals of accuracy

--
Jacek

closed 2024-01-07 by SzczepanLeon