SzczepanLeon/esphome-components #67: No telegrams are received

geeuz, 2024-01-08:
Hello,

I am quite new to the subject and am hoping for your help because I am getting desperate. 

I want to try to read out my Diehl Hydrus with an ESP32 via Esphome and CC1101, but I don't see any incoming telegrams.
I am sure that I have connected the CC1101 correctly, and I have already tested a second CC1101 to rule out a defective module.
I don't yet know whether I have the correct AES key, but even if I don't, telegrams should still be arriving if I have set "log_unknown: True", right? The telegrams should appear in the ESP32 log, correct?

In general, it looks a bit different to me than in the instructions, I think this is due to the newer versions. Perhaps I don't understand it and some component of wmbusmeters is still missing completely or is configured incorrectly.

I have installed the Wmbusmeters HA plugin in version 1.14.0, in the Plugin Configuration I've set UDP Port to 9011.

This is my wmbusmeters configuration:

[image]
[image]

And this is the configuration on my ESP32 in Esphome (meter_id and Key changed):

```
esphome:
  name: "espgarten01"
  friendly_name: "espgarten01"

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [wmbus]

logger:
  level: VERBOSE

captive_portal:

time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Vienna
    servers:
      - 0.at.pool.ntp.org
      - 0.pool.ntp.org
      - 1.pool.ntp.org

    on_time_sync:
      then:
        - logger.log:
            level: WARN
            tag: "system"
            format: "Synchronized sntp clock"
			
wmbus:
  clients:
    - name: "wmbusmeters_udp"
      ip_address: "192.168.1.54"
      port: 9011
      format: RTLWMBUS
      transport: UDP
  mosi_pin: GPIO23  # MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19  # MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18   # SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05    # CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16  # GDO0 Clock output. High Impedance !
  gdo2_pin: GPIO17  # GDO2 status signals. High Impedance !
  log_unknown: True
  
sensor:
  - platform: wmbus
    meter_id: 0x43A12XY
    type: hydrus
    key: "f851728910e66d83f851728910e66d83"
    add_prefix: false
    lqi:
      id: wmbus_cc1101_lqi
      name: "Watermeter CC1101 LQI"
      entity_category: "diagnostic"
      unit_of_measurement: "lqi"
      state_class: "measurement"
    rssi:
      id: wmbus_cc1101_rssi
      name: "Watermeter CC1101 RSSI"
      unit_of_measurement: "dbm"
      entity_category: "diagnostic"
    total_water_m3:
      name: "Watermeter Display"
      id: "waterdisplay"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

In Homeassistant I have created the following sensor in my mqtt.yaml (which is included in the configuration.yaml), but I don't know if this is correct or even necessary if I have already configured the sensor on the ESP32:

```
sensor:
  - name: "water_from_hydrus"
    unique_id: water_from_hydrus_total_m3
    object_id: water_from_hydrus_total_m3
    state_topic: "wmbusmeters/water_from_hydrus"
    value_template: "{{ value_json['total_m3']|default(0.000) }}"
    unit_of_measurement: m³
    device_class: water
    state_class: total_increasing
    icon: mdi:water
```

I would be very grateful for your help.

SzczepanLeon, 2024-01-08:
1) remove all sensors from config
2) collect VERY_VERBOSE or VERBOSE traces

geeuz, 2024-01-08:
Ok, I've removed the sensors, but still no telegram incoming with VERY_VERBOSE on the Esp32.
I've tried on two different ESPs with different CC1101 modules.
I have let it run in the log for considerably longer than these 10 minutes in the log, but nothing comes up.

[logs_espgartentest01_run.txt](https://github.com/SzczepanLeon/esphome-components/files/13863155/logs_espgartentest01_run.txt)

SzczepanLeon, 2024-01-08:
Maybe You are too far from watermeter?

geeuz, 2024-01-08:
Maybe, I will test tomorrow closer to the watermeter.
Both CC1101 are 868 modules.

SzczepanLeon, 2024-01-08:
Maybe You are too far from watermeter? SPI wiring seems to be OK, but there is no check for GDOx. You can try to reverse it.

SzczepanLeon, 2024-01-08:
> Do I understand correctly that I don't need wmbusmeters if I only want the data in Homeassistant from the configured sensor on the Esp32?

Yes it is correct, but not all meters are supported.

geeuz, 2024-01-08:
Yes I'm sure, it's a Diehl Type 171A, 868, OMS

I'm just wondering why I'm not getting any telegrams at all, not even from other meters, as the range should be quite large at 868MHz.
My meter is about 10 meters away in the ground shaft.

ondras76, 2024-01-08:
is there a wall in the way? my meter is in a concrete hole with an iron cover. I have to have the wmbus meter right next to the water meter, otherwise I don't have any telegram.

geeuz, 2024-01-08:
Yes, also in a concrete hole with an iron cover. Ok, I will test it directly next to the meter. As this is in the allotment I am not on site, I will be there in the next 2-3 days. In the meantime, many thanks to you both!

geeuz, 2024-01-09:
Many thanks to you, I have now received telegrams, it was indeed the distance. I was sure that the range was much greater, but I guess nothing gets through the iron cover.

The telegrams received from my Hydrus are unencrypted, so now I can just get on with setting everything up.

Many thanks again to you both!

ondras76, 2024-01-09:
good to hear it. I had the same with my meter.

closed 2024-01-09 by SzczepanLeon