SzczepanLeon/esphome-components #68: Support for Kamstrup Multical 21 watermeter

lukasz-burak, 2024-01-18:
Hey,
I'd like to request support for Kamstrup Multical 21 watermeter.
It's being decoded in wmbusmeters just fine (https://wmbusmeters.org/analyze/2a442d2c998734761B168d2091d37cac21e1d68cdaffcd3dc452Bd802913ff7B1706ca9e355d6c2701cc24:multical21:28F64A24988064A079AA2C807D6102AE).

Can you implement such support?

SzczepanLeon, 2024-02-02:
Please try version _pre_ 3:

```
external_components:
  - source: github://SzczepanLeon/esphome-components@pre_v3.0
    refresh: 0d
    components: [ wmbus ]
```

For now only fields:

  - total_m3
  - target_m3
  
are supported.

**Please collect VERBOSE logs.**

SzczepanLeon, 2024-02-06:
Driver available in version 3.0.0

closed 2024-02-06 by SzczepanLeon