SzczepanLeon/esphome-components #69: Support for Apator Elf 2 (microclima) HeatMeter

sswierczyna, 2024-01-22:
Hey,

I'd like to request support for `Apator Elf 2`  HeatMeter.
It's being decoded in wmbusmeters just fine using `microclima` driver
(https://wmbusmeters.org/analyze/4e4401060411142342047a600040250668BeBed0B6daB3c09c536a1ac1385163541a6648aBc32f725342aa4530292f8d70d4B818B7898178fcc2e53e95dd0a0e0031837888B08B8ec086993B69f3c5:auto:59C4CEBE0A687F7E2B3CCBFD71718AC0).
Can you implement such support?

Or any general advice/sample PR how to add implementation for particular device into this repo.
Thanks in advance!

Decoded frame contains a lot of interesting data
```
telegram=|4E4401060411142342047A600040252F2F_0C0E658700000C13686802014C0E79200000426CFF2C0B3B3913000B2D6000000A5A63020A5E2402046D082A163102FD1700002F2F2F2F2F2F2F2F2F2F2F|

Auto driver  : not found!
Best driver  : microclima 30/30
Using driver : microclima 30/30
000   : 4e length (78 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 0106 dll-mfct (APA)
004   : 04111423 dll-id (23141104)
008   : 42 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 60 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 4025 tpl-cfg 2540 (synchronous AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 0C dif (8 digit BCD Instantaneous value)
018   : 0E vif (Energy MJ)
019 C!: 65870000 ("total_energy_consumption_kwh":2434.722222)
023   : 0C dif (8 digit BCD Instantaneous value)
024   : 13 vif (Volume l)
025 C!: 68680201 ("total_volume_m3":1026.868)
029   : 4C dif (8 digit BCD Instantaneous value storagenr=1)
030   : 0E vif (Energy MJ)
031 C!: 79200000 ("consumption_at_set_date_1_kwh":577.5)
035   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
036   : 6C vif (Date type G)
037 C!: FF2C ("set_date":"2023-12-31")
039   : 0B dif (6 digit BCD Instantaneous value)
040   : 3B vif (Volume flow l/h)
041 C!: 391300 ("volume_flow_m3h":1.339)
044   : 0B dif (6 digit BCD Instantaneous value)
045   : 2D vif (Power 10² W)
046 C!: 600000 ("power_kw":6)
049   : 0A dif (4 digit BCD Instantaneous value)
050   : 5A vif (Flow temperature 10⁻¹ °C)
051 C!: 6302 ("flow_temperature_c":26.3)
053   : 0A dif (4 digit BCD Instantaneous value)
054   : 5E vif (Return temperature 10⁻¹ °C)
055 C!: 2402 ("return_temperature_c":22.4)
057   : 04 dif (32 Bit Integer/Binary Instantaneous value)
058   : 6D vif (Date and time type)
059 C!: 082A1631 ("meter_datetime":"2024-01-22 10:08")
063   : 02 dif (16 Bit Integer/Binary Instantaneous value)
064   : FD vif (Second extension FD of VIF-codes)
065   : 17 vife (Error flags (binary))
066 C!: 0000 ("status":"OK")
068   : 2F skip
069   : 2F skip
070   : 2F skip
071   : 2F skip
072   : 2F skip
073   : 2F skip
074   : 2F skip
075   : 2F skip
076   : 2F skip
077   : 2F skip
078   : 2F skip

{
    "media":"heat",
    "meter":"microclima",
    "name":"",
    "id":"23141104",
    "consumption_at_set_date_1_kwh":577.5,
    "flow_temperature_c":26.3,
    "power_kw":6,
    "return_temperature_c":22.4,
    "set_date_1_date":"2023-12-31",
    "total_energy_consumption_kwh":2434.722222,
    "total_volume_m3":1026.868,
    "volume_flow_m3h":1.339,
    "meter_datetime":"2024-01-22 10:08",
    "status":"OK",
    "timestamp":"2024-01-22T09:23:23Z"
}

Using: wmbusmeters: 1.14.0-78-g866df74
866df740addf23546c0280cc4930ec259fed2882
```

sswierczyna, 2024-01-22:
I have realised that `Elf2` has similar configuration options to already supported `sharky774` 🎉 

Using below configuration:
```
sensor:
  - platform: wmbus
    meter_id: 0x23141104
    type: sharky774
    key: "59C4CEBE0A687F7E2B3CCBFD71718AC0"
    add_prefix: false
    lqi:
      name: "lqi"
    rssi:
      name: "rssi"
    total_energy_consumption_gj:
      name: "total_energy_consumption_gj"
    total_volume_m3:
      name: "total_volume_m3"
    # power doesnt work for some reasons :(
    power_kw:
      name: "power_kw"
    flow_temperature_c:
      name: "flow_temperature_c"
    return_temperature_c:
      name: "return_temperature_c"
    volume_flow_lh:
      name: "volume_flow_lh"
```

I'am able to read values for:
```[11:11:03][I][wmbus:088]: Using driver 'sharky774' for ID [0x23141104] RSSI: -37 dBm LQI: 128 Mode: T1 T: 4E4401060411142342047A3B0040255FA11D2858FDA88A7F1665AA831544FA471F784ACE45C2E7B9C9AECDD310C190FACF27BA53F968DB61C1714C844D66B9D0E9C2960501FBDAD32B5FCC8D9BECB4 (79)
[11:11:03][D][wmbus:095]: Decrypted T : 4E4401060411142342047A3B0040252F2F0C0E828700000C13818102014C0E79200000426CFF2C0B3B4913000B2D5000000A5A58020A5E2602046D072B163102FD1700002F2F2F2F2F2F2F2F2F2F2F (79)
[11:11:03][D][sensor:094]: 'lqi': Sending state 128.00000  with 0 decimals of accuracy
[11:11:03][D][sensor:094]: 'rssi': Sending state -37.00000 dBm with 0 decimals of accuracy
[11:11:03][D][sensor:094]: 'flow_temperature_c': Sending state 25.80000 °C with 0 decimals of accuracy
[11:11:03][D][sensor:094]: 'return_temperature_c': Sending state 22.60000 °C with 0 decimals of accuracy
[11:11:03][D][sensor:094]: 'total_energy_consumption_gj': Sending state 8.78200 GJ with 3 decimals of accuracy
[11:11:03][D][sensor:094]: 'total_volume_m3': Sending state 1028.18103 m³ with 3 decimals of accuracy
[11:11:03][D][sensor:094]: 'volume_flow_lh': Sending state 1349.00000 l/h with 0 decimals of accuracy
```

For some reason value for `power_kw` is not available even frame seems to be correctly decoded 😞 
Is this a bug?

https://wmbusmeters.org/analyze/4E4401060411142342047A240040256795712E29BFD2E8FAFC02A74B9191D604DE0DC2C743C5FCF02759301527CC75BF7D564EB11DB37C716B9BBE075E0A2C6FED35BCFC72CB5828E0E5E9DE3B1FB7:sharky774:59C4CEBE0A687F7E2B3CCBFD71718AC0
```
...
{
    "media":"heat",
    "meter":"sharky774",
    "name":"",
    "id":"23141104",
    "energy_at_set_date_kwh":577.5,
    "flow_temperature_c":25.5,
    "power_kw":3.9,
    "return_temperature_c":23,
    "set_date":"2023-12-31",
    "total_energy_consumption_kwh":2439.166667,
    "total_volume_m3":1028.008,
    "volume_flow_m3h":1.336,
    "timestamp":"2024-01-22T10:05:27Z"
}
...
```

SzczepanLeon, 2024-01-22:
> For some reason value for power_kw is not available even frame seems to be correctly decoded 😞\nIs this a bug?

No. In telegram is 0B2D but in sharky driver it is 0C2B. It is easy to extend this.

SzczepanLeon, 2024-02-08:
Please try version 3.1.2 -- 0B2D added.

closed 2024-02-08 by SzczepanLeon

sswierczyna, 2024-02-08:
Thank you @SzczepanLeon for update 🙇 
I have updated to version `3.1.2`
Much better. Power seems to be decoded, but reported value is `0.04200 kW` where actual power is `4.2kW`.
So decoded value seems to be in mega watts:)

Are you able to adjust this as well ? Or maybe i should change something in my config ?

logs:
```
[11:56:11][C][wmbus:371]: wM-Bus v3.1.2:
[11:56:11][C][wmbus:384]:   LED:
[11:56:11][C][wmbus:385]:     Pin: GPIO2
[11:56:11][C][wmbus:386]:     Duration: 1000 ms
[11:56:11][C][wmbus:388]:   CC1101 frequency: 868.950 MHz
[11:56:11][C][wmbus:389]:   CC1101 SPI bus:
[11:56:11][C][wmbus:390]:     MOSI Pin: GPIO32
[11:56:11][C][wmbus:391]:     MISO Pin: GPIO19
[11:56:11][C][wmbus:392]:     CLK Pin:  GPIO33
[11:56:11][C][wmbus:393]:     CS Pin:   GPIO23
[11:56:11][C][wmbus:394]:     GDO0 Pin: GPIO22
[11:56:11][C][wmbus:395]:     GDO2 Pin: GPIO21
[11:56:11][C][wmbus:402]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, flowiq2200, hydrocalm3, hydrus, iperl, itron, izar, kamheat, mkradio3, mkradio4, multical21, qheat, qwater, rfmtx1, sharky774, topaseskr, ultrimis, unismart, vario451
[11:56:11][C][wmbus:420]:   Meter:
[11:56:11][C][wmbus:421]:     ID: 588517636 [0x23141104]
[11:56:11][C][wmbus:422]:     Type: sharky774
[11:56:11][C][wmbus:423]:     Mode: T1
[11:56:11][C][wmbus:424]:     Key: '59C4CEBE0A687F7E2B3CCBFD71718AC0'
[11:56:11][C][wmbus:426]:     Sensor 'flow_temperature_c'
[11:56:11][C][wmbus:426]:       Device Class: 'temperature'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: '°C'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 2
[11:56:11][C][wmbus:426]:       Icon: 'mdi:coolant-temperature'
[11:56:11][C][wmbus:426]:     Sensor 'lqi'
[11:56:11][C][wmbus:426]:       Device Class: 'signal_strength'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: ''
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 0
[11:56:11][C][wmbus:426]:     Sensor 'power_kw'
[11:56:11][C][wmbus:426]:       Device Class: 'power'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: 'kW'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 3
[11:56:11][C][wmbus:426]:       Icon: 'mdi:transmission-tower-export'
[11:56:11][C][wmbus:426]:     Sensor 'return_temperature_c'
[11:56:11][C][wmbus:426]:       Device Class: 'temperature'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: '°C'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 2
[11:56:11][C][wmbus:426]:       Icon: 'mdi:coolant-temperature'
[11:56:11][C][wmbus:426]:     Sensor 'rssi'
[11:56:11][C][wmbus:426]:       Device Class: 'signal_strength'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: 'dBm'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 0
[11:56:11][C][wmbus:426]:     Sensor 'total_energy_consumption_gj'
[11:56:11][C][wmbus:426]:       Device Class: 'energy'
[11:56:11][C][wmbus:426]:       State Class: 'total_increasing'
[11:56:11][C][wmbus:426]:       Unit of Measurement: 'GJ'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 3
[11:56:11][C][wmbus:426]:       Icon: 'mdi:heating-coil'
[11:56:11][C][wmbus:426]:     Sensor 'total_volume_m3'
[11:56:11][C][wmbus:426]:       Device Class: 'water'
[11:56:11][C][wmbus:426]:       State Class: 'total_increasing'
[11:56:11][C][wmbus:426]:       Unit of Measurement: 'm³'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 3
[11:56:11][C][wmbus:426]:       Icon: 'mdi:water'
[11:56:11][C][wmbus:426]:     Sensor 'volume_flow_lh'
[11:56:11][C][wmbus:426]:       Device Class: 'water'
[11:56:11][C][wmbus:426]:       State Class: 'measurement'
[11:56:11][C][wmbus:426]:       Unit of Measurement: 'l/h'
[11:56:11][C][wmbus:426]:       Accuracy Decimals: 0
[11:56:11][C][wmbus:426]:       Icon: 'mdi:water'
[11:56:25][D][mbus:035]: Processing T1 A frame
[11:56:25][D][wmbus:096]: Using driver 'sharky774' for ID [0x23141104] RSSI: -35 dBm LQI: 128 Frame: T1 A T: 4E4401060411142342047A7C004025A83F68E9B2F6542E6BCD324A24A0B979F772DCCF3BA4651CC9876B81E7C06ABD43B3AB9CB2A59FB65DFB4ED70D66ECECAD45E1FE591EE6019C03F66F0717BB3D (79)
[11:56:25][D][wmbus:103]: Decrypted T : 4E4401060411142342047A7C0040252F2F0C0E302301000C13566343014C0E49090100426C1F310B3B5609000B2D4200000A5A59020A5E2102046D332B083202FD1700002F2F2F2F2F2F2F2F2F2F2F (79)
[11:56:25][D][sensor:094]: 'lqi': Sending state 128.00000  with 0 decimals of accuracy
[11:56:26][D][sensor:094]: 'rssi': Sending state -35.00000 dBm with 0 decimals of accuracy
[11:56:26][D][sensor:094]: 'flow_temperature_c': Sending state 25.90000 °C with 2 decimals of accuracy
[11:56:26][D][sensor:094]: 'power_kw': Sending state 0.04200 kW with 3 decimals of accuracy
[11:56:26][D][sensor:094]: 'return_temperature_c': Sending state 22.10000 °C with 2 decimals of accuracy
[11:56:26][D][sensor:094]: 'total_energy_consumption_gj': Sending state 12.33000 GJ with 3 decimals of accuracy
[11:56:26][D][sensor:094]: 'total_volume_m3': Sending state 1436.35596 m³ with 3 decimals of accuracy
[11:56:26][D][sensor:094]: 'volume_flow_lh': Sending state 956.00000 l/h with 0 decimals of accuracy
```

my esphome config:
```
  - platform: wmbus
    meter_id: -redacted-
    type: sharky774
    key: "-redacted-"
    add_prefix: false
    lqi:
      name: "lqi"
    rssi:
      name: "rssi"
    total_energy_consumption_gj:
      name: "total_energy_consumption_gj"
    total_volume_m3:
      name: "total_volume_m3"
    power_kw:
      name: "power_kw"
    flow_temperature_c:
      name: "flow_temperature_c"
      accuracy_decimals: 2
    return_temperature_c:
      name: "return_temperature_c"
      accuracy_decimals: 2
    volume_flow_lh:
      name: "volume_flow_lh"
```

SzczepanLeon, 2024-02-08:
Please try version 3.1.4

sswierczyna, 2024-02-08:
In version`3.1.6` seems to be perfect!
Thanks again @SzczepanLeon 
```
[14:08:32][D][mbus:035]: Processing T1 A frame
[14:08:32][D][wmbus:097]: Using driver 'sharky774' for ID [0x23141104] RSSI: -35 dBm LQI: 128 Frame: T1 A T: 4E4401060411142342047A6B00402513F7D3FB6419FCE281F13D212A38BE1B1373C1391225CA85EFB0C9E4D9AAD549B82994D9E4AD7FBCD55BA92D7858985B61A546C4A10EDF4CCCD004D692F7B1D9 (79)
[14:08:32][D][wmbus:104]: Decrypted T : 4E4401060411142342047A6B0040252F2F0C0E642301000C13848443014C0E49090100426C1F310B3B5909000B2D4300000A5A63020A5E2502046D042E083202FD1700002F2F2F2F2F2F2F2F2F2F2F (79)
[14:08:32][D][sensor:094]: 'lqi': Sending state 128.00000  with 0 decimals of accuracy
[14:08:32][D][sensor:094]: 'rssi': Sending state -35.00000 dBm with 0 decimals of accuracy
[14:08:32][D][sensor:094]: 'flow_temperature_c': Sending state 26.30000 °C with 2 decimals of accuracy
[14:08:32][D][sensor:094]: 'power_kw': Sending state 4.30000 kW with 3 decimals of accuracy
[14:08:32][D][sensor:094]: 'return_temperature_c': Sending state 22.50000 °C with 2 decimals of accuracy
[14:08:32][D][sensor:094]: 'total_energy_consumption_gj': Sending state 12.36400 GJ with 3 decimals of accuracy
[14:08:32][D][sensor:094]: 'total_volume_m3': Sending state 1438.48401 m³ with 3 decimals of accuracy
```