SzczepanLeon/esphome-components #70: Adding  temp_radiator_c  ( driver fhkvdataiii )

7maxx9, 2024-01-25:
Hey,
I would like to ask about the possibility of adding temp_radiator_c ( driver fhkvdataiii ) - temp_room_c was added in the last release - thanks. 
This is a very important indicator for these devices (in combination with the previously added temp_room_c ).
The decoder at https://wmbusmeters.org decodes this information:

Auto driver  : fhkvdataiii
Best driver  : fhkvdataiii 12/52
Using driver : fhkvdataiii 00/00
000   : 32 length (50 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 6850 dll-mfct (TCH)
004   : xxxxxxx dll-id (xxxxxxx)
008   : 69 dll-version
009   : 80 dll-type (Heat Cost Allocator)
010   : a0 tpl-ci-field (Mfct specific)
011 C?: 11BF2E0000800300005E073D07000000000000000000000000000000000000000000000000000000 mfct specific
012 C!: *** bf2e previous_date = 2023-05-31T02:00:00Z
014 C!: *** 0000 previous_hca = 0
016 C!: *** 8003 current_date = 2024-01-24T02:00:00Z
018 C!: *** 0000 current_hca = 0
020 C!: *** 5e07 temp_room_c = 18.859999999999999
022 C!: *** 3d07 temp_radiator_c = 18.530000000000001

SzczepanLeon, 2024-02-08:
Please try version 3.1.6

closed 2024-02-08 by SzczepanLeon

7maxx9, 2024-02-13:
works as it should. Thank you very much.:)