SzczepanLeon/esphome-components #75: Hydrus not working

Flo2410, 2024-02-12:
I want to use this component with my Hydrus Diehl water meter. 

Unfortunately, I only get the bellow error message, although it can be decoded on [wmbusmeters.org](https://wmbusmeters.org/analyze/3e44a5118694177170077a69003005fd3859e9aa79a4412019793ddB8620ea05e74e86cc16a2155753Bd117d9880Bf6c70cBd6cBda40207B79a16e82959967:hydrus) 

### Decoded Telegram

```json
{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"71179486",
    "flow_temperature_c":10.5,
    "remaining_battery_life_y":13.3692,
    "target_m3":35.791,
    "target_datetime":"2024-01-31 23:59",
    "total_m3":37.512,
    "status":"OK",
    "timestamp":"2024-02-12T20:16:03Z"
}
```

### Error Message

```
[21:15:35][D][mbus:035]: Processing T1 A frame
[21:15:35][D][wmbus:097]: Using driver 'hydrus' for ID [0x71179486] RSSI: -83 dBm LQI: 128 Frame: T1 A T: 3E44A5118694177170077A69003005FD3859E9AA79A4412019793DDB8620EA05E74E86CC16A2155753BD117D9880BF6C70CBD6CBDA40207B79A16E82959967 (63)
[21:15:35][D][wmbus:161]: Can't get value(s) from telegram for ID [0x71179486]
```

### My Config

```yaml
wmbus:
  mosi_pin: GPIO23
  miso_pin: GPIO19
  clk_pin: GPIO18
  cs_pin: GPIO05
  gdo0_pin: GPIO16
  gdo2_pin: GPIO17

  log_unknown: False

sensor:
  - platform: wmbus
    meter_id: !secret watermeter_id
    type: hydrus
    add_prefix: false

    total_water_m3:
      name: "Total"
      id: "waterdisplay"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter
```

SzczepanLeon, 2024-02-12:
You need key. Seems that wmbusmeters use some default one.

Flo2410, 2024-02-12:
How do I get that default key, which seems to work fine?

SzczepanLeon, 2024-02-12:
Try this one:
**51728910E66D83F851728910E66D83F8**

closed 2024-02-12 by SzczepanLeon

Flo2410, 2024-02-12:
Yes, this works. Thank you very much!