SzczepanLeon/esphome-components #76: FlowIQ2200 not 100% working

evilmumi, 2024-02-12:
After implementation in main I let it run for some days.
Today i found out, that it stopped some days ago
[image]

I use it from https://github.com/zibous/ha-watermeter/
i am not sure where to post issue.
reboot did not fix only power cut for some seconds

some logs attached

```

[21:51:25][V][api.connection:1103]: Hello from client: 'Home Assistant 2024.1.5' | 192.168.0.148 | API Version 1.9
[21:51:25][D][api.connection:1121]: Home Assistant 2024.1.5 (192.168.0.148): Connected successfully
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:25][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:26][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:27][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][V][api.connection:1208]: Cannot send message because of TCP buffer space
[21:51:28][D][sntp:078]: Synchronized time: 2024-02-12 21:51:28
[21:51:28][W][system:447]: Synchronized sntp clock
[21:51:28][D][wmbus:535]: cc1101 state message: Init, error code: 0
[21:51:30][D][wmbus:535]: cc1101 state message: Init, error code: 0
[21:51:39][V][rxLoop:153]: Have 55 bytes from CC1101 Rx
[21:51:39][D][mbus:024]: Processing C1 B frame
[21:51:39][V][mbus:028]: Frame: 3444372C347971233A168D20C8C1044221366650AA535CC0935126D3F3891A2B6DFDDBF7928224C483E1AA8C10EE3FF26CC8760DD3 (53) [with CRC]
[21:51:39][V][mbus:158]: Validating CRC for Block1 + Block2
[21:51:39][D][crc:035]:     calculated: 0xCC5B, read: 0x0DD3  !!!
[21:52:27][V][rxLoop:153]: Have 19 bytes from CC1101 Rx
[21:52:27][D][mbus:024]: Processing C1 B frame
[21:52:27][V][mbus:028]: Frame: 10221B961A3CB8919D0B469065E8022110 (17) [with CRC]
[21:52:27][V][mbus:158]: Validating CRC for Block1 + Block2
[21:52:27][D][crc:035]:     calculated: 0x025A, read: 0x2110  !!!
[21:52:30][D][wmbus:535]: cc1101 state message: Init, error code: 0
[21:53:15][V][rxLoop:153]: Have 55 bytes from CC1101 Rx
[21:53:15][D][mbus:024]: Processing C1 B frame
[21:53:15][V][mbus:028]: Frame: 3444372C347971233A168D20CEE0044221125A3492B3DC570B786FF2202F8E1A1DBE119DFFE6803CD43E58FD1FA203312BDF97690C (53) [with CRC]
[... 40 lines cut ...]
[21:54:46][D][mbus:024]: Processing C1 B frame
[21:54:46][V][mbus:028]: Frame: 3C04372C427971233A168D20A3E00442216F110D1CBEBA97C6D9CAA3F0EB68D97706CEA693F8591A81901849E0144C96ABCFAECFB31608E4793D47B264 (61) [with CRC]
[21:54:46][V][mbus:158]: Validating CRC for Block1 + Block2
[21:54:46][D][crc:035]:     calculated: 0x28DA, read: 0xB264  !!!
[21:55:24][V][rxLoop:153]: Have 55 bytes from CC1101 Rx
[21:55:24][D][mbus:024]: Processing C1 B frame
[21:55:24][V][mbus:028]: Frame: 3444372C347971233A168D20D600054221F7D9C75EBCA665D21D0DBB6FC225D061B00B5C0BD51E3826B1ECE82419F0D7D9123EBE73 (53) [with CRC]
[21:55:24][V][mbus:158]: Validating CRC for Block1 + Block2
[21:55:24][V][crc:031]:     calculated: 0xBE73, read: 0xBE73
[21:55:24][V][mbus:055]: Frame: 3244372C347971233A168D20D600054221F7D9C75EBCA665D21D0DBB6FC225D061B00B5C0BD51E3826B1ECE82419F0D7D9123E (51) [without CRC]
[21:55:25][D][wmbus:096]: Using driver 'flowiq2200' for ID [0x23717934] RSSI: -92 dBm LQI: 139 Frame: C1 B T: 3244372C347971233A168D20D600054221F7D9C75EBCA665D21D0DBB6FC225D061B00B5C0BD51E3826B1ECE82419F0D7D9123E (51)
[21:55:25][V][wmbus:282]: Validating CRC for ELL payload
[21:55:25][V][crc:046]:     calculated: 0xA27E, read: 0xA27E
[21:55:25][D][wmbus:103]: Decrypted T : 3244372C347971233A168D20D6000542217EA2790509CBF4000000007CE203007C0506600006600001320B110000FEEE03000F (51)
[21:55:25][V][wmbus:120]: Publishing 'max_flow_lh' = 1404.0000
[21:55:25][V][wmbus:120]: Publishing 'status' = 0.0000
[21:55:25][V][wmbus:120]: Publishing 'target_water_m3' = 254.5880
[21:55:25][V][wmbus:120]: Publishing 'total_water_m3' = 257.7900
[21:55:25][I][wmbus:754]: Reset current value to: 0.000, Waterdisplay value: 257.790, last value: 257.790
[21:55:25][D][wmbus:768]: Diff millisecond is: 64610
[21:55:25][D][wmbus:535]: cc1101 state message: Warte auf Daten, error code: 1
[21:55:25][V][wmbus:120]: Publishing 'volume_flow_lh' = 0.0000
[21:55:30][D][wmbus:535]: cc1101 state message: Warte auf Daten, error code: 1

```

SzczepanLeon, 2024-02-12:
But I see Publishing traces.

evilmumi, 2024-02-13:
how to test? Could you be so kind and provide "smallest" yaml for that?

evilmumi, 2024-02-23:
It was only once - since now no problems

closed 2024-02-23 by SzczepanLeon