SzczepanLeon/esphome-components #77: Feature Request: More Hydrus data

Flo2410, 2024-02-13:
It would be nice to have all the data available in the telegram also be available as a sensor in Home Assistant for the Hydrus water meter.

Currently, only the `total_water_m3` is supported. 
But the telegram includes more data, which would be nice to have in HA.

Here is a sample telegram: https://wmbusmeters.org/analyze/3e44a5118694177170077a69003005fd3859e9aa79a4412019793ddB8620ea05e74e86cc16a2155753Bd117d9880Bf6c70cBd6cBda40207B79a16e82959967:hydrus

Which has the following data:

```json
{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"71179486",
    "flow_temperature_c":10.5,
    "remaining_battery_life_y":13.3692,
    "target_m3":35.791,
    "target_datetime":"2024-01-31 23:59",
    "total_m3":37.512,
    "status":"OK",
    "timestamp":"2024-02-13T11:25:04Z"
}
```

geeuz, 2024-03-14:
Hi Flo,

that would be great indeed!

I have a hydrus too which sends unencrypted telegrams, I am able to receive them with my CC1101 and can read the data with the wmbusmeters Analyzer.

But it doesn't work to get the data into Esphome, I always get the message "[D][wmbus:161]: Can't get value(s) from telegram for ID [0x709.....]"

Can you please share your configuration?

Here's mine:
```
esphome:
  name: "poolbox"
  friendly_name: "poolbox"

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [wmbus]

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  networks:
  - ssid: !secret wifi_ssidg
    password: !secret wifi_passwordg
  manual_ip:
    static_ip: 192.168.10.42
    gateway: 192.168.10.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Poolbox Fallback Hotspot"
    password: !secret hotspot_password

captive_portal:

time:
  - platform: sntp
    id: time_sntp
    timezone: Europe/Vienna
    servers:
      - 0.at.pool.ntp.org
      - 0.pool.ntp.org
      - 1.pool.ntp.org

    on_time_sync:
      then:
        - logger.log:
            level: WARN
            tag: "system"
            format: "Synchronized sntp clock"

wmbus:
  mosi_pin: GPIO23  # MOSI Attached to Hardware SPI controller MOSI SPI Interface
  miso_pin: GPIO19  # MISO Attached to Hardware SPI controller MISO SPI Interface
  clk_pin: GPIO18   # SCK  Attached to Hardware SPI controller CLK
  cs_pin: GPIO05    # CSN  Attached to Hardware SPI controller
  gdo0_pin: GPIO16  # GDO0 Clock output. High Impedance !
  gdo2_pin: GPIO17  # GDO2 status signals. High Impedance !
  log_unknown: True

sensor:
  - platform: wmbus
    meter_id: 0x709.....
    type: hydrus
    key: ""
    add_prefix: false
    lqi:
      id: wmbus_cc1101_lqi
      name: "Watermeter CC1101 LQI"
      entity_category: "diagnostic"
      unit_of_measurement: "lqi"
      state_class: "measurement"
    rssi:
      id: wmbus_cc1101_rssi
      name: "Watermeter CC1101 RSSI"
      unit_of_measurement: "dbm"
      entity_category: "diagnostic"
    total_water_m3:
      name: "Watermeter Display"
      id: "waterdisplay"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      accuracy_decimals: 3
      icon: mdi:counter

```

ThomDietrich, 2024-06-26:
I would also be interested in this, specifically `remaining_battery_life_y` and `flow_temperature_c`.

```json
{
    "media":"water",
    "meter":"hydrus",
    "name":"",
    "id":"81755588",
    "at_date":"2128-03-31",
    "flow_m3h":0,
    "flow_temperature_c":16.4,
    "total_m3":5.434,
    "total_at_date_m3":188888.887,
    "status":"OK",
    "timestamp":"2024-06-26T14:28:21Z"
}
```

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon

ThomDietrich, 2024-10-08:
Confirmed working. The list of values is still limited to the values described but that is now an issue to raise over at wmbusmeters