SzczepanLeon/esphome-components #79: Can't connect via TCP to 10.0.0.1:7227

VAC91, 2024-02-14:
Mam problem z bardzo długim czasem operacji no i dodatkowo występuje błąd który widać w wycinku logu niżej.
Ogólnie dekoduje mi ramkę, dostaję odczyty co 8 sekund. Czy mam coś źle zrobione lub można to poprawić?
Urządzenie działa na ESP8285 - ESP-01M z zegarem ustawionym na 160MHz

```
[22:00:20][E][wmbus:241]: Can't connect via TCP to 10.0.0.1:7227
[22:00:20][W][component:214]: Component wmbus took a long time for an operation (5.19 s).
[22:00:20][W][component:215]: Components should block for at most 20-30ms.
```

Mój yaml
```
time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO13 
  miso_pin: GPIO12 
  clk_pin: GPIO14  
  cs_pin: GPIO2
  gdo0_pin: GPIO4 
  gdo2_pin: GPIO5

  led_pin: GPIO15
  led_blink_time: "200ms"

  clients:
    - name: "wmbusmeters"
      ip_address: "10.0.0.1"
      port: 7227

sensor:
  - platform: wmbus
    meter_id: 0x2106087d
    type: izar
    lqi:
      name: "Radio LQI"
    rssi:
      name: "Radio RSSI"
    total_water_m3:
      name: "Zimna woda"
    last_month_total_water_m3:
      name: "Poprzedni miesiac"
    current_month_total_water_l:
      name: "Obecny miesiac"
    remaining_battery_life_y:
      name: "Remaining battery life"
    transmit_period_s:
      name: "Transmit period"
    current_alarms:
      name: "Alarm"
    previous_alarms:
      name: "Alarm poprzedni"
```

SzczepanLeon, 2024-02-14:
A po drugiej stronie coś odbiera te ramki po TCP? Z logów wynika że nie. Dlatego masz taki długi czas. Możesz zmienić transport na UDP, wtedy będzie szybciej, ale nie będziesz miał feedbacku czy telegram został przez klienta odebrany.

A tak w ogóle to potrzebujesz wysyłać telegramy do klienta?

VAC91, 2024-02-14:
Ja tylko wpisałem tak jak jest w 2.1.1. Example
Całą sekcjęclients mogę usunąc z yamla?

SzczepanLeon, 2024-02-14:
Jak wysyłasz  bezpośrednio do HA to sekcji client nie potrzebujesz.

VAC91, 2024-02-14:
Tak. Do HA to idzie. Te 100ms to nie jest dużo, pewnie nic z tym warningiem już nie będę mógł zrobić? Pewnie zależy od zastosowanego procesora?
Dziękuję za szybką pomoc, już i tak jestem zadowolony.

closed 2024-02-14 by SzczepanLeon