SzczepanLeon/esphome-components #80: flowiq2200 - No C1 frames

anhorbc, 2024-02-15:
Trying to capture data with the flowiq2200 driver. I am not seeing a single C1 frame. CC1101 is 10cm from the water meter. 
I can only see a bunch of T1 A frames.
Using a 868Mhz cc1101 module, that previously successfully captured frames from the same meter (using different code).
I am able to successfully capture C1 telegrams from the meter with rtlsdr + wmbusmeters.
Not sure how to troubleshoot this further? I have tried switching mode between T1C1 to C1 and back, tried both decimal and hexadecimal meter id. 

Config:
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]
logger:
  id: appslogger
  level: "VERBOSE"
  logs:
    wmbus: "VERBOSE"
    wMBus-lib: "VERBOSE"

wmbus:
  mosi_pin: GPIO18
  miso_pin: GPIO19
  clk_pin:  GPIO21
  cs_pin:   GPIO14
  gdo0_pin: GPIO15
  gdo2_pin: GPIO16
 
sensor:
  - platform: wmbus
    meter_id: my_meter_id
    type: flowiq2200
    key: "my_key"
    add_prefix: true
    mode: T1C1
    total_water_m3:
      id: "my_water"

Output:
[15:14:36][C][wmbus:372]: wM-Bus v3.1.8:
[15:14:36][C][wmbus:389]:   CC1101 frequency: 868.950 MHz
[15:14:36][C][wmbus:390]:   CC1101 SPI bus:
[15:14:36][C][wmbus:391]:     MOSI Pin: GPIO18
[15:14:36][C][wmbus:392]:     MISO Pin: GPIO19
[15:14:36][C][wmbus:393]:     CLK Pin:  GPIO21
[15:14:36][C][wmbus:394]:     CS Pin:   GPIO14
[15:14:36][C][wmbus:395]:     GDO0 Pin: GPIO15
[15:14:36][C][wmbus:396]:     GDO2 Pin: GPIO16
[15:14:36][C][wmbus:403]:   Available drivers:  amiplus, apator08, apator162, apatoreitn, bmeters, c5isf, compact5, dme07, elf, evo868, fhkvdataiii, flowiq2200, hydrocalm3, hydrus, iperl, itron, izar, kamheat, mkradio3, mkradio4, mkradio4a, multical21, qheat, qwater, rfmtx1, sharky774, topaseskr, ultrimis, unismart, vario451
[15:14:36][C][wmbus:421]:   Meter:
[15:14:36][C][wmbus:422]:     ID: my_meter_id[my_meter_id_in_hex]
[15:14:36][C][wmbus:423]:     Type: flowiq2200
[15:14:36][C][wmbus:424]:     Mode: T1C1
[15:14:36][C][wmbus:425]:     Key: 'my_key'
[15:14:36][C][wmbus:427]:     Sensor 'my_meter_id_in_hex my_water'
[15:14:36][C][wmbus:427]:       Device Class: 'water'
[15:14:36][C][wmbus:427]:       State Class: 'total_increasing'
[15:14:36][C][wmbus:427]:       Unit of Measurement: 'm³'
[15:14:36][C][wmbus:427]:       Accuracy Decimals: 3
[15:14:36][C][wmbus:427]:       Icon: 'mdi:water'
[15:22:10][D][sntp:078]: Synchronized time: 2024-02-15 15:22:10
[15:24:44][V][rxLoop:157]: Have 338 bytes from CC1101 Rx, RSSI: -83 dBm LQI: 128
[15:24:44][D][mbus:035]: Processing T1 A frame
[15:24:44][V][mbus:039]: Frame: D1A98BD1A1FFFFEDE6C2516D59A9F0B7D9A99CD95259368037BF037E00FE04F62B80F79AE92E9CE1EEEF26E6E7C22BFA0E4CD8489E84AF9BA7A653EF42FF025FCDA98BD1A1FFFFEDE6C2516D592BF0B7D9F6000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[15:24:44][V][3of6:083]: Decode 3 out of 6 failed..
[15:24:44][W][component:214]: Component wmbus took a long time for an operation (0.06 s).
[15:24:44][W][component:215]: Components should block for at most 20-30ms.
[15:24:44][V][rxLoop:157]: Have 338 bytes from CC1101 Rx, RSSI: -80 dBm LQI: 128
[15:24:44][D][mbus:035]: Processing T1 A frame
[15:24:44][V][mbus:039]: Frame: D1A98BD1A1FFFFEDE6C2516D592BF0B7D9F69CD95259368037BF037E00FE04F62B80F79AE92E9CE1EEEF26E6E7C22BFA0E4CD8489E84AF9BA7A653EF42FF025FCDA98BD1A1FFFFEDE6C2516D592BF0B7D9F6000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[15:24:44][V][3of6:083]: Decode 3 out of 6 failed..
[15:24:44][W][component:214]: Component wmbus took a long time for an operation (0.06 s).
[15:24:44][W][component:215]: Components should block for at most 20-30ms.
[15:24:44][V][rxLoop:157]: Have 338 bytes from CC1101 Rx, RSSI: -80 dBm LQI: 128
[15:24:44][D][mbus:035]: Processing T1 A frame
[15:24:44][V][mbus:039]: Frame: D1A98BD1A1FFFFEDE6C2516D592BF0B7D9F69CD95259365937BF037E00FE04F62B80F79AE92E9CE1EEEF26E6E7C22BFA0E4CD8489E84AF9BA7A653EF42FF025FCDA98BD1A1FFFFEDE6C2516D592BF0B7D9F6000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[15:24:44][V][3of6:083]: Decode 3 out of 6 failed..
<---Lots, and lots of T1 A frames for hours--->

SzczepanLeon, 2024-02-21:
Please try version:
```
external_components:
  - source: github://SzczepanLeon/esphome-components@long_data
    refresh: 0d
    components: [ wmbus ]
```

anhorbc, 2024-02-24:
Turned out to be a grounding issue. After properly grounding the cc1101 module, it is working as expected.

closed 2024-02-24 by anhorbc