SzczepanLeon/esphome-components #81: Kamheat: Can't get value(s) from telegram

Candymirror, 2024-02-21:
I'm trying to read wmbus messages from my kamstrup multical 302 (kamheat). I've been changing and editing the configuration yaml multiple times the last few days and i'm making progress, but am stuck at this point.

Findings sofar. Using the MQTT component seems to interfere with the reception of kamheat T1 telegrams.
I sure my kamheat ID is in decimal 67260379, but I had to set it as hex 0x67260379 to match it.

Now I'm stuck on "Can't get value(s) from telegram". Can it be that encrypted telegrams are not supported for kamheat?

`[10:35:12][V][rxLoop:157]: Have 137 bytes from CC1101 Rx, RSSI: -18 dBm LQI: 128
[10:35:12][D][mbus:035]: Processing T1 A frame
[10:35:12][V][mbus:039]: Frame: 73271C3B13B44E558B39A6932D659C3B4C4E4E63655AC7163997319A665971CA539634CB371C9A39339A732A5672665AC5AB0ECACD29A7135994EB0ECA5B168D6A5AB3499966C5A9C4E393C5CA7469970D396B3498D6938CD8F2C6C9A58F23A36A5D1A8DC58EC56C5AB322E34F14D3CA99932CDCA32CB4F439CA74D0B4F24CDC746B2C666A635364EA (137) [RAW]
[10:35:12][V][mbus:045]: Frame: 4E442D2C7903266730042CD27A190840254DAA5544F79B731DE627264EF04A56D682E8CFFD159282E980B0F68CA5580FD227D4FC6541208CA167B1BED8A9BE2B69C6B402D0D68E3B7D77EFA731EB337C24FCC37E71DC6EDA6A1752 (91) [with CRC]
[10:35:12][V][mbus:088]: Validating CRC for Block1
[10:35:12][V][crc:031]:     calculated: 0x2CD2, read: 0x2CD2
[10:35:12][V][mbus:108]: Validating CRC for Block2
[10:35:12][V][crc:031]:     calculated: 0x4EF0, read: 0x4EF0
[10:35:12][V][mbus:108]: Validating CRC for Block3
[10:35:12][V][crc:031]:     calculated: 0x580F, read: 0x580F
[10:35:12][V][mbus:108]: Validating CRC for Block4
[10:35:12][V][crc:031]:     calculated: 0x69C6, read: 0x69C6
[10:35:12][V][mbus:108]: Validating CRC for Block5
[10:35:12][V][crc:031]:     calculated: 0xC37E, read: 0xC37E
[10:35:12][V][mbus:108]: Validating CRC for Block6
[10:35:12][V][crc:031]:     calculated: 0x1752, read: 0x1752
[10:35:12][V][mbus:055]: Frame: 4E442D2C7903266730047A190840254DAA5544F79B731DE627264A56D682E8CFFD159282E980B0F68CA5D227D4FC6541208CA167B1BED8A9BE2BB402D0D68E3B7D77EFA731EB337C24FC71DC6EDA6A (79) [without CRC]
[10:35:12][D][wmbus:097]: Using driver 'kamheat' for ID [0x67260379] RSSI: -18 dBm LQI: 128 Frame: T1 A T: 4E442D2C7903266730047A190840254DAA5544F79B731DE627264A56D682E8CFFD159282E980B0F68CA5D227D4FC6541208CA167B1BED8A9BE2BB402D0D68E3B7D77EFA731EB337C24FC71DC6EDA6A (79)
[10:35:12][V][utils:291]: 2F2F check after decrypting - OK
[10:35:12][D][wmbus:104]: Decrypted T : 4E442D2C7903266730047A190840252F2F04057C2B00004405792B0000426CFF2C0413C1801000441364310B00022D000002590904025D5904023B00000322DA2601026C153201FF21202F2F2F2F2F (79)
[10:35:12][D][wmbus:161]: Can't get value(s) from telegram for ID [0x67260379]
[10:35:12][W][component:214]: Component wmbus took a long time for an operation (0.19 s).
[10:35:12][W][component:215]: Components should block for at most 20-30ms.`

SzczepanLeon, 2024-02-21:
Please try latest version (3.1.9).

Candymirror, 2024-02-21:
Thanks, making progress. Did set logging to very_verbose. Removed RSSI sensor for now.

LQI and Flow sensors are sending data, "total_energy_consumption_kwh" not yet.
If possible and not much of a hassle, I would like support for "power_kw" and T1 + T2

`[17:25:20][V][mbus:055]: Frame: 4E442D2C7903266730047A340840256A5197994891EFC231F07EEEDEC0F26F9BD2DF0ED388A592076DC54CA2D255CF05AA38BE73793658D21AE5F161537D44477729E4625DD78AE118DA6ADFF04F07 (79) [without CRC]
[17:25:20][VV][wmbus:061]: Have data from CC1101 ...
[17:25:20][D][wmbus:097]: Using driver 'kamheat' for ID [0x67260379] RSSI: -17 dBm LQI: 128 Frame: T1 A T: 4E442D2C7903266730047A340840256A5197994891EFC231F07EEEDEC0F26F9BD2DF0ED388A592076DC54CA2D255CF05AA38BE73793658D21AE5F161537D44477729E4625DD78AE118DA6ADFF04F07 (79)
[17:25:20][VV][wmbus:099]: Key defined, trying to decrypt telegram ...
[17:25:20][VV][utils:223]: (TPL)  CI: 7A  offset: 15
[17:25:20][VV][utils:226]: (TPL)  IV: 2D2C7903266730043434343434343434
[17:25:20][VV][utils:229]: (TPL) KEY: 2507F03CDDF956BB2C96CEF822401200 (16)
[17:25:20][VV][utils:252]: (TPL) AES CBC IV decrypting: 6A5197994891EFC231F07EEEDEC0F26F9BD2DF0ED388A592076DC54CA2D255CF05AA38BE73793658D21AE5F161537D44477729E4625DD78AE118DA6ADFF04F07 (64)
[17:25:21][VV][utils:282]: (TPL) AES CBC IV  decrypted: 2F2F04057C2B00004405792B0000426CFF2C041305961000441364310B00022D000002593F04025DEE0D023B00000322E12601026C153201FF21202F2F2F2F2F (64)
[17:25:21][V][utils:291]: 2F2F check after decrypting - OK
[17:25:21][D][wmbus:104]: Decrypted T : 4E442D2C7903266730047A340840252F2F04057C2B00004405792B0000426CFF2C041305961000441364310B00022D000002593F04025DEE0D023B00000322E12601026C153201FF21202F2F2F2F2F (79)
[17:25:21][VV][drivers:070]: Found register '023B' with '0'->'0.000000'
[17:25:21][VV][drivers:035]: Value added to map
[17:25:21][V][sensor:043]: '0x67260379 My lqi': Received new state 128.000000
[17:25:21][D][sensor:094]: '0x67260379 My lqi': Sending state 128.00000  with 0 decimals of accuracy
[17:25:21][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 2970662663
  state: 128
  missing_state: NO
}
[17:25:21][V][wmbus:121]: Publishing 'volume_flow_lh' = 0.0000
[17:25:21][V][sensor:043]: '0x67260379 flow': Received new state 0.000000
[17:25:21][D][sensor:094]: '0x67260379 flow': Sending state 0.00000 l/h with 0 decimals of accuracy
[17:25:21][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 4230391750
  state: 0
  missing_state: NO
}
[17:25:21][W][component:214]: Component wmbus took a long time for an operation (0.33 s).
[17:25:21][W][component:215]: Components should block for at most 20-30ms.`

Candymirror, 2024-02-21:
Just checked. Values are coming in. Temperatures are not yet correct, they need to be multiplied by 10. 
For actual power I'm not yet sure, because it's 0 at the moment.

"total_energy_consumption_kwh" still isn't mapped.
Guess that for my telegrams it should be: 
add_to_map(ret_val, "total_energy_consumption_kwh", this->get_0405(telegram));

Candymirror, 2024-02-23:
Thanks, SzczepanLeon.

Sensor values are correct now, one last minor change requested though.
Could you change the accuracy of the temperatures to 2 decimals? 

I've changed it already in my local wmbus component

Candymirror, 2024-02-23:
Thanks, I'm just starting with esphome (second project). For me the kamheat is working as expexted. time to get rid of the RPI with rtl-wmbus.

closed 2024-02-23 by SzczepanLeon