SzczepanLeon/esphome-components #83: Home assistant not updated with sensor values.

anhorbc, 2024-02-24:
Log from esp32:
[20:36:03][D][wmbus:097]: Using driver 'flowiq2200' for ID [0x23543977] RSSI: -43 dBm LQI: 128 Frame: C1 B T: 4D44372C773954233A168D20F7D4F50D217D0BC39BE478CE95273F0C312E95A5C062857D970FC239AFFE71AA0E7C7455849B742FA591E80187906C3DB1BBF1689758905DEE4534BB8D9284C1F2C7 (78)
[20:36:03][V][wmbus:283]: Validating CRC for ELL payload
[20:36:03][V][crc:046]:     calculated: 0xB424, read: 0xB424
[20:36:03][D][wmbus:104]: Decrypted T : 4D44372C773954233A168D20F7D4F50D2124B47804FF230000000004134BD501004413C7C50100426C0132023B000092013B9C01A2013B000006FF1B257001197001A1015B0791015B12A101670F (78)
[20:36:03][V][drivers:573]: Found register '04FF23' with '00000000'
[20:36:03][V][sensor:043]: 'My lqi': Received new state 128.000000
[20:36:03][D][sensor:094]: 'My lqi': Sending state 128.00000  with 0 decimals of accuracy
[20:36:03][V][sensor:043]: 'My RSSI': Received new state -43.000000
[20:36:04][D][sensor:094]: 'My RSSI': Sending state -43.00000 dBm with 0 decimals of accuracy
[20:36:04][V][wmbus:121]: Publishing 'status' = 0.0000
[20:36:04][V][sensor:043]: 'status': Received new state 0.000000
[20:36:04][D][sensor:094]: 'status': Sending state 0.00000  with 0 decimals of accuracy
[20:36:04][V][wmbus:121]: Publishing 'target_water_m3' = 116.1670
[20:36:04][V][sensor:043]: 'target_water_m3': Received new state 116.167000
[20:36:04][D][sensor:094]: 'target_water_m3': Sending state 116.16700 m³ with 3 decimals of accuracy
[20:36:04][V][wmbus:121]: Publishing 'total_water_m3' = 120.1390
[20:36:04][V][sensor:043]: 'total_water_m3': Received new state 120.139000
[20:36:04][D][sensor:094]: 'total_water_m3': Sending state 120.13900 m³ with 3 decimals of accuracy
[20:36:04][V][wmbus:121]: Publishing 'volume_flow_lh' = 0.0000
[20:36:04][V][sensor:043]: 'volume_flow_lh': Received new state 0.000000
[20:36:04][D][sensor:094]: 'volume_flow_lh': Sending state 0.00000 l/h with 0 decimals of accuracy

Device in home assistant:
[image]

SzczepanLeon, 2024-02-24:
This is something not related to **wmbus** component. It is something between _ESPHOME_ and _HA_.

closed 2024-02-24 by SzczepanLeon

anhorbc, 2024-02-24:
Possible. Strange though that LQI and RSSI is showing up. ESPHome bug report posted at https://github.com/esphome/issues/issues/5527

SzczepanLeon, 2024-02-24:
My component is not official-one. Did you try to remove and add again that ESPHOME in HA?