SzczepanLeon/esphome-components #84: Lack of name results in invisible sensors

anhorbc, 2024-02-24:
Missing sensor names makes sensors invisible in HA.
Example:
YAML:
sensor:
  - platform: wmbus
    meter_id: 0x12345678
    type: flowiq2200
    key: "MY_SECRET_KEY"
    add_prefix: false
    mode: C1
    total_water_m3:
      id: total_water_m3
    target_water_m3:
      id: target_water_m3
    volume_flow_lh:
      id: volume_flow_lh
    max_flow_lh:
      id: max_flow_lh
    status:
      id: status
= No sensors showing up in home assistant.

The following does:
YAML:
sensor:
  - platform: wmbus
    meter_id: 0x12345678
    type: flowiq2200
    key: "MY_SECRET_KEY"
    add_prefix: false
    mode: C1
    total_water_m3:
      id: total_water_m3
      name: "total  water"
    target_water_m3:
      id: target_water_m3
      name: "target water"
    volume_flow_lh:
      id: volume_flow_lh
      name: "volume flow"
    max_flow_lh:
      id: max_flow_lh
      name: "max flow"
    status:
      id: status
      name: "Status"

anhorbc, 2024-02-24:
Obviously my lack of understanding of HA is the problem. 
It turns out that if only supplying id, the sensor is "internal only" to esphome. Only by supplying a name the sensor is showing up in HA. Sorry about this, and thanks for your effort in making this work!

closed 2024-02-24 by anhorbc

SzczepanLeon, 2024-02-24:
No problem ;)