SzczepanLeon/esphome-components #88: Wish: MISO/MOSI/CLK sharing

anhorbc, 2024-03-05:
It would be great if the component could use the default SPI component for assignment of MISO/MOSI/CLK. That would allow for sharing the MISO/MOSI/CLK pins with other SPI devices.

spi:
     clk_pin: GPIO14
     mosi_pin: GPIO27
     miso_pin: GPIO26
     interface: hardware
wmbus:
  cs_pin:   GPIO33
  gdo0_pin: GPIO4
  gdo2_pin: GPIO2

romanhectorsa, 2024-03-15:
Hi.
Such functionality would be very useful.
In its current form, the wmbus component conflicts with the ILI9xxx library, which uses SPI. The logs show that the initialization of the wmbus component freezes the screen, connected via SPI. With screens connected via I2C, this effect does not occur.

SzczepanLeon, 2024-12-15:
I don't have plans to add this feature.
You can use second SPI from ESPHome.

closed 2024-12-15 by SzczepanLeon