SzczepanLeon/esphome-components #89: maddalena Electo SJ

Mareckil, 2024-03-15:
Is it possible to add driver for this water meter?
https://www.maddalena.it/en/product/electo-sj/

SzczepanLeon, 2024-03-15:
It is OMS meter. Please try with existing maddalena evo driver.

Mareckil, 2024-03-18:
Hi, we where talking on ArturHome forum and I've left grabbed telegram frame from my meter. But I think it could be usefull to leave it over here to:
Using driver ‘evo868’ for ID [0x24004006] RSSI: -50 dBm LQI: 128 Frame: T1 A T: 4E4424340640002401077A4E0000202F2F041390020000046D2309123304FD17000000000E780640002401FF441300000000426C01018401130000000082016C0101D3013B820400C4016D07130F33 (79)

SGXander, 2024-05-01:
@Mareckil did the evo868 driver work with the electo-sj?

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon