SzczepanLeon/esphome-components #93: multical21 driver only decodes some of the received telegrams

hanzoh, 2024-04-12:
Hi,

you did an awesome job on this component, thank you!

I am getting the correct values from my Multical21 meter, but I noticed in the VERY_VERBOSE log, that it also receives telegrams in between which it cannot decode. Is this expected behavior?

```yaml
sensor:
  - platform: wmbus
    meter_id: 0x63354060
    type: multical21
    key: "AA038FAFD5C2B2468C24B0F1308CCAA5"
    add_prefix: false
    mode: C1
    lqi:
      name: "LQI"
    rssi:
      name: "RSSI"
    total_water_m3:
      name: "Verbrauch"
```

```log
[15:12:23][V][rxLoop:167]: Have 40 bytes from CC1101 Rx, RSSI: -56 dBm LQI: 128
[15:12:23][D][mbus:023]: Processing C1 B frame
[15:12:24][V][mbus:027]: Frame: 25442D2C604035631B168D20A100470421A0B0189DD57F6BD3B7AB8F6138097C3C00E8B6B8AD (38) [with CRC]
[15:12:24][V][mbus:165]: Validating CRC for Block1 + Block2
[15:12:24][V][crc:031]:     calculated: 0xB8AD, read: 0xB8AD
[15:12:24][V][mbus:062]: Frame: 23442D2C604035631B168D20A100470421A0B0189DD57F6BD3B7AB8F6138097C3C00E8B6 (36) [without CRC]
[15:12:24][VV][wmbus:061]: Have data from CC1101 ...
[15:12:24][D][wmbus:097]: Using driver 'multical21' for ID [0x63354060] RSSI: -56 dBm LQI: 128 Frame: C1 B T: 23442D2C604035631B168D20A100470421A0B0189DD57F6BD3B7AB8F6138097C3C00E8B6 (36)
[15:12:24][VV][wmbus:099]: Key defined, trying to decrypt telegram ...
[15:12:24][VV][utils:109]: (ELL)  CI: 8D  offset: 17
[15:12:24][VV][utils:112]: (ELL)  IV: 2D2C604035631B162000470421000000
[15:12:24][VV][utils:115]: (ELL) KEY: AA038FAFD5C2B2468C24B0F1308CCAA5 (16)
[15:12:24][VV][utils:124]: (ELL) AES_CTR decrypting: A0B0189DD57F6BD3B7AB8F6138097C3C00E8B6 (19)
[15:12:24][VV][utils:152]: (ELL) AES_CTR  decrypted: 767779EDA864F10000AA270200A51B0200080C (19)
[15:12:24][V][wmbus:283]: Validating CRC for ELL payload
[15:12:24][V][crc:046]:     calculated: 0x7776, read: 0x7776
[15:12:24][D][wmbus:104]: Decrypted T : 23442D2C604035631B168D20A100470421767779EDA864F10000AA270200A51B0200080C (36)
[15:12:24][D][wmbus:161]: Can't get value(s) from telegram for ID [0x63354060]
[15:12:24][W][component:232]: Component wmbus took a long time for an operation (138 ms).
[15:12:24][W][component:233]: Components should block for at most 30 ms.
[15:12:39][V][rxLoop:167]: Have 40 bytes from CC1101 Rx, RSSI: -56 dBm LQI: 128
[15:12:39][D][mbus:023]: Processing C1 B frame
[15:12:39][V][mbus:027]: Frame: 25442D2C604035631B168D20A20147042183D81007A4A9C0D89331B7D26BD13CE4F09555D772 (38) [with CRC]
[15:12:39][V][mbus:165]: Validating CRC for Block1 + Block2
[15:12:39][V][crc:031]:     calculated: 0xD772, read: 0xD772
[15:12:39][V][mbus:062]: Frame: 23442D2C604035631B168D20A20147042183D81007A4A9C0D89331B7D26BD13CE4F09555 (36) [without CRC]
[15:12:39][VV][wmbus:061]: Have data from CC1101 ...
[15:12:39][D][wmbus:097]: Using driver 'multical21' for ID [0x63354060] RSSI: -56 dBm LQI: 128 Frame: C1 B T: 23442D2C604035631B168D20A20147042183D81007A4A9C0D89331B7D26BD13CE4F09555 (36)
[15:12:39][VV][wmbus:099]: Key defined, trying to decrypt telegram ...
[15:12:39][VV][utils:109]: (ELL)  CI: 8D  offset: 17
[15:12:39][VV][utils:112]: (ELL)  IV: 2D2C604035631B162001470421000000
[15:12:39][VV][utils:115]: (ELL) KEY: AA038FAFD5C2B2468C24B0F1308CCAA5 (16)
[15:12:39][VV][utils:124]: (ELL) AES_CTR decrypting: 83D81007A4A9C0D89331B7D26BD13CE4F09555 (19)
[15:12:39][VV][utils:152]: (ELL) AES_CTR  decrypted: 767779EDA864F10000AA270200A51B0200080C (19)
[15:12:39][V][wmbus:283]: Validating CRC for ELL payload
[15:12:39][V][crc:046]:     calculated: 0x7776, read: 0x7776
[15:12:39][D][wmbus:104]: Decrypted T : 23442D2C604035631B168D20A201470421767779EDA864F10000AA270200A51B0200080C (36)
[15:12:39][D][wmbus:161]: Can't get value(s) from telegram for ID [0x63354060]
[15:12:39][W][component:232]: Component wmbus took a long time for an operation (137 ms).
[15:12:39][W][component:233]: Components should block for at most 30 ms.
[15:12:50][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294948421 (now=41125)
[15:12:55][V][rxLoop:167]: Have 40 bytes from CC1101 Rx, RSSI: -56 dBm LQI: 128
[15:12:55][D][mbus:023]: Processing C1 B frame
[15:12:55][V][mbus:027]: Frame: 25442D2C604035631B168D20A302470421BEF0766C3A060A7E3BC56CC482175F92DF53A9CC74 (38) [with CRC]
[15:12:55][V][mbus:165]: Validating CRC for Block1 + Block2
[15:12:55][V][crc:031]:     calculated: 0xCC74, read: 0xCC74
[15:12:55][V][mbus:062]: Frame: 23442D2C604035631B168D20A302470421BEF0766C3A060A7E3BC56CC482175F92DF53A9 (36) [without CRC]
[... 63 lines cut ...]
[15:13:20][VV][api.service:585]: on_ping_response: PingResponse {}
[15:13:38][VV][api.service:576]: on_ping_request: PingRequest {}
[15:13:38][VV][api.service:043]: send_ping_response: PingResponse {}
[15:13:50][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=41125 (now=101126)
[15:13:58][V][rxLoop:167]: Have 40 bytes from CC1101 Rx, RSSI: -56 dBm LQI: 128
[15:13:58][D][mbus:023]: Processing C1 B frame
[15:13:58][V][mbus:027]: Frame: 25442D2C604035631B168D20A712470421DE8DAA22047273C3283B43D9B505D051B2BC1E90CF (38) [with CRC]
[15:13:58][V][mbus:165]: Validating CRC for Block1 + Block2
[15:13:58][V][crc:031]:     calculated: 0x90CF, read: 0x90CF
[15:13:58][V][mbus:062]: Frame: 23442D2C604035631B168D20A712470421DE8DAA22047273C3283B43D9B505D051B2BC1E (36) [without CRC]
[15:13:58][VV][wmbus:061]: Have data from CC1101 ...
[15:13:58][D][wmbus:097]: Using driver 'multical21' for ID [0x63354060] RSSI: -56 dBm LQI: 128 Frame: C1 B T: 23442D2C604035631B168D20A712470421DE8DAA22047273C3283B43D9B505D051B2BC1E (36)
[15:13:58][VV][wmbus:099]: Key defined, trying to decrypt telegram ...
[15:13:58][VV][utils:109]: (ELL)  CI: 8D  offset: 17
[15:13:58][VV][utils:112]: (ELL)  IV: 2D2C604035631B162012470421000000
[15:13:58][VV][utils:115]: (ELL) KEY: AA038FAFD5C2B2468C24B0F1308CCAA5 (16)
[15:13:58][VV][utils:124]: (ELL) AES_CTR decrypting: DE8DAA22047273C3283B43D9B505D051B2BC1E (19)
[15:13:58][VV][utils:152]: (ELL) AES_CTR  decrypted: 767779EDA864F10000AA270200A51B0200080C (19)
[15:13:58][V][wmbus:283]: Validating CRC for ELL payload
[15:13:58][V][crc:046]:     calculated: 0x7776, read: 0x7776
[15:13:58][D][wmbus:104]: Decrypted T : 23442D2C604035631B168D20A712470421767779EDA864F10000AA270200A51B0200080C (36)
[15:13:58][D][wmbus:161]: Can't get value(s) from telegram for ID [0x63354060]
[15:13:58][W][component:232]: Component wmbus took a long time for an operation (131 ms).
[15:13:58][W][component:233]: Components should block for at most 30 ms.
```

mortenx, 2024-04-18:
Same problem ...Can't get value 
it seems like only some telegrams are working

`[15:18:22][D][sntp:080]: Synchronized time: 2024-04-18 15:16:19
[15:18:54][D][mbus:023]: Processing C1 B frame
[15:18:54][D][wmbus:097]: Using driver 'multical21' for ID [0x76395254] RSSI: -72 dBm LQI: 168 Frame: C1 B T: 21442D2C545239761B168D281AA830C820E00023BCF54216DCC001B0D4202FB106F0 (34)
[15:18:54][D][wmbus:104]: Decrypted T : 21442D2C545239761B168D281AA830C8203AB97934DD82AF71000600000002000000 (34)
[15:18:54][D][wmbus:161]: Can't get value(s) from telegram for ID [0x76395254]

[15:21:51][D][mbus:034]: Processing T1 A frame
[15:22:09][D][mbus:023]: Processing C1 B frame
[15:22:09][D][wmbus:097]: Using driver 'multical21' for ID [0x76395254] RSSI: -72 dBm LQI: 156 Frame: C1 B T: 21442D2C545239761B168D281CD030C820D96CD99578EA9173DC08789CD761993124 (34)
[15:22:09][D][wmbus:104]: Decrypted T : 21442D2C545239761B168D281CD030C8203AB97934DD82AF71000600000002000000 (34)
[15:22:09][D][wmbus:161]: Can't get value(s) from telegram for ID [0x76395254]`

cenobitedk, 2024-07-17:
> what does the number in Decrypted T between the brackets indicate (34), (37) ?

That's the length of the telegram.

The Multical 21 sends out 2 different telegram lengths, the short one most frequently (every 16 sec.) and the long one once in a while. I can't remember the frequency of the long one, sorry, but I think it's something about 94 sec. 🤷‍♂️

I can see that the multical21 driver does not currently support the short telegram, but I can probably build support for it.

cenobitedk, 2024-07-20:
@hanzoh @mortenx Latest version support compact frames as well! :) fixed in #111

closed 2024-09-20 by SzczepanLeon