SzczepanLeon/esphome-components #94: IZAR total_water_m3 and last_month_total_water_m3 not published

rueckert, 2024-04-21:
Hi, 

I am not very experience with ESPHome and the wmbus compontent and I hope someone can give me some advice.

For my izar water meter device, the following sensors are implemented: 

_total_water_m3
last_month_total_water_m3
current_month_total_water_l
transmit_period_s
remaining_battery_life_y
current_alarms
previous_alarms_

However, I am only able to recieve the last four sensors, i.e., 'transmit_period_s', 'remaining_battery_life_y', 'current_alarms', and 'previous_alarms'.  I tried changing the hardware pins, removed all unnecessary template sensors from  https://github.com/zibous/ha-watermeter/tree/master and consulted the large language model G* with no success. 

Here is the ESPHome Log:  

_[11:37:03][D][sntp:080]: Synchronized time: 2024-04-21 11:37:03
[11:37:06][D][mbus:034]: Processing T1 A frame
[11:37:06][D][wmbus:097]: Using driver 'izar' for ID [0x00940787] RSSI: -59 dBm LQI: 128 Frame: T1 A T: 1944F416870794000019A24148001330D20A8B661C04AF6B1E5C (26)
[11:37:06][D][sensor:094]: 'Water Meter current alarm': Sending state 0.00000  with 0 decimals of accuracy
[11:37:06][D][sensor:094]: 'Water Meter previous alarm': Sending state 4.00000  with 0 decimals of accuracy
[11:37:06][D][sensor:094]: 'Water Meter Battery Life': Sending state 4.00000  with 0 decimals of accuracy
[11:37:06][D][sensor:094]: 'Water Meter Transmit Periode': Sending state 8.00000 s with 0 decimals of accuracy
[11:37:06][W][component:237]: Component wmbus took a long time for an operation (75 ms).
[11:37:06][W][component:238]: Components should block for at most 30 ms._

The telegram message itself contains all sensors: https://wmbusmeters.org/analyze/1944F416870794000019A24148001330D20A8B661C04AF6B1E5C:izar 

_telegram=|1944F416870794000019A2_4148001330D20A8B661C04AF6B1E5C|

Auto driver  : not found!
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : f416 dll-mfct (EWT)
004   : 94000019 dll-id (19000094)
008   : 87 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 4148001330D20A8B661C04AF6B1E5C mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"19000094",
    "last_month_total_m3":241.019,
    "remaining_battery_life_y":4,
    "total_m3":252.518,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-03-31",
    "previous_alarms":"leakage",
    "timestamp":"2024-04-21T09:39:51Z"
}_

Maybe, I have an issue with my ESPHome YAML configuration.

Here is my file:

esphome:
  name: water-meter-esp
  comment: ReadsTheTotalWaterConsumption
  
## ----------------------------------------------------------------
## HARDWARE az-delivery-devkit-v4
##          ESP32 240MHz, 520KB RAM, 4MB Flash
## ----------------------------------------------------------------
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
 
external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 12h
    components: [ wmbus ]

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
    
ota:
# Serwer with a statistic data and OTA board update
  password: !secret ota_pswd
  safe_mode: false
  on_begin:
    then:
      - logger.log:
          format: "OTA Start"
          tag: "OTA"
          level: WARN
  on_progress:
    then:
      - logger.log:
          level: WARN
          tag: "OTA"
          format: "OTA progress %0.1f%%"
          args: ["x"]
  on_end:
    then:
      - logger.log:
          format: "OTA End"
          tag: "OTA"
          level: WARN
  on_error:
    then:
      - logger.log:
          format: "OTA update error %d"
          tag: "OTA"
          level: ERROR
          args: ["x"]

web_server:
  port: 80
  version: 2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "NCCWaterMeter"
    password: "7mgQwFaY"

captive_portal:

time:
  platform: sntp

mqtt:
  broker: 192.168.1.112  
  port: "1883"  
  username: NCCMQTT  
  password: mqtt0924

## ------------------------------------------------------------------
##           WMBUS CC1101 --> ESP32 az-delivery-devkit-v4
## ------------------------------------------------------------------
##
##
##                                                               o 1 (3.3V)
##                                                               |
##   ╭――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――x――o―╮
##   |                                                             |
##   |                                                             |
## - | 5v               az-delivery-devkit-v4                      | -- ANT
##   |                                                             |
##   |                          16 17 5  18 19               23    |
##   ╰――x――x――x――x――x――x――x――x――o――x――o――o――o――o――o――o――o――o――o――o―╯
##                              |  |  |  |  |                 |   |
##                              o  |  |  o  |                 |   ╰-o - 2 (GND)
##                              7  o  |  4  o                 o
##                            GDO0 6  | CLK 5                 3
##                               GD02 o    MISO              M0SI
##                                    8
##                                   CSN
##
## ------------------------------------------------------------------
wmbus:
  #mosi_pin: GPIO23  ## SI:   braun   3: MOSI Attached to Hardware SPI controller MOSI SPI Interface
  #miso_pin: GPIO19  ## SO:   grün    5: MISO Attached to Hardware SPI controller MISO SPI Interface
  #clk_pin: GPIO18   ## SCLK: violett 4: SCK  Attached to Hardware SPI controller CLK
  #cs_pin: GPIO05    ## CSN:  orange  8: CSN  Attached to Hardware SPI controller
  #gdo0_pin: GPIO16  ## GD00: gelb    7: RX Clock output. High Impedance !
  #gdo2_pin: GPIO17  ## GD02: weiss   6: TX FIFO status signals. High Impedance !

  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin:  GPIO14
  cs_pin:   GPIO15
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4

sensor:
  - platform: wmbus
    meter_id: 0x00940787
    type: izar
    key: ""
    add_prefix: false
    mode: T1

    total_water_m3:
      id: "watermeter_total"
      name: "Water Meter Display"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: "water"
      #accuracy_decimals: 2
      #icon: mdi:counter
    
    #last_month_total_water_m3: 
      #id: "watermeter_totallastmonth"
      #name: "Water Meter Total Last Month"
      #unit_of_measurement: "m³"
      #device_class: "water"
      #accuracy_decimals: 2
    
    transmit_period_s:
      name: "Water Meter Transmit Periode"
      id: "watermeterlastreading"
      entity_category: "diagnostic"
      icon: mdi:timelapse

    #current_month_total_water_l:
    #  name: "Water Meter Current Month Total in Liters"
    #  unit_of_measurement: "l"
    
    #remaining_battery_life_y: 
    #  name: "Water Meter Battery Life"
    #  id: "watermeter_batterie"
    #  entity_category: "diagnostic"
    #  icon: mdi:battery

    #current_alarms: 
    #  id: "watermeter_current_alarms"
    #  name: "Water Meter current alarm"
    #  entity_category: "diagnostic"
    #  icon: mdi:alarm-light
    
    #previous_alarms: 
    #  id: "watermeter_prev_alarms"
    #  name: "Water Meter previous alarm"
    #  entity_category: "diagnostic"
    #  icon: mdi:alarm-light

balroX, 2024-05-15:
It looks like we have the same problem.
I get all the values in the log and can convert these values into the correct counter reading using https://wmbusmeters.org/analyze.Unfortunately, i don't see them in the gui. 

Log:
```
[20:13:55][V][rxLoop:167]: Have 45 bytes from CC1101 Rx, RSSI: -63 dBm LQI: 128
[20:13:55][D][mbus:034]: Processing T1 A frame
[20:13:55][V][mbus:045]: Frame: 36571C99934D4EC5933A569C5933538F4CAC98E68D65659634B9A59B4B1668E94E2F1A7139C9B2966D31CA3A4B (45) [RAW]
[20:13:55][VV][3of6:095]: Decode 3 out of 6 OK.
[20:13:55][V][mbus:052]: Frame: 1944A511780729640717BCE8A261500013A9AC8062923DFD24AE9ACDEBF3 (30) [with CRC]
[20:13:55][V][mbus:095]: Validating CRC for Block1
[20:13:55][V][crc:031]:     calculated: 0xBCE8, read: 0xBCE8
[20:13:55][V][mbus:115]: Validating CRC for Block2
[20:13:55][V][crc:031]:     calculated: 0xEBF3, read: 0xEBF3
[20:13:55][V][mbus:062]: Frame: 1944A511780729640717A261500013A9AC8062923DFD24AE9ACD (26) [without CRC]
[20:13:55][VV][wmbus:061]: Have data from CC1101 ...
[20:13:55][D][wmbus:097]: Using driver 'izar' for ID [0x64290778] RSSI: -63 dBm LQI: 128 Frame: T1 A T: 1944A511780729640717A261500013A9AC8062923DFD24AE9ACD (26)
[20:13:55][VV][drivers:035]: Value added to map
[20:13:55][VV][drivers:035]: Value added to map
[20:13:55][VV][drivers:035]: Value added to map
[20:13:55][VV][drivers:035]: Value added to map
[20:13:55][V][sensor:043]: 'Watermeter CC1101 LQI': Received new state 128.000000
[20:13:55][D][sensor:094]: 'Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: [2802092902](tel:2802092902)
  state: 128
  missing_state: NO
}
[20:13:55][V][sensor:043]: 'Watermeter CC1101 RSSI': Received new state -63.000000
[20:13:55][D][sensor:094]: 'Watermeter CC1101 RSSI': Sending state -63.00000 dbm with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: [4124674381](tel:4124674381)
  state: -63
  missing_state: NO
}
[20:13:55][V][wmbus:121]: Publishing 'current_alarms' = 0.0000
[20:13:55][V][sensor:043]: 'Watermeter current alarm': Received new state 0.000000
[20:13:55][D][sensor:094]: 'Watermeter current alarm': Sending state 0.00000  with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 520694377
  state: 0
  missing_state: NO
}
[20:13:55][V][wmbus:121]: Publishing 'previous_alarms' = 4.0000
[20:13:55][V][sensor:043]: 'Watermeter previous alarm': Received new state 4.000000
[20:13:55][D][sensor:094]: 'Watermeter previous alarm': Sending state 4.00000  with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: [4069010409](tel:4069010409)
  state: 4
  missing_state: NO
}
[20:13:55][V][wmbus:121]: Publishing 'remaining_battery_life_y' = 8.0000
[20:13:55][V][sensor:043]: 'Watermeter Battery Life': Received new state 8.000000
[20:13:55][D][sensor:094]: 'Watermeter Battery Life': Sending state 8.00000  with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: [2499643748](tel:2499643748)
  state: 8
  missing_state: NO
}
[20:13:55][V][wmbus:121]: Publishing 'transmit_period_s' = 8.0000
[20:13:55][V][sensor:043]: 'Watermeter transmit periode': Received new state 8.000000
[20:13:55][D][sensor:094]: 'Watermeter transmit periode': Sending state 8.00000 s with 0 decimals of accuracy
[20:13:55][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: [1383804497](tel:1383804497)
  state: 8
  missing_state: NO
}
[20:13:55][W][component:237]: Component wmbus took a long time for an operation (316 ms).
[20:13:55][W][component:238]: Components should block for at most 30 ms.
[20:13:56][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=732135 (now=742135)
``` 

WMBusmeters Analyzer
```

telegram=|1944A511780729640717A2_21500013A9F73D443EBAB243C2FEC7|

Auto driver  : izar
Best driver  : unknown 00/00
Using driver : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a511 dll-mfct (DME)
004   : 29640717 dll-id (17076429)
008   : 78 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 21500013A9F73D443EBAB243C2FEC7 mfct specific

{
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"17076429",
    "last_month_total_m3":814.425,
    "remaining_battery_life_y":8,
    "total_m3":819.443,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-05-01",
    "previous_alarms":"leakage",
    "timestamp":"2024-05-15T18:04:42Z"
}

Using: wmbusmeters: 1.16.1-23-gefcc41d
efcc41d107eaea1f78388386d4004b25112b0c04

```

rueckert, 2024-05-16:
Dear @balroX,
I did not invest into that topic any further. It would be great if someone with more experience would give us some hint. I installed the component 'wmbus' locally and tried to debug the code but could not find yet the right place were to look at. 

I added some debug print messages to the function 'mBusDecodeFormatA' in the mbus.cpp file [1] but could not detect any issue. 

[1]
```
bool mBusDecodeFormatA(const WMbusData &t_in, WMbusFrame &t_frame) {
    uint8_t L = t_in.data[0];

    // Validate CRC
    ESP_LOGV(TAG, "Validating CRC for Block 1");
    if (!crcValid(t_in.data, (BLOCK1A_SIZE - 2))) {
      return false;
    }

    // Check length of package is sufficient
    uint8_t numDataBlocks = (L - 9 + 15) / 16; 
    ESP_LOGV(TAG, "--- 1 - Number of Blocks: %u", numDataBlocks);                                          // Data blocks are 16 bytes long + 2 CRC bytes (not counted in L)
    if ((L < 9) || (((L - 9 + (numDataBlocks * 2))) > (t_in.length - BLOCK1A_SIZE))) {   // add CRC bytes for each data block
      ESP_LOGV(TAG, "Package (%u) too short for packet Length: %u", t_in.length, L);
      ESP_LOGV(TAG, "  %u > %u", (L - 9 + (numDataBlocks * 2)), (t_in.length - BLOCK1A_SIZE));
      return false;
    }

    t_frame.frame.insert(t_frame.frame.begin(), t_in.data, ( t_in.data + (BLOCK1A_SIZE - 2)));
    ESP_LOGV(TAG, "--- 2 -Blocks Nr: %u processed", 1);

    // Get all remaining data blocks and concatenate into data array (removing CRC bytes)
    for (uint8_t n{0}; n < numDataBlocks; ++n) {
      const uint8_t *blockStartPtr = (t_in.data + BLOCK1A_SIZE + (n * 18));  // Pointer to where data starts. Each block is 18 bytes
      uint8_t blockSize    = (MIN((L - 9 - (n * 16)), 16));                  // Maximum block size is 16 Data (without 2 CRC)

      ESP_LOGV(TAG, "--- 3 -Determined Block Size for Block %u is %u", (n + 2), blockSize);
      // Validate CRC
      ESP_LOGV(TAG, "Validating CRC for Block %u", (n + 2));
      if (!crcValid(blockStartPtr, (blockSize))) {
        ESP_LOGV(TAG, "--- X - CRC NOT VALID for Block %u", (n + 2));
        return false;
      }

      // Get block data
      t_frame.frame.insert((t_frame.frame.begin() + ((n * 16) + BLOCK1A_SIZE - 2)), blockStartPtr, (blockStartPtr + blockSize));
      ESP_LOGV(TAG, "--- 4 - Blocks Nr: %u processed", n+2);    
    }

    return true;
  }
```

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon