SzczepanLeon/esphome-components #98: need some help with Qwater

alkue1, 2024-06-02:
Hi, 
recently i got watermeters from Qundis installed and stumbled over (https://github.com/zibous/ha-watermeter).  But now i´m stuck with wmbus. it seems to be an error with decoding??

>[14:16:57][D][mbus:013]: Processing C1 A frame
[14:16:57][D][crc:035]:     calculated: 0x0DF9, read: 0x800D  !!!
[14:16:57][D][mbus:013]: Processing C1 A frame
[14:16:57][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 128.00000 lqi with 0 decimals of accuracy
[14:16:57][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -48.00000 % with 0 decimals of accuracy
[14:16:57][D][wmbus:182]: Meter ID [0x52339899] RSSI: -48 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344999833521A07780DFF5F350082D00000F00007C113FFFF42030000FFFF000000001F3568020000008000800080008000800080008000800080008000801A002F046D100D0236 (74)
[14:16:58][D][sensor:094]: 'Device Boot counter': Sending state 25.00000  with 0 decimals of accuracy
[14:17:02][D][mbus:013]: Processing C1 A frame
[14:17:02][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 129.00000 lqi with 0 decimals of accuracy
[14:17:02][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -74.00000 % with 0 decimals of accuracy
[14:17:02][D][wmbus:182]: Meter ID [0x15565139] RSSI: -74 dBm LQI: 129 Frame: C1 A not found in configuration T: 49449344395156153608780DFF5F350082A90000EF0007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D100D0236 (74)
[14:17:05][D][sensor:094]: 'wifi_signal_db': Sending state -35.00000 dBm with 0 decimals of accuracy
[14:17:05][D][sensor:094]: 'Device WLAN Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:17:27][D][mbus:013]: Processing C1 A frame
[14:17:27][D][crc:035]:     calculated: 0x2D45, read: 0xED1E  !!!
[14:17:36][D][mbus:013]: Processing C1 A frame
[14:17:36][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 131.00000 lqi with 0 decimals of accuracy
[14:17:36][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -87.00000 % with 0 decimals of accuracy
[14:17:36][D][wmbus:182]: Meter ID [0x14157183] RSSI: -87 dBm LQI: 131 Frame: C1 A not found in configuration T: 574468508371151494088C0029900F002C25BC210E00F6F1A8015CBE84EA7ABC00300710F6E033014C9F6FDE4A4FCCD55CCE006B865ECDC56F023BA59490EBCD66376C3C9502220327B4C5CD0D880C109852D5D40FBD4986 (88)
[14:17:47][D][mbus:013]: Processing C1 A frame
[14:17:47][D][sensor:094]: '0x00 Watermeter CC1101 LQI': Sending state 137.00000 lqi with 0 decimals of accuracy
[14:17:47][D][sensor:094]: '0x00 Watermeter CC1101 RSSI': Sending state -83.00000 % with 0 decimals of accuracy
[14:17:47][D][wmbus:182]: Meter ID [0x52180675] RSSI: -83 dBm LQI: 137 Frame: C1 A not found in configuration T: 49449344750618521A06780DFF5F350082460000F10007C113FFFF81010000FFFF000000001F3545010000008000800080008000800080008000800300000000000B002F046D110D0236 (74)

This is my config
```
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

## ----------------------------------------------------------------
## EXTERNAL COMPONENTS
## ----------------------------------------------------------------
external_components:

  # uses the latest version from SzczepanLeon
  # https://github.com/SzczepanLeon/esphome-components
  # You can make ESPHome check the repository every time by setting this option to 0s
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0s
    

  # use local component from
  # https://github.com/zdzichu6969/esphome-components
  #- source:
  #    type: local
  #    path: custom_components
  #  components: [backup]

  # use local component from
  # https://github.com/TheStaticTurtle/esphome_syslog
  # - source:
  #     type: local
  #     path: custom_components
  #   components: [syslog]

## ---------------------------------------------------
## syslog service
## tail -f /var/log/syslog | grep "water-meter"
## ---------------------------------------------------
# syslog:
#   ip_address: !secret syslog_server
#   port: !secret syslog_port
#   min_level: DEBUG
#   enable_logger: true

[... 306 lines cut ...]
  # Received Signal Strength (RSSI) is a measure of incoherent
  ## (raw) RF power in a channel.
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "Device WLAN Signal"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

  # device boot counter
  - platform: template
    name: Device Boot counter
    id: bootcounter
    icon: mdi:counter
    accuracy_decimals: 0
    state_class: "measurement"
    entity_category: "diagnostic"
    lambda: return (id(boot_counter));

```

And this is the result of wmbusmeters.org : https://wmbusmeters.org/analyze/49449344989833521A07780DFF5F3500827500007E0007C113FFFF57040000FFFF000000001F35220300000080008000800080008000800080008000800080008020002F046D040D0236:qwater

Could you help me finding the Problem?

SzczepanLeon, 2024-06-02:
Please set proper meter-id in configuration.

alkue1, 2024-06-02:
thanks for your quick reply!

changing it to:
```
- platform: wmbus
    # Meter ID (usually from sticker). Can be specified as decimal or hex.
    # only hex is working for my watermeter !
    # see: https://github.com/SzczepanLeon/esphome-components/issues/6
    # edit watermeterid in the secrets file
    meter_id: 0x52339899
    type: qwater
    add_prefix: true
```

made this specific device disapear in my logs.  :

```
[14:49:33][D][mbus:013]: Processing C1 A frame
[14:49:33][D][wmbus:182]: Meter ID [0x15565137] RSSI: -50 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344375156153608780DFF5F350082A00000F10007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D310D0236 (74)
[14:49:33][D][mbus:013]: Processing C1 A frame
[14:49:33][D][wmbus:182]: Meter ID [0x52180675] RSSI: -84 dBm LQI: 130 Frame: C1 A not found in configuration T: 49449344750618521A06780DFF5F3500824A0000800007C113FFFF81010000FFFF000000001F3545010000008000800080008000800080008000800300000000000B002F046D310D0236 (74)
[14:49:35][D][mbus:013]: Processing C1 A frame
[14:49:35][D][wmbus:182]: Meter ID [0x15565144] RSSI: -87 dBm LQI: 135 Frame: C1 A not found in configuration T: 49449344445156153608780DFF5F350082CB00005F0107B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D310D0236 (74)
[14:49:42][D][mbus:013]: Processing C1 A frame
[14:49:42][D][wmbus:182]: Meter ID [0x52180680] RSSI: -48 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344800618521A06780DFF5F350082C000005F0107C113FFFF23010000FFFF000000001F35830000000080008000800080008000800080008003000000000005002F046D310D0236 (74)
[14:49:49][D][sensor:094]: 'wifi_signal_db': Sending state -33.00000 dBm with 0 decimals of accuracy
[14:49:49][D][sensor:094]: 'Device WLAN Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:49:55][D][mbus:013]: Processing C1 A frame
[14:49:55][D][crc:035]:     calculated: 0x9BDA, read: 0xA097  !!!
[14:50:05][D][mbus:013]: Processing C1 A frame
[14:50:05][D][crc:035]:     calculated: 0xCAAC, read: 0xC959  !!!
[14:50:14][D][sensor:094]: 'Device Boot counter': Sending state 26.00000  with 0 decimals of accuracy
[14:50:26][D][mbus:013]: Processing C1 A frame
[14:50:26][D][wmbus:182]: Meter ID [0x15565139] RSSI: -65 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344395156153608780DFF5F350082AD0000100007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D320D0236 (74)
[14:50:34][D][mbus:013]: Processing C1 A frame
[14:50:34][D][crc:035]:     calculated: 0x9C2C, read: 0xC120  !!!
[14:50:49][D][sensor:094]: 'wifi_signal_db': Sending state -31.00000 dBm with 0 decimals of accuracy
[14:50:49][D][sensor:094]: 'Device WLAN Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:50:54][D][mbus:013]: Processing C1 A frame
[14:50:54][D][wmbus:182]: Meter ID [0x67555767] RSSI: -56 dBm LQI: 128 Frame: C1 A not found in configuration T: 58449344675755673C377807796757556793443C040DFF5F350082DD0000F00007C006FFFF71400100FF2C549900001F35234001002B023D025B0248028C02BC02E802DF02D4023803690391032F02FD170000046D040E0236 (89)
[14:51:00][D][mbus:013]: Processing C1 A frame
[14:51:14][D][mbus:013]: Processing C1 A frame
[14:51:14][D][crc:035]:     calculated: 0xBF38, read: 0x6BED  !!!
[14:51:14][D][sensor:094]: 'Device Boot counter': Sending state 26.00000  with 0 decimals of accuracy
[14:51:16][D][mbus:013]: Processing C1 A frame
[14:51:16][D][wmbus:182]: Meter ID [0x15565134] RSSI: -78 dBm LQI: 132 Frame: C1 A not found in configuration T: 49449344345156153608780DFF5F3500828D0000100007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D330D0236 (74)
[14:51:23][D][mbus:013]: Processing C1 A frame
[14:51:23][D][wmbus:182]: Meter ID [0x15565138] RSSI: -55 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344385156153608780DFF5F350082A70000EF0007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D330D0236 (74)
[14:51:25][D][mbus:013]: Processing C1 A frame
[14:51:25][D][wmbus:182]: Meter ID [0x15565136] RSSI: -80 dBm LQI: 131 Frame: C1 A not found in configuration T: 314493443651561536087A9A0000200B6E0000004B6E000000426CFFFFCB086E000000C2086C1F35326CFFFF046D330D0236 (50)
[14:51:28][D][mbus:013]: Processing C1 A frame
[14:51:28][D][wmbus:182]: Meter ID [0x15565137] RSSI: -50 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344375156153608780DFF5F350082A000007E0007B06EFFFF00000000FFFF000000001F35000000000080008000800080008000800080008000800080008000802F046D330D0236 (74)
[14:51:33][D][mbus:013]: Processing C1 A frame
[14:51:33][D][wmbus:182]: Meter ID [0x15565134] RSSI: -79 dBm LQI: 131 Frame: C1 A not found in configuration T: 314493443451561536087A8E0000200B6E0000004B6E000000426CFFFFCB086E000000C2086C1F35326CFFFF046D330D0236 (50)
[14:51:33][D][mbus:013]: Processing C1 A frame
[14:51:33][D][wmbus:182]: Meter ID [0x52180680] RSSI: -46 dBm LQI: 128 Frame: C1 A not found in configuration T: 49449344800618521A06780DFF5F350082C00000F00007C113FFFF23010000FFFF000000001F35830000000080008000800080008000800080008003000000000005002F046D330D0236 (74)
[14:51:49][D][sensor:094]: 'wifi_signal_db': Sending state -36.00000 dBm with 0 decimals of accuracy
[14:51:49][D][sensor:094]: 'Device WLAN Signal': Sending state 100.00000 % with 0 decimals of accuracy
[14:51:54][D][mbus:013]: Processing C1 A frame
[14:51:54][D][crc:035]:     calculated: 0x249C, read: 0x7990  !!!
[14:52:14][D][sensor:094]: 'Device Boot counter': Sending state 26.00000  with 0 decimals of accuracy
[14:52:16][D][mbus:013]: Processing C1 A frame
```

what does these calaculation (eg  'calculated: 0x249C, read: 0x7990  !!!') mean?

SzczepanLeon, 2024-06-02:
It is CRC. From frame and/or calculated.

alkue1, 2024-06-02:
Do you have any suggestions why there is no result for the meters? is there a point i could start investigating /debug?

SzczepanLeon, 2024-06-02:
Are you 100% sure that meter is transmitting wMBus frames on that frequency?

alkue1, 2024-06-02:
i assumed that, because when i enter these packets (from the log) to the online wmbus website it shows me the correct meter values (link in my first post).... the meter is "Qwater", but there are also some Qcaloric devices shown in the log.

sippeeey, 2024-06-07:
Hi @alkue1, i have a similar problem with the Qwater 5.5 Device.

My telegram data looks: https://wmbusmeters.org/analyze/49449344708828521a072198780dff5f350082a70035865B2883cf79596625201701faccdcedd4506930ee48B78676cf439BeB7e6054de780a19938a15235695B5c3e7af7140e85Bfaefe1dd046d011307365690:qwater

but my actual consumption is: 1885m³

Is there a solution?

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon