SzczepanLeon/esphome-components #102: Installation based on espidf framework fails

grest, 2024-06-17:
I'd like to use espidf framework, instead of arduino - as I need more advance mqtt client options which are available only on espidf. Unfortunatelly during the installation I got the error

`INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/esp-wmbus-test.yaml...
INFO Detected timezone 'Europe/Warsaw'
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esp-wmbus-test (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- wmbus-drivers @ 0.0.0+20240617075635.sha.1ad30e4
Compiling .pioenvs/esp-wmbus-test/src/esphome/components/wmbus/rf_cc1101.o
Compiling .pioenvs/esp-wmbus-test/src/esphome/components/wmbus/wmbus.o
In file included from src/esphome/components/wmbus/rf_cc1101.h:9,
                 from src/esphome/components/wmbus/rf_cc1101.cpp:1:
src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

**********************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:ELECHOUSE_CC1101_SRC_DRV.h
*
**********************************************************************************

 #include <ELECHOUSE_CC1101_SRC_DRV.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp-wmbus-test/src/esphome/components/wmbus/rf_cc1101.o] Error 1
In file included from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.h:17:10: fatal error: WiFiClient.h: No such file or directory

********************************************************************
* Looking for WiFiClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClient.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClient.h
*
********************************************************************

 #include <WiFiClient.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp-wmbus-test/src/esphome/components/wmbus/wmbus.o] Error 1
========================= [FAILED] Took 10.07 seconds =========================
`

spitfire, 2024-10-04:
same here:

```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/t-call.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing t-call (board: esp-wrover-kit; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing git+https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#master
INFO Installing git+https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#master
git version 2.39.5
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-5aiig3og'...
Library Manager: SmartRC-CC1101-Driver-Lib@2.5.7+sha.b8c6af4 has been installed!
INFO SmartRC-CC1101-Driver-Lib@2.5.7+sha.b8c6af4 has been installed!
Library Manager: Installing git+https://github.com/SzczepanLeon/wmbus-drivers#1.3.13
INFO Installing git+https://github.com/SzczepanLeon/wmbus-drivers#1.3.13
git version 2.39.5
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-u3t8fp4y'...
Note: switching to 'c856f7d1eadb337ae3e6f59f3eb3ca652e371735'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Library Manager: wmbus-drivers@0.0.0+20241004124235.sha.c856f7d has been installed!
INFO wmbus-drivers@0.0.0+20241004124235.sha.c856f7d has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Generating assembly for certificate bundle...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.6
|-- wmbus-drivers @ 0.0.0+20241004124235.sha.c856f7d

```
 (...)
```
In file included from src/esphome/components/wmbus/rf_cc1101.h:9,
                 from src/esphome/components/wmbus/rf_cc1101.cpp:1:
src/esphome/components/wmbus/cc1101_rf_settings.h:3:10: fatal error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory

**********************************************************************************
* Looking for ELECHOUSE_CC1101_SRC_DRV.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ELECHOUSE_CC1101_SRC_DRV.h"
* Web  > https://registry.platformio.org/search?q=header:ELECHOUSE_CC1101_SRC_DRV.h
*
**********************************************************************************

 #include <ELECHOUSE_CC1101_SRC_DRV.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/t-call/src/esphome/components/wmbus/rf_cc1101.o] Error 1
In file included from src/esphome/components/wmbus/wmbus.cpp:1:
src/esphome/components/wmbus/wmbus.h:17:10: fatal error: WiFiClient.h: No such file or directory

********************************************************************
* Looking for WiFiClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClient.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClient.h
*
********************************************************************

 #include <WiFiClient.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/t-call/src/esphome/components/wmbus/wmbus.o] Error 1
========================= [FAILED] Took 27.68 seconds =========================

```

SzczepanLeon, 2024-10-04:
espidf is not supported in current version.

VivantSenior, 2024-10-07:
Duplicate of #11