SzczepanLeon/esphome-components #103: Error when trying to update to ESPHome 2024.6.1

vhamburger, 2024-06-20:
Hey, 
When trying to update to ESPHome 2024.6.1 I get the following error on Update/Install:
 >INFO ESPHome 2024.6.1
  INFO Reading configuration /config/esphome/wasserzahler.yaml...
  INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
  INFO Generating C++ source...
  ERROR Circular dependency detected! Please run with -v option to see what functions failed to complete.

Device is currently working on an older version just fine - but upgrade is impossible.

SzczepanLeon, 2024-06-20:
Please share full YAML.
But ...
I can compile with minimal configuration in YAML so probably this issue is not related.to my component.

SzczepanLeon, 2024-09-20:
Please try version 4.0.0

closed 2024-09-20 by SzczepanLeon